zephyrproject-rtos/zephyr #73677: esp32s3: can't set the flash mode to QIO

Elemecca, 2024-06-03:
**Describe the bug**

On `esp32s3` the `CONFIG_ESPTOOLPY_FLASHMODE_QIO` Kconfig setting is available, but choosing it causes the build to fail.

**To Reproduce**

Steps to reproduce the behavior:
1. Start from the `hello_world` sample
2. Set `CONFIG_ESPTOOLPY_FLASHMODE_QIO=y` in `prj.conf`
3. `west build -b esp32s3_devkitm/esp32s3/procpu`
5. Get a linker error

**Expected behavior**

The bootstrap code should switch the flash to QIO mode during startup (either in MCUBoot prior to loading the application or during application startup if MCUBoot is disabled).

**Impact**

The flash mode is effectively forced to DIO, meaning:
- Cache misses on application code in flash stall for twice as long as they should.
- MCUBoot loads the application half as fast as it should.

For my application in production this would be a showstopper, but right now in prototyping I can work around it.

**Logs and console output**

Relevant section of the build log:
```plain
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/maia/code/coco-chameleon/firmware/chameleon-firmware/build/zephyr/zephyr_pre0.map 
: && ccache /home/maia/.local/opt/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/maia/code/coco-chameleon/firmware/chameleon-firmware/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/maia/.local/opt/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0"  -L/home/maia/code/coco-chameleon/firmware/chameleon-firmware/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/src/linker/esp32s3.rom.alias.ld  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.ld  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/../../components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld  -T/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/zephyr/esp32s3/../../components/soc/esp32s3/ld/esp32s3.peripherals.ld  -Wl,--wrap=longjmp  --specs=picolibc.specs  -lc  -lgcc && cd /home/maia/code/coco-chameleon/firmware/chameleon-firmware/build/zephyr && /usr/bin/cmake -E true
/home/maia/.local/opt/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(flash_qio_mode.c.obj):(.literal.bootloader_enable_qio_mode+0x20): undefined reference to `spi_flash_wrap_probe'
/home/maia/.local/opt/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(flash_qio_mode.c.obj):(.literal.bootloader_enable_qio_mode+0x24): undefined reference to `spi_flash_wrap_disable'
/home/maia/.local/opt/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(flash_qio_mode.c.obj): in function `bootloader_enable_qio_mode':
/home/maia/code/coco-chameleon/firmware/modules/hal/espressif/components/bootloader_support/bootloader_flash/src/flash_qio_mode.c:100: undefined reference to `spi_flash_wrap_probe'
/home/maia/.local/opt/zephyr-sdk-0.16.5-1/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: /home/maia/code/coco-chameleon/firmware/modules/hal/espressif/components/bootloader_support/bootloader_flash/src/flash_qio_mode.c:101: undefined reference to `spi_flash_wrap_disable'
collect2: error: ld returned 1 exit status
```

**Environment**
 - OS: Ubuntu 23.10 on amd64
 - Toolchain: Zephyr SDK 0.16.5-1
 - Zephyr: fe03e0c4bb3f8a658efe69b04c430d1a15474c6c (`main` as of this morning)

EricNRS, 2024-06-11:
I just bumped into this for the ESP32-S3 target.  The SOC_CACHE_SUPPORT_WRAP state is what includes these functions:

https://github.com/zephyrproject-rtos/hal_espressif/blob/5191505f915c0b1c706222f4709925a453e0e858/components/bootloader_support/bootloader_flash/src/flash_qio_mode.c#L99-L102

Commenting out these lines doesn't allow the system to boot.  Likewise, adding the implementation file `components/spi_flash/spi_flash_wrap.c` to `modules/hal/espressif/zephyr/esp32s3/CMakeLists.txt` links correctly, but does not boot.

Further down the `CMakeLists.txt` file, `spi_flash_wrap.c` is commented out (https://github.com/zephyrproject-rtos/hal_espressif/blame/5191505f915c0b1c706222f4709925a453e0e858/zephyr/esp32s3/CMakeLists.txt#L157), so I suspect there was an issue with the support to be debugged later.

@sylvioalves  - do you remember what the issue was and the best way to fix it?

EricNRS, 2024-08-11:
This is still an issue, please keep open.

PR #79286 (2024-10-16): bugfix: esp32: allows QIO and QOUT flash modes
Allows QIO and QOUT flash modes to work in:
- esp32s2
- esp32s3
- esp32c2
- esp32c3
- esp32c6

Fixes #73677
commit 7e7672cb (2024-10-16): bugfix: esp32: allows QIO and QOUT flash modes

Allows QIO and QOUT flash mode to work on:
- esp32s2
- esp32s3
- esp32c2
- esp32c3
- esp32c6

Fixes #73677
closed 2024-10-16 by henrikbrixandersen