zephyrproject-rtos/zephyr #73680: CMake install only supports 'focal' and 'jammy' - 'noble' is not yet supported

znmeb, 2024-06-03:
This is a minor quibble. I am building a `Distrobox` container to run Zephyr builds and I started with Ubuntu 24.04 LTS "Noble Numbat". The CMake install aborted, saying "Only 'focal' and 'jammy' are supported". You might want to put a note in the documentation to that effect, although it's quite possible the CMake team will get 'noble' support soon as well.

henrikbrixandersen, 2024-06-04:
Have you tried installing CMake via pip? `pip3 install cmake`.

znmeb, 2024-06-04:
> Have you tried installing CMake via pip? `pip3 install cmake`.

I didn't realize it was available with pip - I know it's in `conda-forge` but the directions said to use `venv` and I'm not enough of a Python programmer to change up other folks' directions, especially if there's a "requirements.txt" file in the process. :-)

kartben, 2024-08-01:
> [...] although it's quite possible the CMake team will get 'noble' support soon as well.

indeed, the following seems to just work so I will close this issue if that's OK

```
docker run -it ubuntu:noble
apt-get update
apt-get install cmake
```

closed 2024-08-01 by kartben