zephyrproject-rtos/zephyr #73703: Zephyr 3.7 security related modules update overview

aescolar, 2024-06-04:
Umbrella issue for tracking security related modules which may require an update before Zephyr 3.7.

- [x] https://github.com/zephyrproject-rtos/mbedtls (@ceolin , @d3zd3z , @tomi-font)
  - Seems to be on upstream latest 3.6.0 LTS https://github.com/Mbed-TLS/mbedtls
- [x] https://github.com/zephyrproject-rtos/tinycrypt (@ceolin )
  - Upstream not being developed anymore https://github.com/intel/tinycrypt
- [ ] https://github.com/zephyrproject-rtos/trusted-firmware-m (@d3zd3z , @ithinuel )
- [ ] https://github.com/zephyrproject-rtos/trusted-firmware-a (@povergoing, @sgrrzhf, @ithinuel )
- [ ] https://github.com/zephyrproject-rtos/uoscore-uedhoc (@rlubos , @StefanHri )

dleach02, 2024-06-05:
Where should we also track any other modules that need to be updated for the? LVGL is one that comes to mind.

aescolar, 2024-06-06:
> Where should we also track any other modules that need to be updated for the? LVGL is one that comes to mind.

@faxe1008 ^^

What we discussed is that that would be the responsibility of each west project maintainer.
Though, @dleach02, you are welcome to make a list and check the status of each module and/or ping each maintainer.

rlubos, 2024-06-06:
[uoscore-uedhoc](https://github.com/zephyrproject-rtos/uoscore-uedhoc) upmerge PRs:
https://github.com/zephyrproject-rtos/uoscore-uedhoc/pull/4
https://github.com/zephyrproject-rtos/zephyr/pull/73848
Went quite smooth, no surprises.

dkalowsk, 2024-06-11:
@povergoing @SgrrZhf @ithinuel does it make sense to move Zephyr 3.7 LTS up to TF-A v 2.10 which is the TF-A LTS release.  The biggest challenge I see with it (not having tested it) is that 2.10 drops support for the cryptocell712, and I don't know who is or isn't using that in Zephyr.

dleach02, 2024-06-14:
that was supposed to be the plan to align all the TFM/TFA/mbedTLS to LTS with the Zephyr LTS.

aescolar, 2024-07-15:
Closing as all were updated

closed 2024-07-15 by aescolar