zephyrproject-rtos/zephyr #73708: net: lib: mqtt_sn_client: net.mqtt_sn.client does not build

katgiadla, 2024-06-04:
**Describe the bug**
The test `tests/net/lib/mqtt_sn_client/net.mqtt_sn.client` does not build.

Observed for:
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160/ns`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf9160dk/nrf9160` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/net/lib/mqtt_sn_client -p nrf9160dk/nrf9160/ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
FAILED: bin/bl2.axf 
: && opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -mcpu=cortex-m33+nofp -O2 -g -DNDEBUG -mcpu=cortex-m33+nofp --entry=Reset_Handler -specs=nano.specs -specs=nosys.specs -Wl,-check-sections -Wl,-fatal-warnings -Wl,--gc-sections -Wl,--no-wchar-size-warning -Wl,--print-memory-usage -Wl,-Map=zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/bl2.map -mfloat-abi=soft -T [/zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/platform/target/nrf91/nrf_common/CMakeFiles/bl2_scatter.dir/common/gcc/tfm_common_bl2.ld](url) bl2/CMakeFiles/bl2.dir/src/flash_map.o bl2/CMakeFiles/bl2.dir/src/security_cnt.o bl2/CMakeFiles/bl2.dir/src/default_flash_map.o bl2/CMakeFiles/bl2.dir/src/provisioning.o bl2/CMakeFiles/bl2.dir/src/thin_psa_crypto_core.o bl2/CMakeFiles/bl2.dir/platform/ext/common/syscalls_stub.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/bl2_main.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/keys.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/flash_map_extended.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/flash_map_legacy.o bl2/CMakeFiles/bl2.dir/platform/ext/target/nordic_nrf/common/core/startup.o bl2/CMakeFiles/bl2.dir/platform/ext/target/nordic_nrf/common/core/startup_nrf91.o -o bin/bl2.axf  bl2/ext/mcuboot/bootutil/libbootutil.a  platform/libplatform_bl2.a  bl2/mbedcrypto/library/libmbedcrypto.a  bl2/mbedcrypto/3rdparty/p256-m/libbl2_p256m.a  bl2/mbedcrypto/library/libmbedcrypto.a  bl2/mbedcrypto/3rdparty/p256-m/libbl2_p256m.a  bl2/mbedcrypto/3rdparty/everest/libbl2_everest.a && :
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_nvmc.o): in function `nrfx_nvmc_page_erase':
zephyr/zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:263: undefined reference to `spm_log_msgval'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_nvmc.o): in function `nrfx_nvmc_words_write':
zephyr/zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:479: undefined reference to `spm_log_msgval'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld: zephyr/zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:480: undefined reference to `spm_log_msgval'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld: zephyr/zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:481: undefined reference to `spm_log_msgval'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_uarte.o): in function `nrf_gpio_pin_port_decode':
zephyr/zephyr/modules/hal/nordic/nrfx/hal/nrf_gpio.h:873: undefined reference to `spm_log_msgval'
opt/zephyr-sdk/arm-zephyr-eabi/bin/lib/gcc/arm-zephyr-eabi/12.2.0/arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_uarte.o):zephyr/zephyr/modules/hal/nordic/nrfx/hal/nrf_gpio.h:882: more undefined references to `spm_log_msgval' follow
Memory region         Used Size  Region Size  %age Used
           FLASH:       50124 B        64 KB     76.48%
             RAM:         23 KB       256 KB      8.98%
collect2: error: ld returned 1 exit status
[5/15] Linking C static library secure_fw/partitions/lib/runtime/libtfm_sprt.a
ninja: build stopped: subcommand failed.
FAILED: modules/trusted-firmware-m/tfm-prefix/src/tfm-stamp/tfm-build tfm/bin/bl2.elf tfm/bin/bl2.bin tfm/bin/bl2.hex tfm/bin/tfm_s.elf tfm/bin/tfm_s.bin tfm/bin/tfm_s.hex tfm/bin/tfm_s_signed.bin tfm/bin/tfm_s_ns_signed.bin tfm/api_ns/interface/lib/s_veneers.o tfm/api_ns/interface/src/tfm_attest_api.c tfm/api_ns/interface/src/tfm_crypto_api.c tfm/api_ns/interface/src/tfm_fwu_api.c tfm/api_ns/interface/src/tfm_its_api.c tfm/api_ns/interface/src/tfm_platform_api.c tfm/api_ns/interface/src/tfm_ps_api.c tfm/api_ns/interface/src/tfm_tz_psa_ns_api.c tfm/api_ns/interface/src/tfm_ioctl_core_ns_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/modules/trusted-firmware-m/tfm-prefix/src/tfm-stamp/tfm-build zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/bl2.elf zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/bl2.bin zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/bl2.hex zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/tfm_s.elf zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/tfm_s.bin zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/tfm_s.hex zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/tfm_s_signed.bin zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/bin/tfm_s_ns_signed.bin zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/lib/s_veneers.o zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_attest_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_crypto_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_fwu_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_its_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_platform_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_ps_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_tz_psa_ns_api.c zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm/api_ns/interface/src/tfm_ioctl_core_ns_api.c 
cd zephyr/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/tfm && toolchains/d61be77f22/usr/local/bin/cmake --build .
[2/275] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.6.0-5099-g127cb9edb6c6`

rlubos, 2024-06-06:
I've created https://github.com/zephyrproject-rtos/zephyr/issues/73862 with a detailed description of the corresponding bug, that  causes the build failure here. Need to wait for its resolution.

carlescufi, 2024-06-12:
Closing since this is a subset of the more general issue here: https://github.com/zephyrproject-rtos/zephyr/issues/73862

closed 2024-06-12 by carlescufi