zephyrproject-rtos/zephyr #73716: Contributor self nominate jeppenodgaard

jeppenodgaard, 2024-06-04:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Jeppe Odgaard
1. GitHub username: @jeppenodgaard

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

Author:
- https://github.com/zephyrproject-rtos/zephyr/pull/64621
- https://github.com/zephyrproject-rtos/zephyr/pull/64774
- https://github.com/zephyrproject-rtos/zephyr/pull/65614
- https://github.com/zephyrproject-rtos/zephyr/pull/68048
- https://github.com/zephyrproject-rtos/zephyr/pull/68689
- https://github.com/zephyrproject-rtos/zephyr/pull/68972
- https://github.com/zephyrproject-rtos/zephyr/pull/69047
- https://github.com/zephyrproject-rtos/zephyr/pull/70159
- https://github.com/zephyrproject-rtos/zephyr/pull/70182

Reviewer:
- https://github.com/zephyrproject-rtos/zephyr/pull/64412
- https://github.com/zephyrproject-rtos/zephyr/pull/71169
- https://github.com/zephyrproject-rtos/zephyr/pull/71338
- https://github.com/zephyrproject-rtos/zephyr/pull/72833
- https://github.com/zephyrproject-rtos/zephyr/pull/72990
- https://github.com/zephyrproject-rtos/zephyr/pull/73141
- https://github.com/zephyrproject-rtos/zephyr/pull/73164

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

jeppenodgaard, 2024-06-04:
The issue should probably have been created before this related PR:
https://github.com/zephyrproject-rtos/zephyr/pull/72896

teburd, 2024-06-05:
+1 from me

nashif, 2024-06-12:
https://github.com/zephyrproject-rtos/zephyr/invitations

closed 2024-06-12 by nashif

jeppenodgaard, 2024-06-13:
My PR approval currently does not count:
https://github.com/zephyrproject-rtos/zephyr/pull/73141#pullrequestreview-2115941532

I think I currently have triage access. The [collaborator PR](https://github.com/zephyrproject-rtos/zephyr/pull/72896) was opened and merged before this issue. Could that be causing problems?

kartben, 2024-06-13:
> My PR approval currently does not count: [#73141 (review)](https://github.com/zephyrproject-rtos/zephyr/pull/73141#pullrequestreview-2115941532)
> 
> I think I currently have triage access. The [collaborator PR](https://github.com/zephyrproject-rtos/zephyr/pull/72896) was opened and merged before this issue. Could that be causing problems?

and you accepted the aforementioned invite, yes?

jeppenodgaard, 2024-06-14:
> and you accepted the aforementioned invite, yes?

Yes. It shows up under https://github.com/settings/organizations now.

jeppenodgaard, 2024-06-19:
@nashif do you have any idea why [my approvals](https://github.com/zephyrproject-rtos/zephyr/pull/73164) does not count?
It says "Collaborator" next to my name and I am ["Drivers: Sensors" collaborator](https://github.com/zephyrproject-rtos/zephyr/blob/main/MAINTAINERS.yml#L1860)