zephyrproject-rtos/zephyr #73723: tests: drivers: can: api: failure on mr_canhubk3

henrikbrixandersen, 2024-06-04:
**Describe the bug**
The `tests/drivers/can/api/` test suite fails to complete on the `mr_canhubk3` board. It gets stuck on `test_filters_preserved_through_classic_to_fd_mode_change`.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b mr_canhubk3/s32k344 tests/drivers/can/api/`
2. `west flash`
3. Notice test getting stuck on console

**Expected behavior**
Test suite runs to completion with success.

**Impact**
Potential breakage in the FlexCAN driver for the NXP S32K platform.

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-5275-gca19e6bbaa54 ***
Running TESTSUITE can_classic
===================================================================
START - test_add_filter
 PASS - test_add_filter in 0.001 seconds
===================================================================
START - test_add_invalid_ext_filter
E: invalid filter with extended (29-bit) CAN ID 0x1fffffff, CAN ID mask 0x20000000
E: invalid filter with extended (29-bit) CAN ID 0x20000000, CAN ID mask 0x1fffffff
 PASS - test_add_invalid_ext_filter in 0.015 seconds
===================================================================
START - test_add_invalid_null_filter
 PASS - test_add_invalid_null_filter in 0.001 seconds
===================================================================
START - test_add_invalid_std_filter
E: invalid filter with standard (11-bit) CAN ID 0x7ff, CAN ID mask 0x800
E: invalid filter with standard (11-bit) CAN ID 0x800, CAN ID mask 0x7ff
 PASS - test_add_invalid_std_filter in 0.013 seconds
===================================================================
START - test_bitrate_limits
 PASS - test_bitrate_limits in 0.001 seconds
===================================================================
START - test_classic_get_capabilities
 PASS - test_classic_get_capabilities in 0.001 seconds
===================================================================
START - test_filters_added_while_stopped
 PASS - test_filters_added_while_stopped in 0.002 seconds
===================================================================
START - test_filters_preserved_through_bitrate_change
 PASS - test_filters_preserved_through_bitrate_change in 0.003 seconds
===================================================================
START - test_filters_preserved_through_mode_change
 PASS - test_filters_preserved_through_mode_change in 0.003 seconds
===================================================================
START - test_get_core_clock
 PASS - test_get_core_clock in 0.001 seconds
===================================================================
START - test_get_state
 PASS - test_get_state in 0.001 seconds
===================================================================
START - test_invalid_sample_point
 PASS - test_invalid_sample_point in 0.001 seconds
===================================================================
START - test_max_ext_filters
 PASS - test_max_ext_filters in 0.001 seconds
===================================================================
START - test_max_std_filters
 PASS - test_max_std_filters in 0.001 seconds
===================================================================
START - test_receive_timeout
 PASS - test_receive_timeout in 0.101 seconds
===================================================================
START - test_recover
[... 108 lines cut ...]
TESTSUITE can_transceiver succeeded
Running TESTSUITE can_utilities
===================================================================
START - test_can_bytes_to_dlc
 PASS - test_can_bytes_to_dlc in 0.001 seconds
===================================================================
START - test_can_dlc_to_bytes
 PASS - test_can_dlc_to_bytes in 0.001 seconds
===================================================================
START - test_can_frame_matches_filter
 PASS - test_can_frame_matches_filter in 0.001 seconds
===================================================================
TESTSUITE can_utilities succeeded
Running TESTSUITE canfd
===================================================================
START - test_canfd_get_capabilities
 PASS - test_canfd_get_capabilities in 0.001 seconds
===================================================================
START - test_filters_preserved_through_classic_to_fd_mode_change
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: ca19e6bbaa541d606dbbb8a6d267f448d231c00e

henrikbrixandersen, 2024-06-04:
@manuargue, @dleach02: FYI, I am looking into it.

henrikbrixandersen, 2024-06-04:
Same issue observed on the `frdm_mcxn947` board while enabling FlexCAN in https://github.com/zephyrproject-rtos/zephyr/pull/72481

henrikbrixandersen, 2024-06-04:
Similar issue on the `ucans32k1sic` board.

dleach02, 2024-06-04:
do you have a bisect of when this was introduced?

henrikbrixandersen, 2024-06-04:
> do you have a bisect of when this was introduced?

Not yet, will look further into it later today.

manuargue, 2024-06-18:
hi @henrikbrixandersen I did a little of debugging and although I couldn't figured out yet what's wrong, I've noticed the following that it may hopefully bisecting this issue:
```
      zephyr      / hal_nxp
bad   07f132b1b50 
      69360d2f386 
good  eeec09eb9ac / 9b2693c7659 (manifest) or with 862e001504b (master)
```
so it seems that the culprit is in zephyr side and not in hal_nxp, and one of these commits may have introduce the issue:
```
git log --follow --oneline eeec09eb9ac..0561d74d31c -- drivers/can/can_mcux_flexcan.c

633065ec3b7 drivers: can: mcux: flexcan: calculate and set proper TDCO
68096cedaec drivers: can: rename struct can_driver_config fields
0291c551a20 drivers: use `DT_INST_NODE_HAS_COMPAT`
71fe0f413bd drivers: can: remove unnecessary asserts
6d35ec718fc drivers: can: add support for specifying minimum supported CAN bitrate
a57db0ddcb9 drivers: can: rework support for manual bus-off recovery
0c13f3888d2 dts: bindings: can: remove deprecated properties for initial timing
5e9da132001 drivers: can: propagate CAN controller operation mode to CAN transceiver
4340724fd04 drivers: can: use common accessor for getting maximum supported bitrate
3436c93387f drivers: can: remove run-time RTR filtering, add build-time RTR filter
75117a0debe drivers: can: remove CAN_FILTER_FDF flag
```
I'll update if I find anything interesting.

bperseghetti, 2024-06-25:
Have we found anything that seems to be the root cause of this? I believe this is also still affecting mcx as well.

dleach02, 2024-06-25:
If someone has a platform that reproduces the problem can they bisect things to try to identify when things when bad?

dleach02, 2024-06-25:
Can someone bisect this issue if they have a system that can reproduce the problem?

henrikbrixandersen, 2024-06-25:
> Can someone bisect this issue if they have a system that can reproduce the problem?

We've tried. The bisect results are inconclusive. I will find some time to dig even deeper into this, but I've been busy with other tasks.

henrikbrixandersen, 2024-06-27:
Solved here: #75082

PR #75082 (2024-06-29): tests: drivers: can: api: restore default bitrates
Restore the default CAN bitrates after having tested setting bitrate and timing. This ensures the arbitration phase bitrate is proportional to the data phase bitrate, allowing CAN FD transmission tests to complete.

Fixes: #73723
commit 1c0f241b (2024-06-29): tests: drivers: can: api: restore default bitrates

Restore the default CAN bitrates after having tested setting bitrate and
timing. This ensures the arbitration phase bitrate is proportional to the
data phase bitrate, allowing CAN FD transmission tests to complete.

Fixes: #73723
closed 2024-06-29 by nashif