zephyrproject-rtos/zephyr #73736: Deinitialize disk devices within all in tree filesystems

danieldegrasse, 2024-06-04:
Once #72700 merges, all in tree filesystem implementations (EXT2, LittleFS) need to be updated to deinitialize disk devices after unmounting completes successfully. This will allow hotpluggable devices using these filesystems (like SD cards) to be removed.

ptg4, 2026-10-18:
Hi @danieldegrasse

I'm part of a group of four students, and we are interested in taking this on as part of a six-week course project.
It doesn't seem like it's still being actively worked on, but we wanted to leave a message here just to double check.

danieldegrasse, 2026-10-19:
> 7\. Hi [@danieldegrasse](https://github.com/danieldegrasse)
>         I'm part of a group of four students, and we are interested in taking this on as part of a six-week course project.
>         It doesn't seem like it's still being actively worked on, but we wanted to leave a message here just to double check.

It isn't, no- if you'd like to pick this up that would be great :)