zephyrproject-rtos/zephyr #73751: stm32: clock_control_on might return before peripheral is properly clocked

taltenbach, 2024-06-04:
**Describe the bug**
According to [RM0433](https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), section 8.5.10, the following sequence must be used on STM32H7 MCUs when enabling a peripheral's clock to guarantee the peripheral is properly clocked before accessing its registers:
1. Set the corresponding PERxEN bit in the RCC_xxxxENR register.
2. Read back the RCC_xxxxENR register.
3. Perform a dummy read operation to any register of the enabled peripheral.

(Note that this sequence assumes the domain in which the peripheral is located is in DRun mode. Otherwise, additional steps are required.)

However, the current implementation of `clock_control_on` on STM32H7 devices currently only performs step 1 and 2 (see [here](https://github.com/zephyrproject-rtos/zephyr/blob/459517badcaacad9cf4d1781c43c4b34e62771ae/drivers/clock_control/clock_stm32_ll_h7.c#L359C19-L359C41)).

This means an access to a register of the enabled peripheral just after the call to `clock_control_on` might silently fails, leading to hard to investigate issues. This is probably especially true for peripherals having a low clock frequency, in comparison to the CPU clock.

I suspect this issue to also impact other STM32 series. For example, [RM0481](https://www.st.com/resource/en/reference_manual/rm0481-stm32h52333xx-stm32h56263xx-and-stm32h573xx-armbased-32bit-mcus-stmicroelectronics.pdf), section 11.4.16, suggests that a similar sequence should be followed on STM32H5 MCUs.

Relates to https://github.com/zephyrproject-rtos/zephyr/issues/73580.

**To Reproduce**
In theory, it should be possible that the following sequence fails on a STM32 MCU:
* Call `clock_control_on` to enable the clock of a given peripheral P.
* After `clock_control_on` has returned, immeditaly access a register R of the peripheral P.

**Expected behavior**
The peripheral is properly clock after `clock_control_on` has returned and the access to the register R succeeds.

**Impact**
According to the reference manual, the peripheral might not be properly clocked when `clock_control_on` returns, which means the access to the register R might silently fails.

**Logs and console output**
N/A

**Environment**
 - OS: N/A
 - Toolchain: N/A
 - Commit SHA: Zephyr main (https://github.com/zephyrproject-rtos/zephyr/commit/459517badcaacad9cf4d1781c43c4b34e62771ae)

**Additional context**
Relevant quotes from the RM0433 8.5.10:
> **A maximum delay of two periods of the enabled clock may occur between the enable command and the first rising edge of the clock.** The enable command can be the rising edge of the PERxEN bits of RCC_xxxxENR registers, or a kernel clock request asserted by a peripheral. 

> The following sequence can be followed to avoid this issue:
a) Enable the peripheral clocks (i.e. allocate the peripheral) by writing the corresponding PERxEN bit to ‘1’ in the RCC_xxxxENR register,
b) Read back the RCC_xxxxENR register to make sure that the previous write operation is not pending into a write buffer.
c) [...] _(step not needed if the domain is in DRun mode)_
d) **Perform a dummy read operation into a register of the enabled peripheral. This operation will take at least 2 clock cycles, which is equal to the max delay of the enable command.**
e) The peripheral can then be used.

erwango, 2024-06-27:
Thanks for raising this issue.

This description is correct but some notes should be added:

> **Expected behavior**
The peripheral P is properly clock after `clock_control_on` has returned and the access to the register R succeeds.

Not exactly. API `clock_control_on` specifies: " On success, the clock is enabled and ready when this function returns.".
hence that the clock is enabled and nothing else. Nothing is specified on the requesting peripheral.

Coming to the way to fix this, the most simple would be performing systematic dummy read in each peripheral driver following return from `clock_control_on`.
We can also think of alternatives like getting the requesting peripheral to provide a reg address where the dummy read should be performed in the opaque `clock_control_subsys_t` structure provided as argument to `clock_control_on` and have the clock_control driver performing the read.
Both have pro's and cons but I'd prefer the first one.

taltenbach, 2024-08-27:
@erwango Sorry for the delay! Could you please remove the "Stale" label?

> Not exactly. API clock_control_on specifies: " On success, the clock is enabled and ready when this function returns.".
hence that the clock is enabled and nothing else. Nothing is specified on the requesting peripheral.

Yes, you're right, to be rigorous I should have written something like "The peripheral P is properly clock after `clock_control_on` has returned, which implies the access to the register R cannot fail because of the state of the peripheral's clock" (assuming the clock is not disabled in the meantime of course). However, in most cases I guess an enabled and ready peripheral's clock is a sufficient condition for guaranteeing a successful access to the peripheral's registers.

> Coming to the way to fix this, the most simple would be performing systematic dummy read in each peripheral driver following return from clock_control_on.
We can also think of alternatives like getting the requesting peripheral to provide a reg address where the dummy read should be performed in the opaque clock_control_subsys_t structure provided as argument to clock_control_on and have the clock_control driver performing the read.
Both have pro's and cons but I'd prefer the first one.

I agree the first solution is the most straightforward, however I guess it might be error-prone: it would be easy to forget adding the dummy read after the return from `clock_control_on`. It would also not match the specifications of `clock_control_on` since the latter would no longer guarantee that the peripheral's clock is ready on return.

One idea I had in mind is that we might perhaps perform at the end of `clock_control_on` a dummy 32-bit read to the very beginning of the peripheral's memory area, which I guess can be retrieved from the device tree. The problem is that for some peripherals this might have side effects, e.g. if the read register contains some clear-on-read bits. The second solution you suggested might be an alternative but perhaps the address from which to perform the dummy read could also be instead indicated for each peripheral in the device tree, using a new dedicated property?

I haven't had the chance to use Zephyr for a quite a bit of time so my apologies if my ideas are not very relevant.

closed 2024-11-11 by github-actions[bot]