zephyrproject-rtos/zephyr #73757: [NXP][CAN]CAN-FD Test Fail on i.MX 93 EVK

CkovMk, 2024-06-05:
**Describe the bug**
As title. `zephyr/tests/drivers/can/api` test fails on i.MX 93 EVK. Classic CAN test passes, CAN-FD test fails.

**To Reproduce**
Steps to reproduce the behavior:
1. west build 
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
Test should pass.

**Impact**
It seems CAN-FD is not working on i.MX 93 EVK.

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-3652-g12e5c8a739b2 ***
Running TESTSUITE can_classic
===================================================================
START - test_add_filter
 PASS - test_add_filter in 0.001 seconds
===================================================================
START - test_add_invalid_ext_filter
E: invalid filter with extended (29-bit) CAN ID 0x1fffffff, CAN ID mask 0x20000000
E: invalid filter with extended (29-bit) CAN ID 0x20000000, CAN ID mask 0x1fffffff
 PASS - test_add_invalid_ext_filter in 0.015 seconds
===================================================================
START - test_add_invalid_null_filter
 PASS - test_add_invalid_null_filter in 0.001 seconds
===================================================================
START - test_add_invalid_std_filter
E: invalid filter with standard (11-bit) CAN ID 0x7ff, CAN ID mask 0x800
E: invalid filter with standard (11-bit) CAN ID 0x800, CAN ID mask 0x7ff
 PASS - test_add_invalid_std_filter in 0.013 seconds
===================================================================
START - test_bitrate_limits
 PASS - test_bitrate_limits in 0.001 seconds
===================================================================
START - test_classic_get_capabilities
 PASS - test_classic_get_capabilities in 0.001 seconds
===================================================================
START - test_filters_added_while_stopped
 PASS - test_filters_added_while_stopped in 0.002 seconds
===================================================================
START - test_filters_preserved_through_bitrate_change
 PASS - test_filters_preserved_through_bitrate_change in 0.003 seconds
===================================================================
START - test_filters_preserved_through_mode_change
 PASS - test_filters_preserved_through_mode_change in 0.003 seconds
===================================================================
START - test_get_core_clock
 PASS - test_get_core_clock in 0.001 seconds
===================================================================
START - test_get_state
 PASS - test_get_state in 0.001 seconds
===================================================================
START - test_invalid_sample_point
 PASS - test_invalid_sample_point in 0.001 seconds
===================================================================
START - test_max_ext_filters
 PASS - test_max_ext_filters in 0.001 seconds
===================================================================
START - test_max_std_filters
 PASS - test_max_std_filters in 0.001 seconds
===================================================================
START - test_receive_timeout
 PASS - test_receive_timeout in 0.101 seconds
===================================================================
[... 233 lines cut ...]

SUITE FAIL -  58.33% [canfd]: pass = 7, fail = 5, skip = 0, total = 12 duration = 0.077 seconds
 - PASS - [canfd.test_canfd_get_capabilities] duration = 0.001 seconds
 - FAIL - [canfd.test_filters_preserved_through_classic_to_fd_mode_change] duration = 0.015 seconds
 - FAIL - [canfd.test_filters_preserved_through_fd_to_classic_mode_change] duration = 0.015 seconds
 - PASS - [canfd.test_invalid_sample_point] duration = 0.001 seconds
 - FAIL - [canfd.test_send_receive_classic] duration = 0.014 seconds
 - FAIL - [canfd.test_send_receive_fd] duration = 0.013 seconds
 - FAIL - [canfd.test_send_receive_mixed] duration = 0.013 seconds
 - PASS - [canfd.test_set_bitrate_data_while_started] duration = 0.001 seconds
 - PASS - [canfd.test_set_bitrate_too_high] duration = 0.001 seconds
 - PASS - [canfd.test_set_timing_data_max] duration = 0.001 seconds
 - PASS - [canfd.test_set_timing_data_min] duration = 0.001 seconds
 - PASS - [canfd.test_set_timing_data_while_started] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK 0.16.3
 - Commit SHA or Version used

henrikbrixandersen, 2024-06-05:
I believe this to be a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/73723 - the symptoms are very similar to what I am seeing on other platforms.

aescolar, 2024-06-11:
Closing as duplicate. Please reopen if it is not

closed 2024-06-11 by aescolar