zephyrproject-rtos/zephyr #73794: spi: stm32: Question regarding SPI performance

lmajewski, 2024-06-05:
I was not sure if this issue shall be regarded as bug or possible enhancement.

Recently I was debugging SPI transmission - the task is to send 64 bytes in a chunk with:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/oa_tc6.c#L282

Sending 1 byte takes ~400 ns:

[image]

But there is a large "gap" between each byte send (~1.1us):

[image]

Zephyr: 3.6.99: SHA1: c56134ebff9be049885bd59a0b214c55e8b5335e
hal_stm32    modules/hal/stm32            54724a815d23b7f518fa88eb5fb7e991109e191b https://github.com/zephyrproject-rtos/hal_stm32

The board is `nucleo_g431rb`

The question is as follows - as I'm sending those 64B as a single array (at least this is my intention) - why there is such large overhead when sending a single byte?

Am I missing some options to be enabled? Like DMA for SPI or such?

lmajewski, 2024-06-05:
@FRASTM @ABOSTM - I've grepped through the sources and it looks like the `dmas = <>` properties are used on `stm32f4*` and `stm32h7*`. The [User Manual for stm32g49x](file:///home/lukma/Downloads/stm32g491cc-4.pdf) has the information that this SOC supports DMA/DMAMUX. Is this feature (i.e. DMA + SPI) supported in Zephyr for STM32?

ajarmouni-st, 2024-06-05:
> Is this feature (i.e. DMA + SPI) supported in Zephyr for STM32?

Yes, see `tests/drivers/spi/spi_loopback/boards/nucleo_g431rb.overlay`

You can test it with `west build -p always -b nucleo_g431rb tests/drivers/spi/spi_loopback -T drivers.spi.stm32_spi_dma.loopback`

lmajewski, 2024-06-06:
@ajarmouni-st - Thanks for sharing this information. Indeed - I was able to run this tests (with MOSI/MISO loopback) on my nucleo board. It works :-)

However, with my application - I need to dig deeper as it is not working out of the box.

closed 2024-06-06 by lmajewski

ajarmouni-st, 2024-06-06:
> I need to dig deeper as it is not working out of the box

Potentially because of the SRAM where Eth buffers are located, there are restrictions on which DMA/DMA Mux, SPI, can be used with which SRAM.