zephyrproject-rtos/zephyr #73815: Contributor Nomination (josuah)

ubieda, 2024-06-05:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Josuah Demangeon.
1. GitHub username: @josuah 
1. Organization (optional): N/A

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

### Contributions
- https://github.com/zephyrproject-rtos/zephyr/pull/63858
- https://github.com/zephyrproject-rtos/zephyr/pull/73001

### Reviews
- https://github.com/zephyrproject-rtos/zephyr/pull/63470
- https://github.com/zephyrproject-rtos/zephyr/pull/71463
- https://github.com/zephyrproject-rtos/zephyr/pull/72254

ubieda, 2024-06-05:
CC @teburd

teburd, 2024-06-05:
+1 from me

nashif, 2024-06-12:
https://github.com/zephyrproject-rtos/zephyr/invitations

closed 2024-06-12 by nashif

josuah, 2024-06-12:
Thank you all for your support! Now is time for me to read the doc and previous issue/pulls as I keep reviews going... :)