zephyrproject-rtos/zephyr #73820: Atmel Sam E51 fails CAN timing test

PeeJay, 2024-06-06:
**Describe the bug**
I'm developing a custom same51 board and was failing to can_set_bitrate(). After some googling I found the test suite and ran it on my board, where I discovered the can clock was set to only 4MHz by default (which is next to useless). I discovered the divider in the same5x.dtsi was set to 12, hence the low clockrate. Overriding that in the board config to 1 gives 48 MHz which is much better, but still not ideal.

**To Reproduce**
Run the test `tests/drivers/can/timing` on any same51 board

**Expected behavior**
> a CAN FD core clock of 20, 40, or 80 MHz for good node interoperability

**Logs and console output**
Default divider of 12 (4 MHz):
```
*** Booting Zephyr OS build v3.6.0-5453-gc250c330c6bf ***
Running TESTSUITE can_timing
===================================================================
testing on device can@42000400 @ 4000000 Hz
Warning: CiA 601-3 recommends a CAN FD core clock of 20, 40, or 80 MHz for good node interoperability
START - test_timing
testing bitrate 10000, sample point 87.5%: sjw = 12, prop_seg = 0, phase_seg1 = 174, phase_seg2 = 25, prescaler = 2 OK, sample point error 0.0%
testing bitrate 20000, sample point 87.5%: sjw = 12, prop_seg = 0, phase_seg1 = 174, phase_seg2 = 25, prescaler = 1 OK, sample point error 0.0%
testing bitrate 50000, sample point 87.5%: sjw = 5, prop_seg = 0, phase_seg1 = 69, phase_seg2 = 10, prescaler = 1 OK, sample point error 0.0%
testing bitrate 125000, sample point 87.5%: sjw = 2, prop_seg = 0, phase_seg1 = 27, phase_seg2 = 4, prescaler = 1 OK, sample point error 0.0%
testing bitrate 250000, sample point 87.5%: sjw = 1, prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 1 OK, sample point error 0.0%
testing bitrate 500000, sample point 87.5%: 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:179: test_timing_values: (sp_err <= SAMPLE_POINT_MARGIN is false)
sample point error 125 too large
 FAIL - test_timing in 0.083 seconds
===================================================================
START - test_timing_data
testing bitrate 1000000, sample point 75.0%: sjw = 1, prop_seg = 0, phase_seg1 = 2, phase_seg2 = 1, prescaler = 1 OK, sample point error 0.0%
testing bitrate 2000000, sample point 75.0%: bitrate not supported
testing bitrate 4000000, sample point 75.0%: bitrate not supported
testing bitrate 5000000, sample point 75.0%: bitrate not supported
testing bitrate 8000000, sample point 75.0%: bitrate not supported
 PASS - test_timing_data in 0.037 seconds
===================================================================
TESTSUITE can_timing failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [can_timing]: pass = 1, fail = 1, skip = 0, total = 2 duration = 0.120 seconds
 - FAIL - [can_timing.test_timing] duration = 0.083 seconds
 - PASS - [can_timing.test_timing_data] duration = 0.037 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

Divider of 1 (48 MHz):
```
*** Booting Zephyr OS build v3.6.0-5453-gc250c330c6bf ***
Running TESTSUITE can_timing
===================================================================
testing on device can@42000400 @ 48000000 Hz
Warning: CiA 601-3 recommends a CAN FD core clock of 20, 40, or 80 MHz for good node interoperability
START - test_timing
testing bitrate 10000, sample point 87.5%: sjw = 15, prop_seg = 0, phase_seg1 = 209, phase_seg2 = 30, prescaler = 20 OK, sample point error 0.0%
testing bitrate 20000, sample point 87.5%: sjw = 15, prop_seg = 0, phase_seg1 = 209, phase_seg2 = 30, prescaler = 10 OK, sample point error 0.0%
testing bitrate 50000, sample point 87.5%: sjw = 15, prop_seg = 0, phase_seg1 = 209, phase_seg2 = 30, prescaler = 4 OK, sample point error 0.0%
testing bitrate 125000, sample point 87.5%: sjw = 12, prop_seg = 0, phase_seg1 = 167, phase_seg2 = 24, prescaler = 2 OK, sample point error 0.0%
testing bitrate 250000, sample point 87.5%: sjw = 12, prop_seg = 0, phase_seg1 = 167, phase_seg2 = 24, prescaler = 1 OK, sample point error 0.0%
testing bitrate 500000, sample point 87.5%: sjw = 6, prop_seg = 0, phase_seg1 = 83, phase_seg2 = 12, prescaler = 1 OK, sample point error 0.0%
testing bitrate 800000, sample point 80.0%: sjw = 6, prop_seg = 0, phase_seg1 = 47, phase_seg2 = 12, prescaler = 1 OK, sample point error 0.0%
testing bitrate 1000000, sample point 75.0%: sjw = 6, prop_seg = 0, phase_seg1 = 35, phase_seg2 = 12, prescaler = 1 OK, sample point error 0.0%
 PASS - test_timing in 0.101 seconds
===================================================================
START - test_timing_data
testing bitrate 1000000, sample point 75.0%: sjw = 3, prop_seg = 0, phase_seg1 = 17, phase_seg2 = 6, prescaler = 2 OK, sample point error 0.0%
testing bitrate 2000000, sample point 75.0%: sjw = 3, prop_seg = 0, phase_seg1 = 17, phase_seg2 = 6, prescaler = 1 OK, sample point error 0.0%
testing bitrate 4000000, sample point 75.0%: sjw = 1, prop_seg = 0, phase_seg1 = 8, phase_seg2 = 3, prescaler = 1 OK, sample point error 0.0%
testing bitrate 5000000, sample point 75.0%: bitrate not supported
testing bitrate 8000000, sample point 75.0%: 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/can/timing/src/main.c:179: test_timing_values: (sp_err <= SAMPLE_POINT_MARGIN is false)
sample point error 84 too large
 FAIL - test_timing_data in 0.063 seconds
===================================================================
TESTSUITE can_timing failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [can_timing]: pass = 1, fail = 1, skip = 0, total = 2 duration = 0.164 seconds
 - PASS - [can_timing.test_timing] duration = 0.101 seconds
 - FAIL - [can_timing.test_timing_data] duration = 0.063 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: v3.6.0-5453-gc250c330c6bf

henrikbrixandersen, 2024-06-06:
@PeeJay Are you able to submit a fix for the wrong CAN module clock? I do not have access to any Atmel SAM based boards.

PeeJay, 2024-06-11:
Will do.
I've now run into another issue while testing - randomly when recieving can packets I'm mostly getting `<err> can_mcan: Message RAM access failure`. About 1/8 are received successfully. When I connect the debugger (a Jlink) the error instantly stops. I haven't specifically enabled or disabled the MPU or cache.

henrikbrixandersen, 2024-06-27:
[image]

Are you able to check under which of these conditions it occurs?

PeeJay, 2024-07-06:
I suspect it is the first item on the list. It's almost like one of the internal clocks gets changed when the debugger is attached.
I have added the following debug output in can_mcan.c:624 (can_mcan_line_0_isr)
```
err = can_mcan_read_reg(dev, CAN_MCAN_IR, &ir);
	if (err != 0) {
		return;
	}

	LOG_WRN("%x", ir);
```
With debugger attached I get the correct program operation:
```
[00:00:05.010,000] <wrn> can_mcan: 201
[00:00:05.014,000] <inf> main: Received CAN frame
[00:00:05.019,000] <wrn> can_mcan: 1200
[00:00:05.024,000] <wrn> can_mcan: 1200
[00:00:05.028,000] <wrn> can_mcan: 1200
[00:00:05.036,000] <wrn> can_mcan: 210
[00:00:05.040,000] <inf> main: Received CAN frame
[00:00:05.053,000] <wrn> can_mcan: 1200
[00:00:05.058,000] <wrn> can_mcan: 1200
[00:00:05.062,000] <inf> main: Received CAN frame
[00:00:05.067,000] <wrn> can_mcan: 1200
[00:00:05.085,000] <wrn> can_mcan: 210
[00:00:05.090,000] <inf> main: Received CAN frame
[00:00:05.095,000] <wrn> can_mcan: 1200
[00:00:05.100,000] <wrn> can_mcan: 1200
[00:00:05.104,000] <wrn> can_mcan: 1200
[00:00:05.112,000] <wrn> can_mcan: 201
[00:00:05.116,000] <inf> main: Received CAN frame
[00:00:05.129,000] <wrn> can_mcan: 1200
[00:00:05.133,000] <wrn> can_mcan: 1200
[00:00:05.138,000] <wrn> can_mcan: 210
[00:00:05.142,000] <inf> main: Received CAN frame
[00:00:05.147,000] <wrn> can_mcan: 1200
[00:00:05.163,000] <wrn> can_mcan: 201
[00:00:05.167,000] <inf> main: Received CAN frame
[00:00:05.173,000] <wrn> can_mcan: 1200
[00:00:05.177,000] <wrn> can_mcan: 1200
[00:00:05.182,000] <wrn> can_mcan: 1200
[00:00:05.188,000] <wrn> can_mcan: 210
[00:00:05.192,000] <inf> main: Received CAN frame
[00:00:05.206,000] <wrn> can_mcan: 1200
[00:00:05.211,000] <wrn> can_mcan: 1200
[00:00:05.215,000] <inf> main: Received CAN frame
[00:00:05.220,000] <wrn> can_mcan: 1200
[00:00:05.239,000] <wrn> can_mcan: 210
[00:00:05.243,000] <inf> main: Received CAN frame
```

Note what happens here at around 9 seconds when I click disconnect:
```
[00:08:08.962,000] <inf> main: Received CAN frame
[00:08:08.967,000] <wrn> can_mcan: 1200
[00:08:08.972,000] <wrn> can_mcan: 1200
[00:08:08.976,000] <wrn> can_mcan: 1200
[00:08:08.983,000] <wrn> can_mcan: 210
[00:08:08.987,000] <inf> main: Received CAN frame
[00:08:09.001,000] <wrn> can_mcan: 1200
[00:08:09.006,000] <wrn> can_mcan: 1200
[00:08:09.010,000] <inf> main: Received CAN frame
[00:08:09.015,000] <wrn> can_mcan: 1200
[00:08:09.033,000] <wrn> can_mcan: 210
[00:08:09.038,000] <inf> main: Received CAN frame
[00:08:09.043,000] <inf> main: Received CAN frame
[00:08:09.048,000] <inf> main: Received CAN frame
[00:08:09.053,000] <inf> main: Received CAN frame
[00:08:09.058,000] <inf> main: Received CAN frame
[00:08:09.063,000] <inf> main: Received CAN frame
[00:08:09.068,000] <inf> main: Received CAN frame
[00:08:09.073,000] <inf> main: Received CAN frame
[00:08:09.078,000] <inf> main: Received CAN frame
[00:08:09.084,000] <inf> main: Received CAN frame
[00:08:09.089,000] <inf> main: Received CAN frame
[00:08:09.094,000] <inf> main: Received CAN frame
[00:08:09.099,000] <wrn> can_mcan: 200
[00:08:16.084,000] <wrn> can_mcan: 1200
[00:08:16.089,000] <inf> main: Received CAN frame
[00:08:16.094,000] <wrn> can_mcan: 20200
[00:08:16.098,000] <err> can_mcan: Message RAM access failure
[00:08:16.105,000] <inf> main: Received CAN frame
[00:08:18.459,000] <wrn> can_mcan: 1200
[00:08:18.464,000] <inf> main: Received CAN frame
[00:08:18.489,000] <wrn> can_mcan: 20200
[00:08:18.493,000] <err> can_mcan: Message RAM access failure
[00:08:19.886,000] <wrn> can_mcan: 1200
[00:08:19.890,000] <inf> main: Received CAN frame
[00:08:19.895,000] <inf> main: Received CAN frame
[00:08:20.864,000] <wrn> can_mcan: 1200
[00:08:20.868,000] <inf> main: Received CAN frame
[00:08:21.562,000] <wrn> can_mcan: 1200
[00:08:21.567,000] <inf> main: Received CAN frame
[00:08:21.592,000] <wrn> can_mcan: 20200
[00:08:21.596,000] <err> can_mcan: Message RAM access failure
[00:08:24.665,000] <wrn> can_mcan: 1200
[00:08:24.670,000] <inf> main: Received CAN frame
[00:08:24.675,000] <inf> main: Received CAN frame
[00:08:24.680,000] <wrn> can_mcan: 200
[00:08:26.342,000] <wrn> can_mcan: 1200
[00:08:26.346,000] <inf> main: Received CAN frame
[00:08:28.298,000] <wrn> can_mcan: 1200
[00:08:28.302,000] <inf> main: Received CAN frame
[00:08:28.307,000] <inf> main: Received CAN frame
[00:08:28.312,000] <wrn> can_mcan: 200
[00:08:28.337,000] <wrn> can_mcan: 20200
[00:08:28.341,000] <err> can_mcan: Message RAM access failure
[00:08:29.035,000] <wrn> can_mcan: 1200
[00:08:29.040,000] <inf> main: Received CAN frame
```

I intermittently receive frames with large pauses in between.

PeeJay, 2024-07-26:
One more data point - I just accidently discovered that if I send a can message every 5 ms it works as intended, but if I then change it to every 10 ms I get the errors, and that is without touching anything on the board running zephyr.

```
[00:05:37.011,000] <inf> main: Received CAN frame
[00:05:37.016,000] <inf> main: Received CAN frame
[00:05:37.163,000] <err> can_mcan: Message RAM access failure
[00:05:37.169,000] <inf> main: Received CAN frame
[00:05:37.177,000] <err> can_mcan: Message RAM access failure
[00:05:37.316,000] <err> can_mcan: Message RAM access failure
[00:05:37.464,000] <err> can_mcan: Message RAM access failure
[00:05:37.471,000] <inf> main: Received CAN frame
[00:05:37.478,000] <err> can_mcan: Message RAM access failure
[00:05:37.618,000] <err> can_mcan: Message RAM access failure
[00:05:43.500,000] <err> can_mcan: Message RAM access failure
[00:05:43.506,000] <inf> main: Received CAN frame
[00:05:43.511,000] <inf> main: Received CAN frame
[00:05:43.516,000] <inf> main: Received CAN frame
[00:05:43.521,000] <inf> main: Received CAN frame
[00:05:43.527,000] <inf> main: Received CAN frame
[00:05:43.532,000] <inf> main: Received CAN frame
[00:05:43.537,000] <inf> main: Received CAN frame
[00:05:43.542,000] <inf> main: Received CAN frame
[00:05:43.547,000] <inf> main: Received CAN frame
[00:05:43.552,000] <inf> main: Received CAN frame
[00:05:43.557,000] <inf> main: Received CAN frame
[00:05:43.562,000] <inf> main: Received CAN frame
[00:05:43.568,000] <inf> main: Received CAN frame
[00:05:43.573,000] <inf> main: Received CAN frame
[00:05:43.578,000] <inf> main: Received CAN frame
```

PeeJay, 2024-07-31:
Time to close out this bug - I think the problem is somewhere in the RTOS scheduler. 
I ported the working Arduino library into a Zephyr test program but encountered the same issue that the native Zephyr can library exhibits, so I think that rules out any problem with the can code.
Just out of curiosity I disabled the tickless kernel option which results in an immediate crash when the debugging session is stopped.
If Microchip can't be bothered to better support their devices outside of MPLab I'll just use an STM32 instead. Thanks to @henrikbrixandersen for at least trying to help!

nandojve, 2024-08-01:
CC: @scottwcpg , @MyGh64605