zephyrproject-rtos/zephyr #73821: drivers: can: mcp251xfd: driver does not set Transceiver Delay Compensation Offset (TDCO)

henrikbrixandersen, 2024-06-06:
**Describe the bug**
The Microchip MCP251xFD driver does not calculate and set the Transceiver Delay Compensation Offset (TDCO) when CAN FD is enabled. TDCO is needed for correct sampling of own TX line at higher data phase bitrates.

**To Reproduce**
Steps to reproduce the behavior:
1. Inspect the driver code.
2. Notice TDCO is not set up.

**Expected behavior**
TSCO should be calculated using `CAN_CALC_TDCO()` and written to the respective register when using CAN FD.

**Impact**
Likely issues with using higher data phase bitrates.

**Additional context**
Please see the following commits for inspiration on how to calculate and set TDCO:
- https://github.com/zephyrproject-rtos/zephyr/commit/633065ec3b78bc15722b02804ad9d72450ebec71
- https://github.com/zephyrproject-rtos/zephyr/commit/bfad7bc00e734010d059d5ba7eb4cd2187eb1c8d

talih0, 2024-06-09:
I do enable TDC and set TDCO, but it's calculated using a slightly different way:

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/can/can_mcp251xfd.c#L473

which is the same as CAN_CALC_TDCO() but without 1U for sync segment.

The calculation in this driver is the way that's recommended in the mcp251xfd reference manual and that's used in the Linux kernel:

https://lxr.missinglinkelectronics.com/linux/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c#L795

[image]

I've always found the missing +1 from sync segment a bit confusing, and actually reached out to microchip in the forums  about it:
https://forum.microchip.com/s/topic/a5C3l000000MdJVEA0/t381176

henrikbrixandersen, 2024-06-10:
Yeah, I've seen multiple places (data sheets, reference manuals, implementations) not including the SYNC segment in the TDCO calculation. Logically, it should be part of it.

Could we update the mcp251xfd driver to use the common "Zephyr" way of calculating the TDCO? That would provide a uniform TDC behavior across Zephyr CAN FD drivers.

As far as I can tell, the current implementation completely skips TDC if the TDCO is out of bounds, where I believe the better solution would be to program it as close as possible (e.g. at `MCP251XFD_REG_TDC_TDCO_MAX` even if the calculated TDCO is higher)?

PR #74373 (2024-06-17): drivers: can: mcp251xfd: Use macro to set TDCO
Use CAN_CALC_TDCO() macro to set TDCO. There is a minor change how the tdco is calculated. In the macro the SYNC segement is taken into account while it wasn't used before.

Fixes #73821.
commit d2607f09 (2024-06-17): drivers: can: mcp251xfd: Use macro to set TDCO

Use CAN_CALC_TDCO() macro to set TDCO. There is a minor change how the
tdco is calculated. In the macro the SYNC segement is taken into
account while it wasn't used before.

Fixes #73821.
closed 2024-06-17 by nashif