zephyrproject-rtos/zephyr #73823: drivers: can: xmc4xxx: driver does not set Transceiver Delay Compensation Offset (TDCO)

henrikbrixandersen, 2024-06-06:
**Describe the bug**
The Infineon XMC4xxx driver does not calculate and set the Transceiver Delay Compensation Offset (TDCO) when CAN FD is enabled. TDCO is needed for correct sampling of own TX line at higher data phase bitrates.

**To Reproduce**
Steps to reproduce the behavior:
1. Inspect the driver code.
2. Notice TDCO is not set up.

**Expected behavior**
TSCO should be calculated using `CAN_CALC_TDCO()` and written to the respective register when using CAN FD.

**Impact**
Likely issues with using higher data phase bitrates.

**Additional context**
Please see the following commits for inspiration on how to calculate and set TDCO:
- https://github.com/zephyrproject-rtos/zephyr/commit/633065ec3b78bc15722b02804ad9d72450ebec71
- https://github.com/zephyrproject-rtos/zephyr/commit/bfad7bc00e734010d059d5ba7eb4cd2187eb1c8d

talih0, 2024-06-09:
@henrikbrixandersen, can_xmc4xxx doesn't support CAN-FD

henrikbrixandersen, 2024-06-09:
> @henrikbrixandersen, can_xmc4xxx doesn't support CAN-FD

Sorry, my mistake. I had a grep one-liner, which used `CAN_MODE_FD` to see which drivers had CAN FD support. That wrongly matched this driver.

closed 2024-06-09 by henrikbrixandersen