zephyrproject-rtos/zephyr #73824: drivers: can: s32: canxl: driver does not set Transceiver Delay Compensation Offset (TDCO)

henrikbrixandersen, 2024-06-06:
**Describe the bug**
The NXP S32 CANXL driver does not calculate and set the Transceiver Delay Compensation Offset (TDCO) when CAN FD is enabled. TDCO is needed for correct sampling of own TX line at higher data phase bitrates.

**To Reproduce**
Steps to reproduce the behavior:
1. Inspect the driver code.
2. Notice TDCO is not set up.

**Expected behavior**
TSCO should be calculated using `CAN_CALC_TDCO()` and written to the respective register when using CAN FD.

**Impact**
Likely issues with using higher data phase bitrates.

**Additional context**
Please see the following commits for inspiration on how to calculate and set TDCO:
- https://github.com/zephyrproject-rtos/zephyr/commit/633065ec3b78bc15722b02804ad9d72450ebec71
- https://github.com/zephyrproject-rtos/zephyr/commit/bfad7bc00e734010d059d5ba7eb4cd2187eb1c8d

manuargue, 2024-06-06:
@Dat-NguyenDuy pls have a look at this

congnguyenhuu, 2024-06-07:
Hello @henrikbrixandersen , I tried reproducing this issue but I'm NOT possible.
I tried a test CAN traffic with multiple CAN-FD frames between 2 instance CANXL_0 and CANXL_1 on board S32Z270.

I set CAN PE clock 160MHz and bitrate for  arbitration phase 125kbit/s, for data phase 8Mbit/s
[image]

I have this test for reproducing 
[can_fd_traffic.zip](https://github.com/user-attachments/files/15692038/can_fd_traffic.zip)
and the result test
[image]

Do you have any suggestion about bitrate and test steps that help me to reproduce this issue. Thank you.

henrikbrixandersen, 2024-06-07:
> Hello @henrikbrixandersen , I tried reproducing this issue but I'm NOT possible. I tried a test CAN traffic with multiple CAN-FD frames between 2 instance CANXL_0 and CANXL_1 on board S32Z270.

Reproducing this greatly depends on the actual CAN tranceiver used, but nevertheless this needs to be set up. 

You need to set `BTDCC[FTDCEN]` to enable TDC for CAN FD, calculate the TDCO using `CAN_CALC_TDCO()` and write it to `BTDCC[FTDCOFF]`. Check the reference manual chapter on "Transceiver Delay Compensation".

congnguyenhuu, 2024-06-07:
> Reproducing this greatly depends on the actual CAN tranceiver used, but nevertheless this needs to be set up.
>
> You need to set `BTDCC[FTDCEN]` to enable TDC for CAN FD, calculate the TDCO using [...]

yes, I opened PR to fix

PR #73893 (2024-06-11): drivers: can: nxp_s32_canxl: calculate and set the TDCO
Calculate and set the Transceiver Delay Compensation Offset (TDCO).

Fixes #73824
commit 0781caf4 (2024-06-11): drivers: can: nxp_s32_canxl: calculate and set the TDCO

Calculate and set the Transceiver Delay Compensation Offset (TDCO).

Fixes #73824
closed 2024-06-11 by aescolar