zephyrproject-rtos/zephyr #73831: twister -n/--no-clean does not pass tests after first build

Thalley, 2024-06-06:
**Describe the bug**
If you run a test with `twister` with `--no-clean` then it passes on the first run, but fails on subsequent runs. 

**To Reproduce**
Steps to reproduce the behavior:
1. scripts/twister --verbose -p unit_testing --no-clean -i -T tests/net/buf_simple/
2. See `1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 2.14 seconds`
3. scripts/twister --verbose -p unit_testing --no-clean -i -T tests/net/buf_simple/
4. See `0 of 1 test configurations passed (0.00%), 0 failed, 1 errored, 0 skipped with 0 warnings in 2.27 seconds`

Switching to other test cases work, it is only if you run the same test twice. It works as expected without `--no-clean`.

**Expected behavior**
Expect `--no-clean` to fail the test case on subsequent builds

**Impact**
Annoyance

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Linux
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 16.5-1
 - Commit SHA or Version used: e414acfabcc6d948d83136f32971b49b8349f9e6

**Additional context**
N/A

nashif, 2024-06-07:
does not seem to be a twister issue, you can reproduce the same error with:

`west build -b unit_testing tests/net/buf_simple/`

then

`west build -b unit_testing tests/net/buf_simple/ --cmake-only`

the error being:

```
CMake Error at /home/nashif/zephyrproject/zephyr/cmake/modules/extensions.cmake:3351 (message):
  BOARD is not being defined on the CMake command-line, in the environment or
  by the app.
Call Stack (most recent call first):
  /home/nashif/zephyrproject/zephyr/cmake/modules/boards.cmake:61 (zephyr_check_cache)
  /home/nashif/zephyrproject/zephyr/cmake/modules/unittest.cmake:9 (include)
  /home/nashif/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /home/nashif/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:7 (find_package)
```

@tejlmand  any idea?

nashif, 2024-06-25:
@tejlmand  ping

closed 2024-08-05 by carlescufi