zephyrproject-rtos/zephyr #73862: TF-M won't build with CONFIG_DEBUG_OPTIMIZATIONS

rlubos, 2024-06-06:
**Describe the bug**
Since commit f451201228531de98364a04130365598563a1dca (TF-M update to version 2.1.0), it's no longer possible to build with `CONFIG_DEBUG_OPTIMIZATIONS=y`, at least for nRF platforms.

As a consequence, projects enabling `CONFIG_DEBUG_OPTIMIZATIONS=y` or `CONFIG_DEBUG=y` fail to build for non-secure platforms, as reported in https://github.com/zephyrproject-rtos/zephyr/issues/73708.

**To Reproduce**
Either follow steps in https://github.com/zephyrproject-rtos/zephyr/issues/73708

or 

simply try build Hello World sample with the following command:
`west build -b nrf9160dk/nrf9160/ns -- -DCONFIG_DEBUG_OPTIMIZATIONS=y`

**Expected behavior**
TF-M build should not break if project enabled debug optimizations.

**Impact**
Showstopper

**Logs and console output**
```bash
[373/383] Linking C executable bin/tfm_s.axf
Memory region         Used Size  Region Size  %age Used
           FLASH:      194908 B       254 KB     74.94%
             RAM:       50892 B        88 KB     56.48%
[379/383] Linking C executable bin/bl2.axf
FAILED: bin/bl2.axf 
: && /home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -mcpu=cortex-m33+nofp -O2 -g -DNDEBUG -mcpu=cortex-m33+nofp --entry=Reset_Handler -specs=nano.specs -specs=nosys.specs -Wl,-check-sections -Wl,-fatal-warnings -Wl,--gc-sections -Wl,--no-wchar-size-warning -Wl,--print-memory-usage -Wl,-Map=/home/robert/repos/zephyrproject/zephyr/samples/hello_world/build/tfm/bin/bl2.map -mfloat-abi=soft -T /home/robert/repos/zephyrproject/zephyr/samples/hello_world/build/tfm/platform/target/nrf91/nrf_common/CMakeFiles/bl2_scatter.dir/./__/__/__/__/common/gcc/tfm_common_bl2.ld bl2/CMakeFiles/bl2.dir/src/flash_map.o bl2/CMakeFiles/bl2.dir/src/security_cnt.o bl2/CMakeFiles/bl2.dir/src/default_flash_map.o bl2/CMakeFiles/bl2.dir/src/provisioning.o bl2/CMakeFiles/bl2.dir/src/thin_psa_crypto_core.o bl2/CMakeFiles/bl2.dir/__/platform/ext/common/syscalls_stub.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/bl2_main.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/keys.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/flash_map_extended.o bl2/CMakeFiles/bl2.dir/ext/mcuboot/flash_map_legacy.o bl2/CMakeFiles/bl2.dir/__/platform/ext/target/nordic_nrf/common/core/startup.o bl2/CMakeFiles/bl2.dir/__/platform/ext/target/nordic_nrf/common/core/startup_nrf91.o -o bin/bl2.axf  bl2/ext/mcuboot/bootutil/libbootutil.a  platform/libplatform_bl2.a  bl2/mbedcrypto/library/libmbedcrypto.a  bl2/mbedcrypto/3rdparty/p256-m/libbl2_p256m.a  bl2/mbedcrypto/library/libmbedcrypto.a  bl2/mbedcrypto/3rdparty/p256-m/libbl2_p256m.a  bl2/mbedcrypto/3rdparty/everest/libbl2_everest.a && :
/home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_nvmc.o): in function `nrfx_nvmc_page_erase':
/home/robert/repos/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:263: undefined reference to `spm_log_msgval'
/home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_nvmc.o): in function `nrfx_nvmc_words_write':
/home/robert/repos/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:479: undefined reference to `spm_log_msgval'
/home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/robert/repos/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:480: undefined reference to `spm_log_msgval'
/home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/robert/repos/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:481: undefined reference to `spm_log_msgval'
/home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_uarte.o): in function `nrf_gpio_pin_port_decode':
/home/robert/repos/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_gpio.h:873: undefined reference to `spm_log_msgval'
/home/robert/zephyr-sdk/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: platform/libplatform_bl2.a(nrfx_uarte.o):/home/robert/repos/zephyrproject/modules/hal/nordic/nrfx/hal/nrf_gpio.h:882: more undefined references to `spm_log_msgval' follow
```

**Environment (please complete the following information):**
 - Linux
 - Toolchain: Zephyr SDK 0.16.5-1
 - 0d56a3c3f4c333b08b0fecde02a9c032ee6c2d2f

aescolar, 2024-06-11:
Is this issue nordic specific?

ithinuel, 2024-06-11:
I just ran `west build -p -b mps2/an521/cpu0/ns samples/hello_world -- -DCONFIG_DEBUG_OPTIMIZATIONS=y` and it built fine.

rlubos, 2024-06-12:
> Is this issue nordic specific?

I'm far from being TFM expert, but in my opinion it's not. It does manifest on Nordic devices, yes, but that's because they use asserts in the driver code:
https://github.com/zephyrproject-rtos/hal_nordic/blob/ab5cb2e2faeb1edfad7a25286dcb513929ae55da/nrfx/drivers/src/nrfx_nvmc.c#L263
which evaluates to `SPM_ASSERT`:
https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/bdc4df1734b870de43237d56eb1b2a7af016ee95/platform/ext/target/nordic_nrf/common/core/common/nrfx_glue.h#L70
Which leads to errors because `spm_log_msgval()` is not compiled in.

> Could you be missing a west update ?

I don't, the issue was originally reported by our CI.

Unassingning myself, if I knew the right solution, I wouldn't be bothering opening an issue.

juhaylinen, 2024-06-12:
One solution to fix this is to add the missing file to cmake script but I'm not sure if that's the best way to do it.

ithinuel, 2024-06-12:
`spm_log.c` is conditionally added to the `tfm_spm` cmake target in tf-m.

The failure occurs when linking `bl2.axf` so either there's a missing dependency injection, or the condition adding `spm_log.c` is not met.

aescolar, 2024-06-12:
CC @Vge0rge

adeaarm, 2024-06-12:
> One solution to fix this is to add the missing file to cmake script but I'm not sure if that's the best way to do it.
> 
> I'll create a PR to TFM project

From a quick look it seems that the flash driver directly refers SPM_ASSERT(). While this is valid for a runtime FW build, in a BL2 build the concept of SPM_ASSERT() does not exist, and it should just behave as a standard assert(). It should probably needs to be fixed internally. Synced up offline with @frkv about this.

frkv, 2024-06-12:
A comment with regards to this. 

We need to make a viable pattern to ensure that the signal to use (or not use) TF-M constructs is accurately guarded with compile definitions that relevant images are built with. There are some shared concepts with bl2, but bl2 image is "its own thing" and we don't have a define that only exists for this build

In TF-M there is a concept for the build targets `bl2 `and `platform_s` and it has been discussed (offline, with @adeaarm) that the best approach may be to mirror some kind of pattern (in compile definitions) so that we can have a accurate signal that "I am in image X"

An accurate signal for being inside a TF-M image build would be to send `-DPLATFORM_S(=1)` as part of the build (for the platform_s target).

A similar construct could be made for BL2, but we are wondering if the target name should be `platform_bl2` with the compile definition `-DPLATFORM_BL2(=1)` or if the bl2 should just continue with its target name and we just add `-DPLATFORM_BL2(=1)` to the build

_I am hesitant to add a minor fix for this here-and-now if it is going to be reworked in a few days time..._

aescolar, 2024-06-24:
@frkv @adeaarm @ithinuel is there any update on this issue?

frkv, 2024-06-24:
Will be discussing this with @adeaarm, today. Will give feedback as soon as possible!

adeaarm, 2024-06-24:
> Will be discussing this with @adeaarm, today. Will give feedback as soon as possible!

Hi, we are working on a refactor in TF-M for this so that private macros such as `SPM_ASSERT()` won't be exported publicly. The issue here is that there is platform code which is using a macro which is meant to be private for the secure binary, and unfortunately such behaviour is not constrained by the TF-M build system. We still aim to provide a fix upstream in the coming weeks.

aescolar, 2024-06-24:
> The issue here is that there is platform code which is using a macro which is meant to be private for the secure binary, and unfortunately such behaviour is not constrained by the TF-M build system. 

I see. Do I understand correctly then that the flash driver in the nrfx hal should also be change to not use this macro anymore?
CC @de-nordic @nika-nordic @anangl

adeaarm, 2024-06-24:
> I see. Do I understand correctly then that the flash driver in the nrfx hal should also be change to not use this macro anymore? CC @de-nordic @nika-nordic @anangl

Yes, Ideally we're working on a fix such that platform code such as the flash driver will just have to use the standard `assert()` macros, and then at build time it will be properly linked to the SPM_ASSERT() only for the runtime FW (while binaries in BL1/2 stages won't).

aescolar, 2024-06-24:
Thanks @adeaarm . So there is nothing we can do in the short term that we can get in for 3.7 release?

adeaarm, 2024-06-24:
> Thanks @adeaarm . So there is nothing we can do in the short term that we can get in for 3.7 release?

A workaround could be to have `_weak` defines of the function below in nRF platform code that just redirects to `printf()`, but i guess this is more of a question for @frkv if there is room to wait for the upstream fix.

> ```
> spm_log_msgval
> ```

frkv, 2024-06-24:
> A workaround could be to have `_weak` defines of the function below in nRF platform code that just redirects to `printf()`, but i guess this is more of a question for @frkv if there is room to wait for the upstream fix.

I am OK with a weak redirect to something that does printf, as long as assert is still handled. I think that would be the "quick fix" that should be doable in short term.

I was hoping we would be able to fix this properly, but waiting a few weeks might be a bit difficult given Zephyr LTS release schedule...

nicola-mazzucato-arm, 2024-06-25:
Hi All, I just posted for review in TF-M a small set that may help with the issue above:
https://review.trustedfirmware.org/q/topic:%22spm_assert%22

Thanks

aescolar, 2024-06-26:
Reopening and this is not in the manifest yet

PR #74953 (2024-06-27): manifest: trusted-firmware-m: fix debug build
Pulls in assert usage fixes.

Fixes #73862.
commit 5f41ec8e (2024-06-27): manifest: trusted-firmware-m: fix debug build

Pulls in assert usage fixes.

Fixes #73862.
closed 2024-06-27 by nashif