zephyrproject-rtos/zephyr #73869: RFC: Add Fallback RTIO implementation to Bus Drivers

ubieda, 2024-06-06:
## Introduction

With the addition of RTIO in-tree, there's a new set of use-cases that can be accomplished by integrating it into the different system-components. One of these is the Asynchronous Sensor API. While this depends on having RTIO implementations on the underlying bus-drivers to be efficient, a default-fallback implementation for the main bus drivers (e.g: I2C, SPI) would **ease the transition of components that depend on RTIO Bus Drivers**, while efforts towards adding drivers support progresses.

### Problem description

Components attempting to migrate towards an RTIO-centric implementation need a better transitional approach than waiting for ubiquitous RTIO Bus Drivers support.

### Proposed change

- Add Default-Fallback I2C RTIO implementation, applicable to drivers lacking RTIO support (Support: #67328).
- Add Default-Fallback SPI RTIO implementation, applicable to drivers lacking RTIO support (Support: #67327).

### Proposed change (Detailed)

- I2C RTIO fallback implementation shall allow users to correctly perform I2C transfers using the RTIO API on non-supported Bus Drivers. See example code below:
https://github.com/zephyrproject-rtos/zephyr/blob/0d56a3c3f4c333b08b0fecde02a9c032ee6c2d2f/tests/drivers/i2c/i2c_ram/src/test_i2c_ram.c#L181-L220

- SPI RTIO fallback implementation shall allow users to correctly perform SPI transfers using the RTIO API on non-supported Bus Drivers. See example code below:
https://github.com/zephyrproject-rtos/zephyr/blob/0d56a3c3f4c333b08b0fecde02a9c032ee6c2d2f/tests/drivers/spi/spi_loopback/src/spi_rtio.c#L131-L166

- These default implementations will not apply to already RTIO-supported Bus Drivers. 

### Concerns and Unresolved Questions

- This should be conditionally included only when necessary, as it would imply increase code-size.
- It should also not be evaluated by its performance, since it would not be optimal for what RTIO is capable of.
- Initial approach would be to integrate work-queue's or threads to decouple the interactions, but a further evaluation is required to refine details.

## Alternatives

Do not develop a fallback implementation and block the development of other components that depend on this until all bus drivers support RTIO.

ubieda, 2024-06-06:
CC @teburd @yperess

teburd, 2024-06-06:
This makes good sense to me, and would let us change sensor drivers without waiting on RTIO implementations of the bus drivers which seems to be a growing issue. I think *this is the right place* to shim for asynchronicity rather than shimming at the sensor layer with the exception of shimming for the fetch + get APIs.

ubieda, 2024-06-06:
> This makes good sense to me, and would let us change sensor drivers without waiting on RTIO implementations of the bus drivers which seems to be a growing issue. I think _this is the right place_ to shim for asynchronicity rather than shimming at the sensor layer with the exception of shimming for the fetch + get APIs.
> 
> Maybe the most obvious option is having a [i2c|spi]_rtio device driver that works as normal but calls to an underlying blocking implementation on a thread that changes priority to match the I/O request priority.

I agree with you. However, I don't think we'll be able to spare the inclusion of a temporary shim in the sensor layer, given the LTS release coming up soon, and the existing drivers with synchronous implementation (#73676).

If we weren't so close to the LTS release I'd vote for going directly in this direction, instead.

swift-tk, 2024-06-19:
Very cool!!

ubieda, 2024-08-12:
Update on this:
- @yperess has a PR up for the I2C Fallback: #75293.
- I'm working on the SPI Fallback implementation (No PR filed yet).

closed 2024-09-05 by ubieda