zephyrproject-rtos/zephyr #73878: Zephyr 3.7 module update overview

dleach02, 2024-06-06:
Umbrella issue for modules which may require an update before Zephyr 3.7.

- [ ] Security modules tracked here: #73703
- [ ] https://github.com/zephyrproject-rtos/hostap (@krish2718, @jukkar)
- [ ] https://github.com/zephyrproject-rtos/zcbor (@de-nordic)
- [ ] https://github.com/zephyrproject-rtos/uoscore-uedhoc (@rlubos, @StefanHri)
- [ ] https://github.com/zephyrproject-rtos/thrift (@cfriedt)
- [ ] https://github.com/zephyrproject-rtos/tflite-micro (@laurenmurphyx64)
- [ ] https://github.com/zephyrproject-rtos/sof (@kv2019i)
- [ ] https://github.com/zephyrproject-rtos/picolibc (@keith-packard, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/percepio (@eriktamlin, @aronlander-pe)
- [ ] https://github.com/zephyrproject-rtos/openthread (@rlubos)
- [ ] https://github.com/zephyrproject-rtos/open-amp (@carlocaione)
- [ ] https://github.com/zephyrproject-rtos/nanopb (@pdgendt)
- [ ] https://github.com/zephyrproject-rtos/mcuboot (@d3zd3z, @de-nordic, @nordicjm)
- [ ] https://github.com/zephyrproject-rtos/lz4 (@Navin-Sankar)
- [ ] https://github.com/zephyrproject-rtos/lvgl (@faxe1008, @brgl, @pdgendt)
- [ ] https://github.com/zephyrproject-rtos/loramac-node (@Mani-Sadhasivam)
- [ ] https://github.com/zephyrproject-rtos/liblc3 (@Casper-Bonde-Bose, @MariuszSkamra)
- [ ] https://github.com/zephyrproject-rtos/fatfs (@de-nordic)
- [ ] https://github.com/zephyrproject-rtos/edtt (@aescolar, @wopu-ot, @thoh-ot)
- [ ] https://github.com/zephyrproject-rtos/cmsis-nn (@XenuIsWatching, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/cmsis-dsp (@XenuIsWatching, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/cmsis (@stephanosio, @microbuilder, @povergoing)
- [ ] https://github.com/zephyrproject-rtos/chre (@yperess)
- [ ] https://github.com/zephyrproject-rtos/canopennode (@henrikbrixandersen)
- [ ] https://github.com/zephyrproject-rtos/babblesim-manifest (@aescolar)
- [ ] https://github.com/zephyrproject-rtos/acpica (@najumon1980)

Folks that are tagged please comment and confirm the status of your modules with respect to readiness for the 3.7 release.

- [Zephyr 3.7 Release Checklist #69408](https://github.com/zephyrproject-rtos/zephyr/issues/69408)
- [Zephyr 3.7 security related modules update overview #73703](https://github.com/zephyrproject-rtos/zephyr/issues/73703)

JordanYates, 2024-06-11:
`loramac-node` is in maintenance mode and has no updates.
`cmsis-nn` update here: #73995

faxe1008, 2024-06-11:
`lvgl` module is at the current viable released version 8.4.0 [https://github.com/lvgl/lvgl/releases/tag/v8.4.0](https://github.com/lvgl/lvgl/releases/tag/v8.4.0). Version 9 breaks at least the monochrome display support so it's not viable for now. PR for that is pending but not realistic for 3.7.

krish2718, 2024-06-11:
No planned updated for `hostap` for `3.7.0`.

carlocaione, 2024-06-12:
For `open-amp` and `libmetal` https://github.com/zephyrproject-rtos/zephyr/pull/74186

keith-packard, 2024-06-12:
No known issues for picolibc for 3.7.0 -- current release is the same as zephyr uses.

JordanYates, 2024-06-13:
`tflite-micro` upgrade is part of #73995, need either perms for the repo or someone to work out the desired branch structure.

henrikbrixandersen, 2024-06-14:
There's a request (https://github.com/zephyrproject-rtos/zephyr/issues/70285) to update CANopenNode to v4.0, but it will not happen in time for v3.7.0.

nordicjm, 2024-06-14:
No update planned for MCUboot in the zephyr 3.7 release

MariuszSkamra, 2024-06-20:
As discussed offline with @sjanc, no plans to update liblc3 in the Zephyr 3.7.0

cfriedt, 2024-06-20:
Thrift docs gave a minor update in https://github.com/zephyrproject-rtos/zephyr/pull/74557

There is a version upgrade, but I would rather save that until after 3.7

XenuIsWatching, 2024-06-22:
cmsis-dsp hasn't had an update come out. I'd rather wait for the next 'taged' release

dleach02, 2024-06-25:
> Thrift docs have a minor update in #74557
> 
> There is a version upgrade, but I would rather save that until after 3.7

> Thrift docs have a minor update in #74557
> 
> There is a version upgrade, but I would rather save that until after 3.7

@cfriedt remember that v3.7 is LTS3. Do you want this version to stick for the next 3+ years?

dleach02, 2024-06-25:
> There's a request (#70285) to update CANopenNode to v4.0, but it will not happen in time for v3.7.0.

> There's a request (#70285) to update CANopenNode to v4.0, but it will not happen in time for v3.7.0.

@henrikbrixandersen so that means the LTS will stick with the older version for the next 3 years

henrikbrixandersen, 2024-06-25:
> @henrikbrixandersen so that means the LTS will stick with the older version for the next 3 years

Correct.

carlocaione, 2024-07-05:
https://github.com/zephyrproject-rtos/zephyr/pull/74186 this is waiting to be merged

aescolar, 2024-07-15:
Closing as 3.7.0 rc3 has been tagged and no more module updates are expected.

closed 2024-07-15 by aescolar