zephyrproject-rtos/zephyr #73881: CMD53 READ Fail on MIMXRT1170/595-EVK Platform.

sidd198, 2024-06-06:
Issue: 
Noticed cmd53_read failing while reading the chip_id and I am seeing the below crash.

Platform:
MIMXRT1170-EVK
MIMXRT595-EVK

I am trying to bring up the wi-fi on NXP MIMXRT1170-EVK with 1YN module.

I have modified the dts file to pick the corresponding driver.

On executing the Wi-Fi CE I am seeing the below crash logs.

[00:00:01.660,000] <err> os: ***** MPU FAULT *****
[00:00:01.660,000] <err> os:   Instruction Access Violation
[00:00:01.660,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x18000000  r2/a3:  0x80012c34
[00:00:01.660,000] <err> os: r3/a4:  0x18000000 r12/ip:  0x0000067c r14/lr:  0x80001000
[00:00:01.660,000] <err> os:  xpsr:  0x60000000
[00:00:01.660,000] <err> os: Faulting instruction address (r15/pc): 0x80001000
[00:00:01.660,000] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
[00:00:01.660,000] <err> os: Current thread: 0x800028e0 (main)
[00:00:01.699,000] <err> os: Halting system

I have attached the changes made for 1170 platform and the complete terminal log.
[1170_changes.txt](https://github.com/user-attachments/files/15686520/1170_changes.txt)
[1170_error_log.txt](https://github.com/user-attachments/files/15686522/1170_error_log.txt)

Please me in solving this issue. 
Please provide the below info: 
How to configure wifi-reg-on and wifi-host-wake on the dts file. 
wifi-reg-on and wifi-host-wake GPIO pin details for the MIMXRT-595 and MINMX-1170 platform.

danieldegrasse, 2024-06-06:
Hi @sidd198, if you run your application with `CONFIG_SD_LOG_LEVEL_DBG=y` and `CONFIG_SDHC_LOG_LEVEL_DBG=y` do you see any additional information? If you have any form of backtrace from a debugger leading up to this crash, that could be helpful

sidd198, 2024-06-07:
@danieldegrasse I add the CONFIG_SD_LOG_LEVEL_DBG=y and CONFIG_SDHC_LOG_LEVEL_DBG=y flags on WI-FI CE prj.conf and i don't see any additional information.

danieldegrasse, 2024-06-07:
One place I would start is disabling the `sdmmc` node under `usdhc1`. If you have the disk driver subsystem enabled (CONFIG_DISK_ACCESS=y), that could be interfering. Do you have `CONFIG_IMX_USDHC_DMA_SUPPORT` enabled? One thing to check would be that the AIROC WIFI driver is setting `data->blocks` and `data->block_size` correctly for the size of the buffer provided in `data->data` (it should be equal in size of `data->blocks * data->block_size`)

sidd198, 2024-06-08:
I have disabled the sdmmc below the code for the same:

zephyrproject_3_6\zephyr\boards\arm\mimxrt1170_evk\mimxrt1170_evk_cm7.dts

&usdhc1 {
		status = "okay";
		detect-dat3;
		pwr-gpios = <&gpio10 2 GPIO_ACTIVE_LOW>;
		sdmmc {
			compatible = "zephyr,sdmmc-disk";
			status = "disabled";
		};
		airoc-wifi {
			status = "okay"; 
			compatible = "infineon,airoc-wifi"; 
			/*wifi-reg-on-gpios    = <&gpio9 30 GPIO_ACTIVE_HIGH>; */
			/*#wifi-host-wake-gpios = <&gpio\_prt0 4 GPIO\_ACTIVE\_HIGH>; */
		}; 
};

I checked the CONFIG_DISK_ACCESS is not enabled and CONFIG_IMX_USDHC_DMA_SUPPORT is enabled.

\zephyrproject_3_6\zephyr\build\zephyr\.config file

CONFIG_DISK_ACCESS is not set

CONFIG_IMX_USDHC_DMA_SUPPORT=y

Also while building I am seeing the below warning massages is there anything I am missing ??

warning: MCUX_SDIF (defined at drivers/sdhc/Kconfig.mcux_sdif:4) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: DT_HAS_NXP_LPC_SDIF_ENABLED (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MCUX_SDIF and/or look up MCUX_SDIF in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: OPENTHREAD_THREAD_STACK_SIZE (defined at subsys/net/l2/openthread/Kconfig:132) was assigned
the value '9000' but got the value ''. Check these unsatisfied dependencies: NET_L2_OPENTHREAD (=n).
See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_OPENTHREAD_THREAD_STACK_SIZE and/or
look up OPENTHREAD_THREAD_STACK_SIZE in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

danieldegrasse, 2024-06-10:
> warning: MCUX_SDIF (defined at drivers/sdhc/Kconfig.mcux_sdif:4) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: DT_HAS_NXP_LPC_SDIF_ENABLED (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MCUX_SDIF and/or look up MCUX_SDIF in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

You are setting `CONFIG_MCUX_SDIF=y` in your `prj.conf`. The value is being assigned `n` because the RT1170 does not have an SDIF peripheral, so the driver is not enabled. You should not need to enable SDHC drivers from your `prj.conf`, so `CONFIG_IMX_USDHC=y` can be dropped too.

> One thing to check would be that the AIROC WIFI driver is setting data->blocks and data->block_size correctly for the size of the buffer provided in data->data (it should be equal in size of data->blocks * data->block_size)

Can you please check this? You can also disable `CONFIG_IMX_USDHC_DMA_SUPPORT` to see if that changes the error message- if the DMA is writing beyond the end of a buffer, the crash will be harder to debug than if the core itself tries to write to the same memory. However, I have seen issues with high datarates on the iMX USDHC peripheral when DMA is disabled, so DDR50 mode might not work well without DMA

sidd198, 2024-06-11:
I did removed the CONFIG_MCUX_SDIF=y & CONFIG_IMX_USDHC=y flags from my prj.conf.
I have configured the sdio mode to SDR25 and voltage: 3.3v(I am using 1YN Murata Module)
Disabled CONFIG_IMX_USDHC_DMA_SUPPORT while building the project with below command
west build -p always -b mimxrt1170_evk_cm7 samples\net\wifi -DCONFIG_WIFI_INIT_PRIORITY=86 -DCONFIG_IMX_USDHC_DMA_SUPPORT=n
I still see the same crash logs.

One thing to check would be that the AIROC WIFI driver is setting data->blocks and data->block_size correctly for the size of the buffer provided in data->data (it should be equal in size of data->blocks * data->block_size)

zephyrproject_3_6\zephyr\subsys\sdio.c file 
   Function :sdio_io_rw_extended_helper is handling the block size based on the data to send. 
WHD is setting block size to SDIO_64B_BLOCK for sdio_func1 & sdio_func2

Let me know if there any other place where i can print and check ??

I have added the changes again review and let me know if i am missing anything 
[complete_1170_changes.txt](https://github.com/user-attachments/files/15781575/complete_1170_changes.txt)

Also can please let me know the GPIO pin for WLAN_REG_ON for 1170 platform  and how to configure the same.

danieldegrasse, 2024-06-11:
> Function :sdio_io_rw_extended_helper is handling the block size based on the data to send.
WHD is setting block size to SDIO_64B_BLOCK for sdio_func1 & sdio_func2

> Let me know if there any other place where i can print and check ??

Can you narrow down the call stack prior to the fault? Any type of backtrace would help greatly to debug what is failing here. If that is not possible, at least some additional debug logging on what part of the WHD driver init sequence is sending CMD53 and causing the fault would be useful. Perhaps adding debug logs in the functions that call `whd_bus_sdio_cmd53`?

> Also can please let me know the GPIO pin for WLAN_REG_ON for 1170 platform and how to configure the same.

This looks like a pin on the 1YN module? I'm unclear what the question is here- the GPIO pin would be based on how the module is connected to the SOC.

@npal-cy do you have any insight here as to what could cause the fault?

sidd198, 2024-06-11:
i am connecting 1YN module through M2 connector and i what to know from SOC(1170) which is the GPIO pin ???
Example: wifi-reg-on-gpios in DTS 
         wifi-reg-on-gpios    = <GPIO PIN CONFIGURATION PIN>; 
  
what should be the  value of GPIO PIN CONFIGURATION PIN if i am connecting 1YN module to M2 connector ??

danieldegrasse, 2024-06-11:
I am not familiar with the pinout of the 1YN module- the EVKB schematic mentions compatibility with the 1DX M.2 module from embedded artists, but does not claim compatibility with the 1YN module. Perhaps you want pin 56?:
[image]
(https://developer.embeddedartists.com/docs-m2/integration/pinning/)
That maps to WL_RST, which seems to need a resistor populated in order to function: 
[image]

Regardless, this pin appears to be pulled up on the schematic, so I *believe* the module would be powered on

sidd198, 2024-06-12:
Below are my DTS changes for 1170 and 595 Platforms.

For 1170 :
&usdhc1 {
	status = "okay";
	detect-dat3;
	pwr-gpios = <&gpio10 2 GPIO_ACTIVE_LOW>;
	sdmmc {
		compatible = "zephyr,sdmmc-disk";
		status = "disabled";
	};
	airoc-wifi {
		status = "okay"; 
		compatible = "infineon,airoc-wifi"; 
		/*wifi-reg-on-gpios    = <NEED WL_REG_ON GPIO PIN INFO>; */           --GPIO For WL_REG_ON not configured.
		/*wifi-host-wake-gpios = <NEED WL_HOSTWAKE GPIO PIN INFO>; */
	};
};

For 595:

&usdhc1 {
       status = "okay";
       pwr-gpios = <&gpio3 20 GPIO_ACTIVE_HIGH>;
       cd-gpios = <&gpio3 19 GPIO_ACTIVE_LOW>;
       pinctrl-0 = <&pinmux_usdhc1>;
       pinctrl-names = "default";
       airoc-wifi {
               status = "okay";
               compatible = "infineon,airoc-wifi";
               wifi-reg-on-gpios = <&gpio0 31 GPIO_ACTIVE_HIGH>;          ---GPIO For Wl_REG_ON Configured.
       }; 
 };

I got the WL_REG_ON GPIO PIN details for 595 Platform( PORT 0 PIN 31) but for 1170 I could not find from the 1170 Schematic?? Hence, I have not configured the WL_REG_PIN configuration for 1170 platform ??

From the 1170 platform schematic I could see WL_RST is connected to GPIO¬_AD_30 ?? what is GPIO PORT and GPIO PIN  info for the same ??

 

[image]
Can you please let me know the GPIO PORT details for 1170 platform not from the 1YN module for WL_REG_ON ??

If I don’t configure the GPIO PIN will there be any SDIO re-enumeration issue or any other issues??

sidd198, 2024-06-12:
Also do you have any other application to test the SDIO communication ???

danieldegrasse, 2024-06-12:
> From the 1170 platform schematic I could see WL_RST is connected to GPIO¬_AD_30 ?? what is GPIO PORT and GPIO PIN info for the same ??

Take a look at this file, which maps the GPIO names to port numbers: https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/nxp/nxp_rt11xx_cm7.dtsi#L426. You can see that the pinmux option for GPIO_AD_30 is called gpio_mux3_io29, so GPIO port 3, pin 29 routes to this pin.

Note that R404 is marked DNP on the schematic. Unless you have explicitly populated this resistor, WL_RST will not be connected. Either way, it seems it would be connected to GPIO_AD_31 if the resistor was present?

> If I don’t configure the GPIO PIN will there be any SDIO re-enumeration issue or any other issues??

I do not know- I unfortunately do not have a 1YN module to test with, and am not an expert on the initialization of the module.

> Also do you have any other application to test the SDIO communication ???

`tests/subsys/sd/sdio` is a good place to start to verify basic SDIO commands work. Beyond that, work is ongoing to support the 1DX module within Zephyr- I have tested the iMXRT1170 using internal enablement for this module in the past, and it has functioned correctly.

> Can you narrow down the call stack prior to the fault? Any type of backtrace would help greatly to debug what is failing here. If that is not possible, at least some additional debug logging on what [...]

I really think something like this is going to be necessary to narrow down what is wrong here- unfortunately I do not have the right hardware to test the setup you are using, so with the information provided thus far, I really can't help much more. If you can narrow down the location of the last `sdhc_transfer` call made before the crash occurs, we can likely go from there.

sidd198, 2024-06-12:
I have 1DX module with me, can you please share me the changes(git diff) you have done to make it work on 1170 platform.
Also do let me know if you have made any hw re-work or jumper setting  on 1170.

npal-cy, 2024-06-12:
Hi @sidd198 
from [1170_error_log.txt](https://github.com/user-attachments/files/15686522/1170_error_log.txt):
```
[00:00:01.660,000] <err> os: ***** MPU FAULT *****
[00:00:01.660,000] <err> os:   Instruction Access Violation
[00:00:01.660,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x18000000  r2/a3:  0x80012c34
[00:00:01.660,000] <err> os: r3/a4:  0x18000000 r12/ip:  0x0000067c r14/lr:  0x80001000
[00:00:01.660,000] <err> os:  xpsr:  0x60000000
[00:00:01.660,000] <err> os: Faulting instruction address (r15/pc): 0x80001000
[00:00:01.660,000] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
[00:00:01.660,000] <err> os: Current thread: 0x800028e0 (main)
[00:00:01.699,000] <err> os: Halting system
```

MPU FAULT can cause by stack overflows, but strange that we have `ZEPHYR FATAL ERROR 20`. could you share zephyr\build\zephyr\.config ?

npal-cy, 2024-06-13:
> I believe your asking about the zephyrproject_3_6\zephyr\build\zephyr.config right ??
> [config.txt](https://github.com/user-attachments/files/15810513/config.txt)

i see that DCACHE is enabled, could you try to disable DCACHE  (`west build ..... -DCONFIG_DCACHE=n`), and see if something changes.

sidd198, 2024-06-20:
Hello @danieldegrasse & @DerekSnell 

By Disabling -DCONFIG_DCACHE=n
SDIO communication is working chip_id read properly, Firmware & NVRAM Downloaded succesfull and terminal got stuck.
Below are logs for the same.

[00:00:00.503,000] <err> infineon_airoc_wifi:  airoc_init
[00:00:00.503,000] <err> infineon_airoc: airoc_wifi_init_primary
[00:00:00.503,000] <err> infineon_airoc: call airoc_wifi_power_on
[00:00:00.503,000] <err> infineon_airoc: airoc_wifi_power_on
[00:00:00.764,000] <err> infineon_airoc: call device_is_ready
[00:00:00.764,000] <err> infineon_airoc: call sd_init
[00:00:00.764,000] <err> sd:  Host controller support 3.3V max
[00:00:00.765,000] <err> usdhc: imx_usdhc_set_io SDHC I/O: bus widthSIDDU :: whd_bus_sdio_init whd_bus_read_register_value BUS width
whd_bus_sdio_init whd_bus_write_register_value BUS width
whd_bus_sdio_init whd_bus_write_register_value SDIOD_CCCR_BLKSIZE_0
whd_bus_sdio_init whd_bus_write_register_value SDIOD_CCCR_F1BLKSIZE_0
whd_bus_sdio_init whd_bus_write_register_value SDIOD_CCCR_F2BLKSIZE_0
whd_bus_sdio_init whd_bus_write_register_value SDIOD_CCCR_F2BLKSIZE_1
whd_bus_sdio_init whd_bus_write_register_value Enable/Disable Client interrupts
whd_bus_sdio_init whd_bus_read_register_value SDIOD_CCCR_IORDY
whd_bus_sdio_init whd_bus_write_register_value Set the ALP
whd_bus_sdio_init whd_bus_read_register_value SDIO_CHIP_CLOCK_CSR
whd_bus_sdio_init whd_bus_write_register_value Clear request for ALP
whd_bus_sdio_init whd_bus_write_register_value Disable the extra SDIO pull-ups
whd_bus_sdio_init whd_bus_write_register_value Enable F1 and F2
whd_bus_sdio_init whd_bus_write_register_value Enable F2 interrupt only
whd_bus_sdio_init whd_bus_read_register_value Enable F2 interrupt only:SDIOD_CCCR_IORDY
whd_bus_sdio_init chip_id 43439
whd_bus_sdio_init whd_bus_sdio_download_firmware
1, clock 0Hz, card power OFF, voltage 3.3V

To Debug further 
I set-up the live debug using MCUXpresso IDE.
 while debuggung live I am seeing that First CMD5 itself failing
call trace :
sdio_card_init()  --> sdio_send_ocr ()   --> sdhc_request()

Error log:

[00:00:01.771,000] <err> usdhc: imx_usdhc_transfer TRANSFER_CMD_FAILED
[00:00:01.771,000] <err> sd: SD CMD8 failed with error -5

Can you please help us to solve the above issues.

danieldegrasse, 2024-07-11:
> By Disabling -DCONFIG_DCACHE=n
SDIO communication is working chip_id read properly, Firmware & NVRAM Downloaded succesfull and terminal got stuck.
Below are logs for the same.

I'm not sure I follow- if DCACHE is disabled, than the firmware downloads successfully and the Wireless card works as expected?

siddesh198, 2024-08-07:
Yes by disabling -DCONFIG_DCACHE=n  Firmware is download..
but when we trigger scan we are seeing IOCTL time out, even though SCAN command reached the wi-fi chip...

We are not receiving the imx_usdhc_isr..
can you please help us what could be the issue...

siddesh198, 2024-08-08:
Just for the confirmation does RT1170 require any Hardware re-work ??
what are the change needs for the out of band configuration ?? any Hardware re-work ??

danieldegrasse, 2024-08-26:
> Just for the confirmation does RT1170 require any Hardware re-work ?? what are the change needs for the out of band configuration ?? any Hardware re-work ??

I'm unclear, how are you confirming the firmware is downloaded? The log you sent shows that the download firmware function runs, but does the card actually respond after that point? It seems like perhaps the firmware download is failing?

sidd198, 2024-09-26:
Hello @danieldegrasse and @DerekSnell 

We are able to load the firmware and scan & connect are working.

but some times I am intermittently notice usdhc_transfer Failure with reason TRANSFER_CMD_FAILED at the multiple stages(sdio_read_cccr, sdio_read_cis  and sdio_set_bus_speed). 

This is seen with 1060 not on 1170.
Is there is hardware re-work need for 1060 or do let me any pointer to debug further.

closed 2024-12-10 by github-actions[bot]