zephyrproject-rtos/zephyr #73885: numaker_m2l31ki is failing to build Kernel memory protection test

danieldegrasse, 2024-06-06:
**Describe the bug**
Since a30c5731aedaf8744a5ab, the `numaker_m2l31ki` board has failed to build the `tests/kernel/mem_protect/mem_protect` test. This seems to be related to disabling `CONFIG_USERSPACE` at the board level. Per the comments in #73209, it seems that the SOC only has 4 MPU regions and therefore cannot support userspace, even though the Cortex M architecture supports it.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `numaker_m2l31ki `
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? Yes, a30c5731aedaf8744a5ab

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b numaker_m2l31ki tests/kernel/mem_protect/mem_protect
2. See error

**Expected behavior**
Build should pass

**Impact**
This issue is blocking CI on #73802

**Logs and console output**
```
warning: MAX_THREAD_BYTES (defined at arch/Kconfig:369) was assigned the value '4' but got the value
''. Check these unsatisfied dependencies: USERSPACE (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MAX_THREAD_BYTES and/or look up
MAX_THREAD_BYTES in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

...
3:26: error: 'CONFIG_MAX_DOMAIN_PARTITIONS' undeclared here (not in a function)
   23 | #define NUM_RW_PARTS    (CONFIG_MAX_DOMAIN_PARTITIONS - PARTS_USED)
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/danieldegrasse/zephyr/zephyrproject/zephyr/tests/kernel/mem_protect/mem_protect/src/mem_domain.c:30:17: note: in expansion of macro 'NUM_RW_PARTS'
   30 |         rw_bufs[NUM_RW_PARTS][MEM_REGION_ALLOC];
      |                 ^~~~~~~~~~~~
/home/danieldegrasse/zephyr/zephyrproject/zephyr/tests/kernel/mem_protect/mem_protect/src/mem_domain.c:35:40: error: expected ')' before 'sizeof'
   35 | K_MEM_PARTITION_DEFINE(ro_part, ro_buf, sizeof(ro_buf),
      |                                        ^~~~~~~
      |                                        )
/home/danieldegrasse/zephyr/zephyrproject/zephyr/tests/kernel/mem_protect/mem_protect/src/mem_domain.c:38:45: error: expected ')' before 'sizeof'
   38 | K_MEM_PARTITION_DEFINE(overlap_part, ro_buf, sizeof(ro_buf),
      |                                             ^~~~~~~
      |                                             )
/home/danieldegrasse/zephyr/zephyrproject/zephyr/tests/kernel/mem_protect/mem_protect/src/mem_domain.c: In function 'test_mem_domain_setup':
/home/danieldegrasse/zephyr/zephyrproject/zephyr/tests/kernel/mem_protect/mem_protect/src/mem_domain.c:52:25: warning: implicit declaration of function 'arch_mem_domain_max_partitions_get'; did you mean 'k_mem_domain_add_partition'? [-Wimplicit-function-declaration]
   52 |         int max_parts = arch_mem_domain_max_partitions_get();
...
```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04)
 - Toolchain Zephyr SDK zephyr-sdk-0.16.5-1
 - Commit SHA or Version used e017006be4b (main)

aescolar, 2024-06-11:
@danieldegrasse Did merging #73886 fix this issue? (if so could you close this)

aescolar, 2024-06-11:
Closing assuming it was fixed by #https://github.com/zephyrproject-rtos/zephyr/pull/73886 please reopen if it wasn't.