zephyrproject-rtos/zephyr #73912: USB failure on lpcxpresso55s36

henrikbrixandersen, 2024-06-07:
**Describe the bug**
Building and running any USB samples on the `lpcxpresso55s36` board leads to compile-time warnings and run-time errors.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b lpcxpresso55s36/lpc55s36 samples/subsys/usb/cdc_acm`
2. `west flash`
3. See error

**Expected behavior**
No errors.

**Impact**
Unable to use the LPCXpresso55S36 with USB.

**Logs and console output**
Building the sample mentioned above leads to the following warnings (due to USB SRAM not being declared in devicetree)?
```
[ 97%] Linking C executable zephyr_pre0.elf
/opt/zephyr-sdk/0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd:linker_zephyr_pre0.cmd:280: warning: memory region `USB_SRAM' not declared
Logical command for additional byproducts on target: zephyr_pre0
[ 97%] Built target zephyr_pre0
[ 97%] Generating linker.cmd
[ 97%] Built target linker_zephyr_final_script_target
[ 97%] Generating isr_tables.c, isr_tables_vt.ld, isr_tables_swi.ld
[ 98%] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[100%] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[100%] Linking C executable zephyr.elf
/opt/zephyr-sdk/0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd:linker.cmd:280: warning: memory region `USB_SRAM' not declared
Memory region         Used Size  Region Size  %age Used
           FLASH:       47200 B       246 KB     18.74%
             RAM:       20040 B       112 KB     17.47%
        IDT_LIST:          0 GB        32 KB      0.00%
       *default*:       47200 B 18446744073709551615 B      0.00%
Generating files from /home/brix/Projects/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: lpcxpresso55s36
[100%] Built target zephyr_final
```

Running the resulting binary leads to the following error:
```
*** Booting Zephyr OS build v3.6.0-5511-gee5025c48128 ***
[00:00:00.000,000] <err> os: ***** MPU FAULT *****
[00:00:00.000,000] <err> os:   Data Access Violation
[00:00:00.000,000] <err> os:   MMFAR Address: 0x0
[00:00:00.000,000] <err> os: r0/a1:  0x200013bc  r1/a2:  0x40000000  r2/a3:  0x00000000
[00:00:00.000,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00005753
[00:00:00.000,000] <err> os:  xpsr:  0x69000000
[00:00:00.000,000] <err> os: Faulting instruction address (r15/pc): 0x00008d30
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x20000de0 (unknown)
[00:00:00.000,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 06f02eb3840358d444014a199665d292bd9e3f8c

**Additional context**
The same samples work fine on the `lpcxpresso55s16` board.

dleach02, 2024-06-07:
@hakehuang, are you seeing this in HW test range?

hakehuang, 2024-06-09:
@dleach02 the use device is not support in [lpcxpresso55s36.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/nxp/lpcxpresso55s36/lpcxpresso55s36.yaml)
and all usb device cases are filter our as below `depends_on`.
```
sample:
  name: CDC ACM USB
tests:
  sample.usb.cdc-acm:
    depends_on: usb_device
```
so I need enable it? or it is still not support yet?

henrikbrixandersen, 2024-06-10:
USB support was added by @mmahadevan108 in https://github.com/zephyrproject-rtos/zephyr/pull/58095 - but it's true that `usb_device` was never added to the list of supported features on the `lpcxpresso55s36` board. There's some discussion going on in that PR regarding the missing USB SRAM node, perhaps @mmahadevan108 has some input on the issue described in this PR.

PR #74908 (2024-06-26): soc: nxp: lpc: lpc55xxx: fix CONFIG_LPC55XXX_USB_RAM defaults
Setting a boolean Kconfig option default to "n" after having set it to "y" does not make the option disabled. Instead, avoid setting default to "y" for SoCs known not to have dedicated USB RAM.

Fixes: #73912
commit 771a362f (2024-06-26): soc: nxp: lpc: lpc55xxx: fix CONFIG_LPC55XXX_USB_RAM defaults

Setting a boolean Kconfig option default to "n" after having set it to "y"
does not make the option disabled. Instead, avoid setting default to "y"
for SoCs known not to have dedicated USB RAM.

Fixes: #73912
closed 2024-06-26 by nashif