zephyrproject-rtos/zephyr #73928: `samples/drivers/flash_shell` has build warning for devices that select `CONFIG_FLASH_HAS_NO_EXPLICIT_ERASE`

danieldegrasse, 2024-06-07:
**Describe the bug**
The flash shell sample builds with warnings for devices that select `CONFIG_FLASH_HAS_NO_EXPLICIT_ERASE`.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? Reproduced on numaker_m2l31ki
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? Yes, 7e658607fc77c51

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b numaker_m2l31ki samples/drivers/flash_shell/
2. See warning

**Expected behavior**
Build should complete without warnings

**Impact**
Blocking CI on #73886

**Logs and console output**
```
[116/159] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_shell.c.obj
/home/danieldegrasse/zephyr/zephyrproject/zephyr/drivers/flash/flash_shell.c: In function 'cmd_erase':
/home/danieldegrasse/zephyr/zephyrproject/zephyr/drivers/flash/flash_shell.c:79:18: warning: unused variable 'size' [-Wunused-variable]
   79 |         uint32_t size;
      |                  ^~~~
/home/danieldegrasse/zephyr/zephyrproject/zephyr/drivers/flash/flash_shell.c:77:18: warning: unused variable 'page_addr' [-Wunused-variable]
   77 |         uint32_t page_addr;
      |                  ^~~~~~~~~
/home/danieldegrasse/zephyr/zephyrproject/zephyr/drivers/flash/flash_shell.c:76:30: warning: unused variable 'flash_dev' [-Wunused-variable]
   76 |         const struct device *flash_dev;
```

de-nordic, 2024-06-07:
Already fixed with 0e7c25fee462a7249a4303471617e5416120d198

danieldegrasse, 2024-06-07:
> Already fixed with https://github.com/zephyrproject-rtos/zephyr/commit/0e7c25fee462a7249a4303471617e5416120d198, can you rebase your PR and check again? I have run the provided command on main and this does not reproduce.

Yup, sorry- should have checked closer to see if a hotfix already existed! I'll close this out

closed 2024-06-07 by danieldegrasse