zephyrproject-rtos/zephyr #73948: Doc: the link to Getting Started Guide is wrong

WeibingChen17, 2024-06-07:
**Describe the bug**

On the home page of Zephyr, "Technical Resources" -> "Developing with Zephyr" -> "Getting Started Guide" is pointing to https://docs.www.zephyrproject.org/latest/develop/getting_started/index.htm. The link is 404.

It should be https://docs.zephyrproject.org/latest/develop/getting_started/index.htm. 

**To Reproduce**
NA

**Expected behavior**
NA

**Impact**
Bad impression for developers

**Logs and console output**
NA

**Environment (please complete the following information):**
 - Web

**Additional context**
NA

Wirgiliusz, 2024-06-09:
I have found more broken links, all in the same manner ("www" in wrong place) - actually looks like every single link with "external address" icon that points to https://docs.zephyrproject.org/ is broken.

I have summarised all the broken links following hierarchy used on the main page (https://www.zephyrproject.org/):
- Technical Resources
  - Developing with Zephyr
    - Getting Started Guide (https://docs.www.zephyrproject.org/latest/develop/getting_started/index.html)
  - Communicating with Zephyr Community
    - Mailing Lists (https://lists.www.zephyrproject.org/g/main/subgroups)
  - Documentation (https://docs.www.zephyrproject.org/latest/)
    - Supported Boards (https://docs.www.zephyrproject.org/latest/boards/index.html)
    - Samples and Demos (https://docs.www.zephyrproject.org/latest/samples/index.html)
    - Contribution Guidelines (https://docs.www.zephyrproject.org/latest/contribute/index.html)
- Community
  - Supported Boards (450+) (https://docs.www.zephyrproject.org/latest/boards/index.html)
  - Connect with Zephyr
    - Mailing Lists (https://lists.www.zephyrproject.org/g/main/subgroups)

kartben, 2024-06-10:
Thanks for the report. Looks like there were some regressions introduced by a recent technical migration of the main Zephyr website. I've asked the webmaster team to look into it.

Azzamean, 2024-06-10:
Thanks. This should be resolved now.

closed 2024-06-11 by kartben

vastep, 2024-06-11:
One more - chat.www.zephyrproject.org

Azzamean, 2024-06-11:
What page is this on? I am not seeing it.

Azzamean, 2024-06-11:
Fixed