zephyrproject-rtos/zephyr #73954: tests: tests/subsys/jwt: lpcxpresso55s69/lpc55s69/cpu0/ns : build failure due to mbedtls upgrading

hakehuang, 2024-06-08:
**Describe the bug**
build fail for tests/subsys/jwt

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 lpcxpresso55s69/lpc55s69/cpu0/ns
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 

**To Reproduce**
Steps to reproduce the behavior:
1.  west buind -b lpcxpresso55s69/lpc55s69/cpu0/ns tests/subsys/jwt

**Expected behavior**
Build PASS

**Impact**
jwt

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk_internal.h:26:67: error: 'psa_to_pk_rsa_errors' undeclared (first use in this function); did you mean 'psa_to_ssl_errors'?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro 'PSA_TO_MBEDTLS_ERR_LIST'
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk.c:1202:16: note: in expansion of macro 'PSA_PK_RSA_TO_MBEDTLS_ERR'
 1202 |         return PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk_internal.h:26:67: note: each undeclared identifier is reported only once for each function it appears in
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro 'PSA_TO_MBEDTLS_ERR_LIST'
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk.c:1202:16: note: in expansion of macro 'PSA_PK_RSA_TO_MBEDTLS_ERR'
 1202 |         return PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk.c: In function 'mbedtls_pk_sign_ext':
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk_internal.h:26:67: error: 'psa_to_pk_rsa_errors' undeclared (first use in this function); did you mean 'psa_to_ssl_errors'?
   26 |                                                                   psa_to_pk_rsa_errors,            \
      |                                                                   ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/psa_util_internal.h:95:35: note: in definition of macro 'PSA_TO_MBEDTLS_ERR_LIST'
   95 |     psa_status_to_mbedtls(status, error_list,                         \
      |                                   ^~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk.c:1339:20: note: in expansion of macro 'PSA_PK_RSA_TO_MBEDTLS_ERR'
 1339 |             return PSA_PK_RSA_TO_MBEDTLS_ERR(status);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk.c: In function 'mbedtls_pk_verify_ext':
/home/jenkins/agent/workspace/bitbucket_build_platform/modules/crypto/mbedtls/library/pk.c:1228:1: error: control reaches end of non-void function [-Werror=return-type]
 1228 | }
      | ^
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v3.6.0-5645-gf414400b8fd8

butok, 2024-06-10:
JFYI: TFM is broken for all platforms, not only for NXP.

aescolar, 2024-06-10:
CC @valeriosetti

valeriosetti, 2024-06-11:
@hakehuang can you please check if https://github.com/zephyrproject-rtos/zephyr/pull/74052 solves the above mentioned issues? Please note that a `west update` is required after you checkout into that PR ;)

PR #74052 (2024-06-13): Multiple samples/tests fail to build in multiple platforms due to MBEDTLS / PSA issues
This PR aims at solving issues found in #73954 by properly adding `PSA_WANT` symbols where necessary.

Resolves #73954.

## Important note

~This commit also updates the `manifest.yaml` file because it depends on a fix that has been added into the official Mbed TLS repository (both `development` and `mbedtls-3.6` LTS branch). So the workflow for this PR is to:~
- ~review/accept https://github.com/zephyrproject-rtos/mbedtls/pull/57 first~
- ~update this PR by removing the `manifest.yaml` change~
- review/appr [... cut]
closed 2024-06-13 by henrikbrixandersen