zephyrproject-rtos/zephyr #73955: tests: can: imx93_evk/mimx9352/a55 : all can test fails

hakehuang, 2024-06-08:
**Describe the bug**
All can related application build failure on imx93_evk/mimx9352/a55

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 imx93_evk/mimx9352/a55
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
07f132b1b50ac803ab775a578a0890c1e46b4ed8 is the first bad commit
commit 07f132b1b50ac803ab775a578a0890c1e46b4ed8
Author: Chekhov Ma <chekhov.ma@nxp.com>
Date:   Fri May 24 13:56:16 2024 +0800

    boards: imx93_evk: enable flexcan driver
    
    Add pinctrl for flexcan2
    Add EXP_SEL gpio hog for board mux selection during boot.
    Add TJA1057 CAN PHY driver.
    Update supported features in board document
    
    Signed-off-by: Chekhov Ma <chekhov.ma@nxp.com>

 boards/nxp/imx93_evk/Kconfig.defconfig           | 14 ++++++++++++++
 boards/nxp/imx93_evk/doc/index.rst               |  8 ++++++++
 boards/nxp/imx93_evk/imx93_evk-pinctrl.dtsi      | 18 ++++++++++++++++++
 boards/nxp/imx93_evk/imx93_evk_mimx9352_a55.dts  | 22 ++++++++++++++++++++--
 boards/nxp/imx93_evk/imx93_evk_mimx9352_a55.yaml |  1 +
 5 files changed, 61 insertions(+), 2 deletions(-)
 create mode 100644 boards/nxp/imx93_evk/Kconfig.defconfig
```

**To Reproduce**
Steps to reproduce the behavior:
 west twister -p imx93_evk/mimx9352/a55 -s tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.txdl_64

**Expected behavior**
Build pass

**Impact**
CAN drivers

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:91:41: error: '__device_dts_ord_31' undeclared here (not in a function); did you mean '__device_dts_ord_32'?
   91 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/devicetree.h:4305:9: note: in expansion of macro 'COND_CODE_1'
 4305 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT),   \
      |         ^~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:69:53: note: in expansion of macro '__DEBRACKET'
   69 | #define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
      |                                                     ^~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_macro.h:224:9: note: in expansion of macro 'COND_CODE_1'
  224 |         COND_CODE_1(_flag, _code, ())
      |         ^~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/can/transceiver/can_transceiver_gpio.c:119:9: note: in expansion of macro 'IF_ENABLED'
  119 |         IF_ENABLED(DT_INST_NODE_HAS_PROP(inst, name##_gpios),           \
      |         ^~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/toolchain/common.h:138:23: note: in expansion of macro '_DO_CONCAT'
  138 | #define _CONCAT(x, y) _DO_CONCAT(x, y)
      |                       ^~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:91:33: note: in expansion of macro '_CONCAT'
   91 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:228:37: note: in expansion of macro 'DEVICE_NAME_GET'
  228 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:245:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  245 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/drivers/gpio.h:333:25: note: in expansion of macro 'DEVICE_DT_GET'
  333 |                 .port = DEVICE_DT_GET(DT_GPIO_CTLR_BY_IDX(node_id, prop, idx)),\
      |                         ^~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/drivers/gpio.h:395:9: note: in expansion of macro 'GPIO_DT_SPEC_GET_BY_IDX'
  395 |         GPIO_DT_SPEC_GET_BY_IDX(DT_DRV_INST(inst), prop, idx)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/drivers/gpio.h:422:9: note: in expansion of macro 'GPIO_DT_SPEC_INST_GET_BY_IDX'
  422 |         GPIO_DT_SPEC_INST_GET_BY_IDX(inst, prop, 0)
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/can/transceiver/can_transceiver_gpio.c:120:36: note: in expansion of macro 'GPIO_DT_SPEC_INST_GET'
  120 |                    (.name##_gpio = GPIO_DT_SPEC_INST_GET(inst, name##_gpios),))
      |                                    ^~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/can/transceiver/can_transceiver_gpio.c:130:17: note: in expansion of macro 'CAN_TRANSCEIVER_GPIO_COND'
  130 |                 CAN_TRANSCEIVER_GPIO_COND(inst, standby)                \
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/imx93_evk_mimx9352_a55/tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.unused/zephyr/include/generated/zephyr/devicetree_generated.h:24987:55: note: in expansion of macro 'CAN_TRANSCEIVER_GPIO_INIT'
24987 | #define DT_FOREACH_OKAY_INST_can_transceiver_gpio(fn) fn(0)
      |                                                       ^~
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_can_transceiver_gpio'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/drivers/can/transceiver/can_transceiver_gpio.c:138:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
  138 | DT_INST_FOREACH_STATUS_OKAY(CAN_TRANSCEIVER_GPIO_INIT)
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-5645-gf414400b8fd8

hakehuang, 2024-06-08:
failure test application list

<html>
<body>

CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
drivers.can.shell | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/drivers/can/shell
drivers.can.host | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/drivers/can/host
drivers.can.api | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/drivers/can/api
drivers.can.api.rtr | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/drivers/can/api
drivers.can.timing | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/drivers/can/timing
canbus.isotp.conformance | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/subsys/canbus/isotp/conformance
canbus.isotp.conformance.fd.unused | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/subsys/canbus/isotp/conformance
canbus.isotp.conformance.fd.txdl_32 | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/subsys/canbus/isotp/conformance
canbus.isotp.conformance.fd.txdl_64 | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/subsys/canbus/isotp/conformance
canbus.isotp.implementation | v3.6.0-5645-gf414400b8fd8 | Build failure | imx93_evk/mimx9352/a55 | tests/subsys/canbus/isotp/implementation

</body>
</html>

dleach02, 2024-06-11:
@CkovMk please sync with @hakehuang on this bug report. Your PR merged after you confirmed that twister passed:

https://github.com/zephyrproject-rtos/zephyr/pull/70201#issuecomment-2102240014

henrikbrixandersen, 2024-06-11:
Raising priority as this is causing CI failures.

CkovMk, 2024-06-12:
@hakehuang does this #73758 fixes build failure? It seems to be an node enable / KConfig enable issue.

CkovMk, 2024-06-12:
In #69330 there was a comment by @JiafeiPan :

https://github.com/zephyrproject-rtos/zephyr/pull/69330#discussion_r1566615601

Which stated that I should get nodes disabled by default, only enable the one required in dts. We came up with a solution that we keep these nodes enabled to let ci run checks on that pr, then disable nodes just before it gets merged.

In #70201 which caused this issue, It was merged in a hurry. I confirmed the twister build pass on my local machine, and I didn't watch it for a while, then I came back to find it was merged. I did not have time to disable related nodes, thus caused such a chaos.

Sorry for that.

As for #73757 , I believe it is an driver issue. Since what I did is just SoC & board level enablement, and all Classic CAN tests have passed, I also have working application that runs this driver (in classic can mode), I think it's okay to keep my pr.

Solution purposal:
1. Disable FlexCAN related nodes (i2c, gpio expander, transceiver, flexcan) in board dts, write separate overlays & KConfig fragments to ensure can test build pass.
2. Enable FlexCAN related nodes, add KConfig options to board defconfig, ensure can test build pass.

ping @dleach02 @henrikbrixandersen @hakehuang

JiafeiPan, 2024-06-12:
> In #69330 there was a comment by @JiafeiPan :
> 
> [#69330 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/69330#discussion_r1566615601)
> 
> Which stated that I should get nodes disabled by default, only enable the one required in dts. We came up with a solution that we keep these nodes enabled to let ci run checks on that pr, then disable nodes just before it gets merged.
> 
> In #70201 which caused this issue, It was merged in a hurry. I confirmed the twister build pass on my local machine, and I didn't watch it for a while, then I came back to find it was approved and merged. CI didn't raise any issue back then. I did not have time to disable related nodes, thus caused such a chaos.
> 
> Sorry for that.
> 
> As for #73757 , I believe it is an driver issue. Since what I did is just SoC & board level enablement, and all Classic CAN tests have passed, I also have working application that runs this driver (in classic can mode), I think it's okay to keep my pr.
> 
> Solution purposal:
> 
> 1. Disable FlexCAN related nodes (i2c, gpio expander, transceiver, flexcan) in board dts, write separate overlays & KConfig fragments to ensure can test build pass.
> 2. Enable FlexCAN related nodes, add KConfig options to board defconfig, ensure can test build pass.
> 
> ping @dleach02 @henrikbrixandersen @hakehuang

I still prefer to use the first proposal.

CkovMk, 2024-06-12:
updated #73758.

PR #74103 (2024-06-12): Revert "soc: imx93: enable flexcan driver"
This reverts commit 69360d2f386ce808caeb715819c4ab1b717571c1..07f132b1b50ac803ab775a578a0890c1e46b4ed8 from #70201 which are causing CI build errors. Apart from the build failures, it appears the CAN tests are not even passing on this board (see https://github.com/zephyrproject-rtos/zephyr/issues/73757 and https://github.com/zephyrproject-rtos/zephyr/pull/73758 which is reported by the same person, who added the functionality?).

Fixes: #73955
commit 51e52eae (2024-06-12): Revert "boards: imx93_evk: enable flexcan driver"

This reverts commit 07f132b1b50ac803ab775a578a0890c1e46b4ed8 which is
causing CI build errors.

Fixes: #73955
commit 58db527d (2024-06-12): Revert "soc: imx93: enable flexcan driver"

This reverts commit 69360d2f386ce808caeb715819c4ab1b717571c1 which is
causing CI build errors.

Fixes: #73955
closed 2024-06-12 by henrikbrixandersen