zephyrproject-rtos/zephyr #73958: drivers: gpio: Apollo3 Blue MCU is not able to SX1262 Dio1 interrupt

srcnert, 2024-06-08:
Apollo3 Blue MCU is not able to get Dio1 interrupt of the SX1262 transceiver. For test purposes, I used sample/subsys/lorawan/class_a example project.

The code is running on RAK11720 module from RAKwireless. It is based on Apollo3 Blue SoC with a Semtech SX1262 transceiver. Currently, I am trying to add rak11720 board support to zephyr project.

This overlay file can be used to compile class_a example project:
[apollo3_evb-overlay.txt](https://github.com/user-attachments/files/15748269/apollo3_evb-overlay.txt)

The main problem is that inside ambiq_gpio_pin_interrupt_configure function pincfg.eGPRdZero flag comes as 1 while setting up SX1262 Dio1 interrupt. If I set it to zero, I am able to get Dio1 interrupt from SX1262. Please check following file to see how I fixed it.
[gpio_ambiq-c.txt](https://github.com/user-attachments/files/15748289/gpio_ambiq-c.txt)

Commit SHA or Version used: main branch 8136f740df63 commit id

AlessandroLuo, 2024-06-10:
@srcnert thank you for the finding, this is a known issue to us, it will be fixed in https://github.com/zephyrproject-rtos/zephyr/pull/73890

srcnert, 2024-06-10:
> @srcnert thank you for the finding, this is a known issue to us, it will be fixed in #73890

Thanks for update.

closed 2024-06-10 by srcnert