zephyrproject-rtos/zephyr #73988: tfm: build is broken

butok, 2024-06-10:
**Describe the bug**
The TFM sample build is failed for all NS platforms.
**To Reproduce**
1. Build any TFM sample according to https://docs.zephyrproject.org/latest/services/tfm/build.html
2. For example: `west build -p auto -b mps2_an521_ns samples/tfm_integration/psa_protected_storage`
3. See build error:
```
CMake Error at cmake/remote_library.cmake:111 (message):
  Failed to apply patches at
  C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/lib/ext/cmsis-src
```
**Expected behavior**
Build PASS

**Impact**
TFM support is brocken

**Logs and console output**
```
C:\_ddm\ZEPHYR\zephyr_vs\zephyr>west build -p always -b mps2_an521_ns samples/tfm_integration/psa_protected_storage
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/samples/tfm_integration/psa_protected_storage
-- CMake version: 3.28.1
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.4", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/.cache
-- Zephyr version: 3.6.99 (C:/_ddm/ZEPHYR/zephyr_vs/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Warning at C:/_ddm/ZEPHYR/zephyr_vs/zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=mps2_an521_ns specified, board automatically changed to:
  mps2/an521/cpu0/ns.
Call Stack (most recent call first):
  C:/_ddm/ZEPHYR/zephyr_vs/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  C:/_ddm/ZEPHYR/zephyr_vs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/_ddm/ZEPHYR/zephyr_vs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- Board: mps2, qualifiers: an521/cpu0/ns
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.6 (C:/Users/nxa12205/zephyr-sdk-0.16.6)
-- Found toolchain: zephyr 0.16.6 (C:/Users/nxa12205/zephyr-sdk-0.16.6)
-- Found Dtc: C:/Users/nxa12205/.mcuxpressotools/dtc-msys2/tools/usr/bin/dtc.exe (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/boards/arm/mps2/mps2_an521_cpu0_ns.dts
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/dts.cmake
Parsing C:/_ddm/ZEPHYR/zephyr_vs/zephyr/Kconfig
Loaded configuration 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/boards/arm/mps2/mps2_an521_cpu0_ns_defconfig'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/samples/tfm_integration/psa_protected_storage/prj.conf'
Configuration saved to 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/nxa12205/zephyr-sdk-0.16.6/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/nxa12205/zephyr-sdk-0.16.6/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Using ccache: C:/Strawberry/c/bin/ccache.exe
CMake Warning at C:/_ddm/ZEPHYR/zephyr_vs/zephyr/modules/trusted-firmware-m/CMakeLists.txt:512 (message):
  TFM_DUMMY_PROVISIONING is enabled:

        The device will be provisioned using dummy keys and is NOT secure!
        This is not suitable for production

-- Configuring done (28.6s)
-- Generating done (0.9s)
-- Build files have been written to: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build
-- west build: building application
[1/229] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.6.99 (C:/_ddm/ZEPHYR/zephyr_vs/zephyr), build: v3.6.0-5666-g453ab8a9a356
[6/229] Generating ../../tfm/CMakeCache.txt
FAILED: tfm/CMakeCache.txt C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/CMakeCache.txt
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\_ddm\ZEPHYR\zephyr_vs\zephyr\build\tfm && C:\Users\nxa12205\.mcuxpressotools\cmake-3.28.1-windows-x86_64\bin\cmake.exe -GNinja -DTFM_TOOLCHAIN_FILE=C:/_ddm/ZEPHYR/zephyr_vs/modules/tee/tf-m/trusted-firmware-m/toolchain_GNUARM.cmake -DCROSS_COMPILE=C:/Users/nxa12205/zephyr-sdk-0.16.6/arm-zephyr-eabi/bin/arm-zephyr-eabi -DCMAKE_BUILD_TYPE=MinSizeRel -DTFM_PLATFORM=arm/mps2/an521 -DCONFIG_TFM_BUILD_LOG_QUIET=ON -DCONFIG_TFM_MEMORY_USAGE_QUIET=OFF -DPython3_EXECUTABLE=C:/Python311/python.exe -DCONFIG_TFM_SPM_BACKEND="IPC" -DBL2=TRUE -DMCUBOOT_IMAGE_VERSION_S=0.0.0+0 -DMCUBOOT_IMAGE_VERSION_NS=0.0.0+0 -DMCUBOOT_SIGNATURE_TYPE=EC-P256 -DMCUBOOT_USE_PSA_CRYPTO=ON -DMCUBOOT_BUILTIN_KEY=ON -DMCUBOOT_HW_KEY=OFF -DMCUBOOT_KEY_S=C:/_ddm/ZEPHYR/zephyr_vs/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-EC-P256.pem -DMCUBOOT_KEY_NS=C:/_ddm/ZEPHYR/zephyr_vs/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-EC-P256_1.pem -DTFM_ISOLATION_LEVEL=1 -DMCUBOOT_IMAGE_NUMBER=2 -DTFM_DUMMY_PROVISIONING=ON -DTFM_EXCEPTION_INFO_DUMP=ON -DMCUBOOT_LOG_LEVEL=INFO -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_PARTITION_NS_AGENT_MAILBOX=OFF -DTFM_PARTITION_PROTECTED_STORAGE=ON -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE=ON -DTFM_PARTITION_CRYPTO=ON -DTFM_PARTITION_PLATFORM=ON -DTFM_PARTITION_INITIAL_ATTESTATION=OFF -DTFM_PARTITION_FIRMWARE_UPDATE=OFF -DMCUBOOT_PATH=C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot -DCONFIG_TFM_ENABLE_FP=OFF -DTFM_TESTS_REVISION_CHECKS=OFF -DMBEDCRYPTO_PATH=C:/_ddm/ZEPHYR/zephyr_vs/modules/crypto/mbedtls C:/_ddm/ZEPHYR/zephyr_vs/modules/tee/tf-m/trusted-firmware-m"
CMake Warning at cmake/version.cmake:22 (message):
  Actual TF-M version is not available from Git repository.  Settled to
  v2.1.0
Call Stack (most recent call first):
  CMakeLists.txt:13 (include)

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.41.0.windows.1")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/nxa12205/zephyr-sdk-0.16.6/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Populating cmsis
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/lib/ext/cmsis-subbuild
Cloning into 'cmsis-src'...
remote: Enumerating objects: 28334, done.
remote: Counting objects: 100% (3471/3471), done.
remote: Compressing objects: 100% (952/952), done.
remote: Total 28334 (delta 2900), reused 2828 (delta 2458), pack-reused 24863
Receiving objects: 100% (28334/28334), 39.76 MiB | 8.09 MiB/s, done.
Resolving deltas: 100% (20830/20830), done.
HEAD is now at d0c460c1 CMSIS-Core(M): Add Cortex-M52 support : add core_m52.h
[1/9] Creating directories for 'cmsis-populate'
[1/9] Performing download step (git clone) for 'cmsis-populate'
[2/9] Performing update step for 'cmsis-populate'
[3/9] No patch step for 'cmsis-populate'
[5/9] No configure step for 'cmsis-populate'
[6/9] No build step for 'cmsis-populate'
[7/9] No install step for 'cmsis-populate'
[8/9] No test step for 'cmsis-populate'
[9/9] Completed 'cmsis-populate'
'C:/Program Files/Git/cmd/git.exe' 'apply' '--verbose' 'C:/_ddm/ZEPHYR/zephyr_vs/modules/tee/tf-m/trusted-firmware-m/lib/ext/cmsis/0001-iar-Add-missing-v8.1m-check.patch'
error: corrupt patch at line 19
CMake Error at cmake/remote_library.cmake:111 (message):
  Failed to apply patches at
  C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/lib/ext/cmsis-src
Call Stack (most recent call first):
  cmake/remote_library.cmake:287 (apply_patches)
  lib/ext/cmsis/CMakeLists.txt:17 (fetch_remote_library)

-- Configuring incomplete, errors occurred!
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Users\nxa12205\.mcuxpressotools\cmake-3.28.1-windows-x86_64\bin\cmake.EXE' --build 'C:\_ddm\ZEPHYR\zephyr_vs\zephyr\build'
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 453ab8a9a356acf475a965a777a370795effa255

aescolar, 2024-06-11:
@butok I cannot reproduce this on Linux neither on main or in the sha pointed (https://github.com/zephyrproject-rtos/zephyr/commit/453ab8a9a356acf475a965a777a370795effa255)
CC @tomi-font @Vge0rge @ithinuel

butok, 2024-06-11:
> @butok I cannot reproduce this on Linux neither on main or in the sha pointed ([453ab8a](https://github.com/zephyrproject-rtos/zephyr/commit/453ab8a9a356acf475a965a777a370795effa255)) CC @tomi-font @Vge0rge @ithinuel

OK. You can close it as not reproduced. 
As west pristine build does not work for me now and I have to delete the build folder manually, I have feeling that it caused by our new company security restrictions.

erwango, 2024-06-18:
I'm getting exactly the same issue on my side on Linux and a colleague on Windows, on 2 different boards using `v3.7.0-rc1-8-gc52094facdd`

butok, 2024-06-18:
>I'm getting exactly the same issue on my side on Linux and a colleague on Windows, on 2 different boards using v3.7.0-rc1-8-gc52094facdd
> @butok Do you still see the issue on your side ?

Yes, I still has this issue. I am not alone ;) 
Reopening issue.

ithinuel, 2024-06-18:
@erwango I cannot reproduce this error on your zephyr fork at c52094facdd. Which board are you trying to build for ? Since the error message is different I am not sure this is the same issue.

@butok I had a chat with @adeaarm and since the patch applies fine under linux, we're wondering if it couldn't be an issue with line ending.

aescolar, 2024-06-18:
@erwango as your error message is different it would seem the problem is different. Could you create another bug report with the repro steps for your issue?
(Otherwise this issue may easily become a bucket of "TFM is broken for me" reports which are not really actionable)

butok, 2024-06-18:
> @butok I had a chat with @adeaarm and since the patch applies fine under linux, we're wondering if it couldn't be an issue with line ending. Would you mind checking that for me please? I don't have a windows machine at hand to test.

On my colleague's Linux machine the build is passed. 
It is failed on Windows 10/11. 
So it can be an issue for building TFM on Windows.

butok, 2024-06-24:
The TFM build problem has been reproduced on several Windows10/11 systems (including my home Win10 PC).
Guess, the issue priority should be increased.

adeaarm, 2024-06-24:
Can you raise this on the tfm mailing list? We tested on 1 window pc and couldn't reproduce, so it would be great if you can attach the patch file so that we can diff.

Thanks, Antonio

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

butok, 2024-06-24:
> so it would be great if you can attach the patch file so that we can diff.

Attached [0001-iar-Add-missing-v8.1m-check.patch](https://github.com/user-attachments/files/15952233/0001-iar-Add-missing-v8.1m-check.patch)

adeaarm, 2024-06-25:
Hi @butok , can you try with this one and let me know?

[0001-iar-Add-missing-v8.1m-check.patch](https://github.com/user-attachments/files/15969794/0001-iar-Add-missing-v8.1m-check.patch)

butok, 2024-06-25:
> Hi @butok , can you try with this one and let me know?

With this patch, the TFM build is successful.

butok, 2024-06-25:
JFYI: Git Windows installer, if to install without changing proposed settings, sets git global autocrlf to "true".

ithinuel, 2024-06-25:
Could it be that what's fetched by cmake isn't following that globally set rule ?

I saw some articles (like this [one](https://markentier.tech/posts/2021/10/autocrlf-true-considered-harmful/)) recommending not to use that and forcing it to `lf` only instead.
Note I'm not doing any development on windows so I don't have opinions on this matter, but it might be worth trying this.

adeaarm, 2024-06-25:
> JFYI: Git Windows installer, if to install without changing proposed settings, sets git global autocrlf to "true".

I believe setting `autocrlf` to `false` and forcing to `eol` to `lf` should be the recommended way of configuring git on Windows, at least for TF-M.

butok, 2024-06-25:
> I believe setting `autocrlf` to `false` and forcing to `eol` to `lf` should be the recommended way of configuring git on Windows, at least for TF-M.

Please do not complicate Windows dev. life.
**Just do the fix in TFM, as it was all these >3 years.**

adeaarm, 2024-06-25:
> > I believe setting `autocrlf` to `false` and forcing to `eol` to `lf` should be the recommended way of configuring git on Windows, at least for TF-M.
> 
> Please do not complicate Windows dev. life. **Just do the fix in TFM, as it was all these >3 years.**

Which fix? TF-M is behaving the same as it always had in these >3 years. As I mentioned already, the patch hosted in TF-M's Gerrit has LF ending, there is nothing to fix that I can see.

butok, 2024-06-25:
> Which fix? TF-M is behaving the same as it always had in these >3 years. As I mentioned already, the patch hosted in TF-M's Gerrit has LF ending, there is nothing to fix that I can see.

You have sent the modified patch that works.

adeaarm, 2024-06-25:
> > Which fix? TF-M is behaving the same as it always had in these >3 years. As I mentioned already, the patch hosted in TF-M's Gerrit has LF ending, there is nothing to fix that I can see.
> 
> You have sent the modified patch that works.

As I mentioned already, the patch that I sent is the patch file that TF-M hosts in Gerrit. It has LF endings as expected. The patch you're trying to apply has modified line endings, which is not the patch that should be applied.

butok, 2024-06-25:
>The patch you're trying to apply has modified line endings, which is not the patch that should be applied.

**The TFM was buildable on Windows 10 from its public release till this moment.**
The issue is reproducible on all available to me Win10 and Win11.
Hope, TFM will be able to find a solution to make it usable again for a Windows community.

ithinuel, 2024-06-25:
@butok Are you using WSL and if not would you mind giving it a try?

I'll try once I'm back home, with a bit of luck it won't mess too much with my windows (I've never used WSL).

butok, 2024-06-26:
> @butok Are you using WSL and if not would you mind giving it a try?

It's buildable in WSL.  But it's not possible to flash a board cause of some USB issues.
WSL is Linux, here were should solve the Windows TFM problem.
BTW: Now, the Standalone TFM is not buildable on Windows too.

butok, 2024-09-02:
Checked. The Zephyr TFM build is passed on Windows, for the latest main (0492235d0dd84ad468acf84a4e3c97a88d98c700)
Closing the issue.

closed 2024-09-02 by butok