zephyrproject-rtos/zephyr #74004: cypress_psoc6_uart driver fails to build in multiple samples/tests

aescolar, 2024-06-10:
**Describe the bug**
drivers/serial/uart_psoc6.c fails to build in many samples/tests in CI

* cy8ckit_062_ble
* cy8ckit_062_wifi_bt

**To Reproduce**
Steps to reproduce the behavior:
1. twister -T samples/drivers/uart/echo_bot/ -s sample.drivers.uart -p cy8ckit_062_wifi_bt/cy8c6247/m4
4. See error

**Expected behavior**
No build errors

**Impact**
Many (>=170) weekly CI build failures 

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/runs/25996829280
```
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:335:28: error: 'cy_psoc6_uart0_irq_config' undeclared here (not in a function); did you mean 'cy_psoc6_uart0_config'?
  335 |         .irq_config_func = cy_psoc6_uart##n##_irq_config
      |                            ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/devicetree.h:4305:9: note: in expansion of macro 'COND_CODE_1'
 4305 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT),   \
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:353:17: note: in expansion of macro 'CY_PSOC6_UART_IRQ_SET_FUNC'
  353 |                 CY_PSOC6_UART_IRQ_SET_FUNC(n)                                   \
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/twister-out/cy8ckit_062_ble@0.0.0_cy8c6347_m0/tests/subsys/modbus/modbus.rtu.build_only/zephyr/include/generated/zephyr/devicetree_generated.h:11462:53: note: in expansion of macro 'CY_PSOC6_UART_INIT'
11462 | #define DT_FOREACH_OKAY_INST_cypress_psoc6_uart(fn) fn(0)
      |                                                     ^~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_cypress_psoc6_uart'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:361:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
  361 | DT_INST_FOREACH_STATUS_OKAY(CY_PSOC6_UART_INIT)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:338:41: error: 'cy_psoc6_uart0_data' undeclared here (not in a function); did you mean 'cy_psoc6_uart0_config'?
  338 | #define CY_PSOC6_UART_DECL_DATA_PTR(n) &cy_psoc6_uart##n##_data
      |                                         ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/devicetree.h:4305:9: note: in expansion of macro 'COND_CODE_1'
 4305 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT),   \
      |         ^~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:1013:17: note: in expansion of macro 'Z_DEVICE_INIT'
 1013 |                 Z_DEVICE_INIT(name, pm, data, config, api, state, deps,                            \
      |                 ^~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:1101:9: note: in expansion of macro 'Z_DEVICE_BASE_DEFINE'
 1101 |         Z_DEVICE_BASE_DEFINE(node_id, dev_id, name, pm, data, config, level,    \
      |         ^~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:198:9: note: in expansion of macro 'Z_DEVICE_DEFINE'
  198 |         Z_DEVICE_DEFINE(node_id, Z_DEVICE_DT_DEV_ID(node_id),                  \
      |         ^~~~~~~~~~~~~~~
/__w/zephyr/zephyr/include/zephyr/device.h:213:9: note: in expansion of macro 'DEVICE_DT_DEFINE'
  213 |         DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
      |         ^~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:355:9: note: in expansion of macro 'DEVICE_DT_INST_DEFINE'
  355 |         DEVICE_DT_INST_DEFINE(n, &uart_psoc6_init, NULL,                        \
      |         ^~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:356:31: note: in expansion of macro 'CY_PSOC6_UART_DECL_DATA_PTR'
  356 |                               CY_PSOC6_UART_DECL_DATA_PTR(n),                   \
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/twister-out/cy8ckit_062_ble@0.0.0_cy8c6347_m0/tests/subsys/modbus/modbus.rtu.build_only/zephyr/include/generated/zephyr/devicetree_generated.h:11462:53: note: in expansion of macro 'CY_PSOC6_UART_INIT'
11462 | #define DT_FOREACH_OKAY_INST_cypress_psoc6_uart(fn) fn(0)
      |                                                     ^~
/__w/zephyr/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_cypress_psoc6_uart'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:361:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
  361 | DT_INST_FOREACH_STATUS_OKAY(CY_PSOC6_UART_INIT)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/serial/uart_psoc6.c:294:13: error: 'uart_psoc6_isr' defined but not used [-Werror=unused-function]
  294 | static void uart_psoc6_isr(const struct device *dev)
      |             ^~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: main as of now b46fb464cbaaad8d094f75487a3502a11126d025

**Additional context**

aescolar, 2024-06-10:
CC @npal-cy

npal-cy, 2024-06-10:
Hi @aescolar, i pushed fix: https://github.com/zephyrproject-rtos/zephyr/pull/74010

closed 2024-06-12 by nashif