zephyrproject-rtos/zephyr #74025: Contributor nominations for @nordic-piks and @katgiadla

PerMac, 2024-06-10:
# Nomination
 
for Contributors

## GitHub User

1. Piotr Kosycarz
1. @nordic-piks 
1. Nordic Semiconductor

2. Katarzyna Giądła
2. @katgiadla 
2. Nordic Semiconductor

## Supporting Documents
Both Piotr and Katarzyna are actively reporting issues detected when executing tests on Nordic hardware. Adding them as contributors to the project and giving them triage rights would reduce the amount of work done by Triage wg for the issues they report. They already provided multiple fixes to test configurations and some improvements to test related tooling used in the project:

https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+author%3Akatgiadla
https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+author%3Anordic-piks

nashif, 2024-06-12:
https://github.com/zephyrproject-rtos/zephyr/invitations

katgiadla, 2024-06-12:
> https://github.com/zephyrproject-rtos/zephyr/invitations

After click into link, I see message:

[image]

could you sent invitation once again, please?

nordic-piks, 2024-06-20:
For me working fine, thanks for nomination!

PerMac, 2024-06-20:
@nashif Can you send/check the invitation for @katgiadla? As reported, the link doesn't work for Katarzyna. Katarzyna also haven't received mail invitation (we checked in spam).

katgiadla, 2024-06-27:
The e-mail with invitation reach to me. Thanks :)

closed 2024-06-27 by katgiadla