zephyrproject-rtos/zephyr #74051: sysbuild: Allow simpler paths (e.g. in DTC_OVERLAY_FILE) when subimage is not in the application folder

nordic-krch, 2024-06-11:
**Is your feature request related to a problem? Please describe.**
If sysbuild is used and subimage is a different application (bootloader, vpr launcher, etc. ) then any paths provided to that application are relative to the location of that subimage application. We have a case where we are using `vpr_launcher` on `cpuapp` for tests running on nrf54h20 `cpuppr`. `vpr_launcher` requires an overlay and path to that overlay looks ugly since it is relative to the `vpr_launcher` application and file is placed in the test folder:

https://github.com/zephyrproject-rtos/zephyr/blob/9e3cbfe8be0e4b4a10853367916fceb42359998c/tests/drivers/counter/counter_basic_api/testcase.yaml#L50

**Describe the solution you'd like**
It would be good to have a cmake variable that would point to the main (test application in my case) so that overlay could be provided as `${APP_DIR}/boards/vpr_launcher.overlay`.

nordicjm, 2025-10-31:
This is already supported by using the system described on https://docs.zephyrproject.org/latest/build/sysbuild/index.html#zephyr-application-configuration-directory you would create a `<app>/sysbuild/vpr_launcher` folder then copy the `prj.conf` and `app.overlay` files there, then create a `boards` sub folder and put your board in there.

boaks, 2026-08-17:
Though I found my self in a similar situation, I want to point out, that AFAIK the support only includes overlays with specific names, and doesn't allow to use multiple overlay with custom names, e.g. for different "aspects". E.g. for mcuboot, only one overlay "mcuboot.overlay" seems to be supported, or the board specific overlays. But in the original overlay idea without sysbuild it was easy to apply several overlays using free names, e.g. "-DEXTRA_DTC_OVERLAY_FILE=ledrgb.overlay -DEXTRA_DTC_OVERLAY_FILE=solarcharger.overlay", which makes it pretty easy to support a set of variants. 
(Sometimes it's not too clear, if such things are simply not considered or if it turns out to be to complex to implement it.)

nordicjm, 2026-08-17:
> Though I found my self in a similar situation, I want to point out, that AFAIK the support only includes overlays with specific names, and doesn't allow to use multiple overlay with custom names, [...]

They are still supported, you can supply as many filenames as you want to `-Dmcuboot_EXTRA_DTC_OVERLAY_FILE=""`

boaks, 2026-08-17:
Doesn't that then require to either use "(long) full-path" or "(long) relative paths"? 
Maybe I missed it, but it's not too easy to see, how that works for sysbuild components.

nordicjm, 2026-08-17:
> Doesn't that then require to either use "(long) full-path" or "(long) relative paths"? Maybe I missed it, but it's not too easy to see, how that works for sysbuild components.

`-Dmcuboot_EXTRA_DTC_OVERLAY_FILE="\\${APP_DIR}/dts/some_file.dts"` -> `<app_dir>/dts/some_file.dts`

boaks, 2026-08-17:
Hmm ... not too sure ... it may be caused by limitations / specific shell (windows/bash) used on my side, but I can't get it working with ${APP_DIR}.

``` 
west -v build ... -Dmcuboot_EXTRA_DTC_OVERLAY_FILE="\${APP_DIR}/boards/one_ns.overlay"

Running CMake: 'C:\ncs\toolchains\dcbdc366a1\opt\bin\cmake.EXE' -DWEST_PYTHON=C:/ncs/toolchains/dcbdc366a1/opt/bin/python.exe '-BC:\ncs\v3.4.0\one\build' -GNinja -DBOARD=nrf9151dk/nrf9151/ns '-DEXTRA_DTC_OVERLAY_FILE='"'"'boards/one_ns.overlay'"'"'' '-Dmcuboot_EXTRA_DTC_OVERLAY_FILE=${APP_DIR}/boards/one_ns.overlay' '-SC:\ncs\v3.4.0\zephyr\share\sysbuild'

...

-- Found BOARD.dts: C:/ncs/v3.4.0/zephyr/boards/nordic/nrf9151dk/nrf9151dk_nrf9151.dts
-- Found devicetree overlay: C:/ncs/v3.4.0/one/sysbuild/mcuboot/boards/nrf9151dk_nrf9151.overlay
-- Found devicetree overlay: /boards/one_ns.overlay
<command-line>: fatal error: /boards/one_ns.overlay: No such file or directory
``` 

Using "\\${APP_DIR}" instead of "\${APP_DIR}" results then in 

``` 
<command-line>: fatal error: //boards/one_ns.overlay: No such file or directory
``` 

so doesn't work either.

Or, maybe I got the idea wrong, and it's up to the caller, to setup APP_DIR with the right path?

boaks, 2026-08-17:
I tried to debug it with some "message" statements in the CMakeLists and sysbuild.cmake:
APP_DIR has the expected value, but it seems to be not used for -Dmcuboot_EXTRA_DTC_OVERLAY_FILE.

boaks, 2026-08-17:
At least using the zephyr 4.4 (NCS 3.4.0) my analysis results in:

cmake/modules/configuration_files.cmake:

```
zephyr_get(EXTRA_DTC_OVERLAY_FILE SYSBUILD LOCAL MERGE REVERSE)
```

but unluckily, APP_DIR isn't available at that point in time and so it's replaced by an empty string.

Adding

```
if(NOT DEFINED APP_DIR AND DEFINED ENV{APP_DIR})
   set(APP_DIR $ENV{APP_DIR})
endif()
```

before that `zephyr_get`, makes it magically working.

@nordicjm 
So, I guess you haven't tested the proposed solution?

tejlmand, 2026-08-19:
@boaks there is currently an issue Zephyr CMake so that only variables already existing in current CMake scope can be used through sysbuild. `APP_DIR` does not exists at the time `EXTRA_DTC_OVERLAY_FILE` is resolved, resulting in `APP_DIR` being empty.

This is fixed in #116740 which looks up unresolved variables during `zephyr_get()` so variables, like APP_DIR`, which exists through sysbuild can be used.

Please take a look at the PR and see if that fixes your use case.
Note, it's still important to escape the `$` using backslah, like `-DEXTRA_DTC_OVERLAY_FILE="\${APP_DIR}/file.overlay"`

tejlmand, 2026-08-24:
> I added the lines (foreach ...) of [#116740](https://github.com/zephyrproject-rtos/zephyr/pull/116740) to zephyr commit [bf801e4](https://github.com/zephyrproject-rtos/zephyr/commit/bf801e4e3d19e1ffa76164346480cb7734dd2800) (as used by NCS 3.4.0), but that still fails.

thanks for testing.

Just tried the same on ncs 3.4.0, and found that it only resolved only correctly global, not local var.
Fixed now in the PR, so feel free to retry.
See: https://github.com/zephyrproject-rtos/zephyr/compare/2ee4e4a42347164c07e295c17d7d15d92f86bda0..5e81eb047be0201a92ac6e2b41bf82b52c34bfc4

tejlmand, 2026-08-24:
@boaks my test command:
```
$ west build -p -b nrf52840dk/nrf52840 samples/hello_world --cmake-only -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -Dmcuboot_EXTRA_CONF_FILE="\${APP_DIR}/test.conf"
-- west build: making build dir /projects/github/ncs/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): sysbuild_default
...
-- 
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
...
Loaded configuration '/projects/github/ncs/zephyr/boards/nordic/nrf52840dk/nrf52840dk_nrf52840_defconfig'
Merged configuration '/projects/github/ncs/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration '/projects/github/ncs/bootloader/mcuboot/boot/zephyr/boards/nrf52840dk_nrf52840.conf'
Merged configuration '/projects/github/ncs/zephyr/snippets/rtt-console/rtt-console.conf'
Merged configuration '/projects/github/ncs/zephyr/samples/hello_world/test.conf'
Merged configuration '/projects/github/ncs/zephyr/build/mcuboot/zephyr/.config.sysbuild'
Configuration saved to '/projects/github/ncs/zephyr/build/mcuboot/zephyr/.config'
...
```

The `${APP_DIR}/test.conf` resolves correctly on NCS v3.4.0 with the updated #116740

boaks, 2026-08-24:
Trying the updated PR with

```
-Dmcuboot_EXTRA_DTC_OVERLAY_FILE="\${APP_DIR}/test.dtsi" 
```

works! Thanks a lot, very nice.

boaks, 2026-08-24:
> try APPLICATION_CONFIG_DIR as the variable instead

That result in "sysbuild/mcuboot/test.dtsi", which may be pretty helpful in other cases.

closed 2026-09-01 by fabiobaltieri