zephyrproject-rtos/zephyr #74066: frdm_k64f

arikgreen, 2024-06-11:
Describe the bug
The tests/subsys/storage/flash_map/storage.flash_map.mpu failed.

```
Tue, 11 Jun 2024 09:52:40 GMT INFO    - frdm_k64f    tests/subsys/storage/flash_map/storage.flash_map.mpu  FAILED Failed (device: xxx, 10.815s)
```

To Reproduce
```
scripts/twister -W -p frdm_k64f --device-testing -T tests/subsys/storage/flash_map -v
```
```
The following issues were found (showing the top 10 items):
Tue, 11 Jun 2024 09:53:10 GMT INFO    - 1) tests/subsys/storage/flash_map/storage.flash_map.mpu on frdm_k64f failed (Failed)
Tue, 11 Jun 2024 09:53:10 GMT INFO    - 
Tue, 11 Jun 2024 09:53:10 GMT INFO    - To rerun the tests, call twister using the following commandline:
Tue, 11 Jun 2024 09:53:10 GMT INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
Tue, 11 Jun 2024 09:53:10 GMT INFO    - 
Tue, 11 Jun 2024 09:53:10 GMT INFO    - west twister -p frdm_k64f -s tests/subsys/storage/flash_map/storage.flash_map.mpu
Tue, 11 Jun 2024 09:53:10 GMT INFO    - or with west:
Tue, 11 Jun 2024 09:53:10 GMT INFO    - west build -p -b frdm_k64f tests/subsys/storage/flash_map -T storage.flash_map.mpu
```

Logs and console output
```
INFO    - frdm_k64f    tests/subsys/storage/flash_map/storage.flash_map.mpu  FAILED Failed (device: xxx, 10.815s)
INFO    - ./twister-out/frdm_k64f/tests/subsys/storage/flash_map/storage.flash_map.mpu/handler.log
ERROR   - ***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build v3.6.0-5742-g2efc447c1b41 (delayed boot 500ms) ***
Running TESTSUITE flash_map
===================================================================
START - test_fixed_partition_node_macros
PASS - test_fixed_partition_node_macros in 0.001 seconds
===================================================================
START - test_flash_area_check_int_sha256
PASS - test_flash_area_check_int_sha256 in 0.159 seconds
===================================================================
START - test_flash_area_disabled_device
PASS - test_flash_area_disabled_device in 0.001 seconds
===================================================================
START - test_flash_area_erase_and_flatten
Assertion failed at ./tests/subsys/storage/flash_map/src/main.c:242: flash_map_test_flash_area_erase_and_flatten: (erased is false)
Erase failed at index 1
FAIL - test_flash_area_erase_and_flatten in 4.744 seconds
===================================================================
START - test_flash_area_erased_val
PASS - test_flash_area_erased_val in 0.001 seconds
===================================================================
START - test_flash_area_get_sectors
PASS - test_flash_area_get_sectors in 0.904 seconds
===================================================================
TESTSUITE flash_map failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  83.33% [flash_map]: pass = 5, fail = 1, skip = 0, total = 6 duration = 5.810 seconds
- PASS - [flash_map.test_fixed_partition_node_macros] duration = 0.001 seconds
- PASS - [flash_map.test_flash_area_check_int_sha256] duration = 0.159 seconds
- PASS - [flash_map.test_flash_area_disabled_device] duration = 0.001 seconds
- FAIL - [flash_map.test_flash_area_erase_and_flatten] duration = 4.744 seconds
- PASS - [flash_map.test_flash_area_erased_val] duration = 0.001 seconds
- PASS - [flash_map.test_flash_area_get_sectors] duration = 0.904 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: bf5ab1639defa729ed7ac2c5f1bffb02
PROJECT EXECUTION FAILED
```

Environment (please complete the following information):
OS: Linux
Zephyr version: v3.6.0-5742-g2efc447c1b41
Zephyr SDK version: 0.16.5-1

de-nordic, 2024-06-11:
@dleach02 seems to be my fault, taking over.

katgiadla, 2024-06-12:
The same issue occurred at `nrf52840dk/nrf52840` and `nrf52dk/nrf52832` (version `v3.6.0-5452-gec9f71e23794`).

```
===================================================================
START - test_flash_area_erase_and_flatten
Assertion failed at ../../../../../../../zephyr/tests/subsys/storage/flash_map/src/main.c:242: flash_map_test_flash_area_erase_and_flatten: (erased is false)
Erase failed at index 1
FAIL - test_flash_area_erase_and_flatten in 4.289 seconds
===================================================================
```

PR #74076 (2024-06-14): tests/storage/flash_map: Fix erase and flatten test
Fix offset read for verification and comparison of read to erased value, where type mismatch has been causing fails in comparisons.

Fixes #74066
commit a593f344 (2024-06-14): tests/storage/flash_map: Fix erase and flatten test

Fix offset read for verification and comparison of read to
erased value, where type mismatch has been causing fails in
comparisons.

Fixes #74066
closed 2024-06-14 by aescolar