zephyrproject-rtos/zephyr #74081: net_mgmt | net_if | WiFi AP+STA mode: Add support for WiFi AP+STA mode.

M-Hazik, 2024-06-11:
**Is your enhancement proposal related to a problem? Please describe.**
During working with the ESP32 on ZephyrOS, wifi driver does not allow for AP+STA mode to be active concurrently, presenting a significant limitation. After modifying ESP32 WiFi driver to enable both modes, we noticed that ZephyrOS uses only single **net_if** interface to manage IP addresses irrespective of the device's mode (STA or AP), thus further restricting AP+STA mode support in ZephyrOS. We also tried to work with virtual interfaces so we can assign one interface to AP and the other to STA but network(**net_mgmt**) requests which uses virtual interface gives error -134.

**Describe the solution you'd like**
Add support to active both(AP+STA) modes simultaneously by defining two interfaces for WiFi one for STA and one for AP.
### Logic For enabling ESP32 AP+STA mode:
-  If controller is working in  AP mode and NET_REQUEST_WIFI_CONNECT is requested but the controller in AP+STA mode and vice versa.
- Two **net_if** interfaces one for AP and other for STA mode.

**Describe alternatives you've considered**
- We tried to create a virtual interface and attached it to default interface of ESP32. When we tried to make network request using this virtual interface it return **error:-134**

M-Hazik, 2024-06-28:
This is how I was creating VIF

[image] 

It gave me error `-ENOTSUP`, For Referenc, screenshot is attached.

[image]

When I was looking for the reason I found  in`wifi_mgmt.c` out that it checks for the function pointer related to `net_mgmt` request. I might have not use VIF properly. can we pass `net_wifi_mgmt_offload` instead of `virtual_interface_api` to it?

From docs i found `NET_DEVICE_DT_INST_DEFINE_INSTANCE`, as per my understanding it creates multiple `netif` bound to the same network device. it did'nt work.  It give me following error. I might have not understand it's usage correctly.

[image].

The PR you mentioned will indeed be helpful for enabling AP+STA mode. because it separates the netif for `STA` and `softAP`. But still some changes are need to be done on driver sides and we are willing to make that changes. 

I would like to ask how would I define multiple netif instances and tell register it lower network layer as STA and AP netif?

krish2718, 2024-06-30:
So, I tried a quick hack using a downstream driver (I am most familiar with) to cook up multiple VIFs, I used See https://github.com/krish2718/zephyr/commit/a4be2ade0247e05cb1d2b402eb1312b94a0d13dd

1. Used `ETH_NET_DEVICE_DT_INST_DEFINE` to define multiple instances, the OPs are all same, just the per-interface data is different.
2. Modified DTS to define multiple `DT_DRV_COMPAT` instances `compatible = "nordic,wlan";`, `wlan0` and wlan1`

And it worked. For testing, I hacked Wi-Fi shell https://github.com/krish2718/zephyr/commit/934f7950597cca83a337e824669999d70a0d83a5 and able to scan on both the interfaces.

It's purely up to the driver to manage these interfaces i.e., whether they belong to same radio or different radio etc. Wi-Fi management is agnostic and just relies on `wifi_mgmt_api` (and WPA supplicant on `wifi_drv_ops`) all of these OPs are designed to be per-interface.

M-Hazik, 2024-07-04:
Thankyou, @krish2718. I tried your suggestions and it did work. I was able to create two net_if. 
After making some quick changes in wifi driver of esp32, I was able to turn on the `APSTA` mode of it.

Now, I'm trying to
1. Add enable DHCP server when softAP is enabled. So, it can assign IP to the clients.
2. Registering `net_if` as `softAP` and `STA`. 

I didn't figure out the way to register `net_if` with `wifi_nm`. To register it, the function needs the name of nm instance which I don't know how to get. 

Moreover, I was wondering why we need to define multiple nodes of `wifi` in .`overlay` to create multiple net_if? Can't wedefine multiple instance of `net_if` with single wifi node?

krish2718, 2024-07-04:
> I didn't figure out the way to register net_if with wifi_nm. To register it, the function needs the name of nm instance which I don't know how to get.

Please see https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/hostap/src/supp_main.c#L69 and https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/hostap/src/supp_main.c#L237 to get and use Wi-Fi NM instance.

> Just wondering why we need to define multiple nodes of wifi in .overlay to create multiple net_if? Can't we define multiple instance of net_if with single wifi node?

The macro `ETH_NET_DEVICE_DT_INST_DEFINE` needs multiple instances in DTS to map to the interfaces, it's a 1-1 mapping.

M-Hazik, 2024-07-09:
Thank you, @krish2718 . I was able to make AP+STA work on esp32. If you have'nt guide me. I might have struck. Here ( https://github.com/zin-HXQ/zephyr/tree/enabled-esp32-wifi-driver-both-mode )is the changes I made for esp32 wifi driver . it still needs little work to do and code cleanup. Thanks alot.

krish2718, 2024-07-18:
Can this be closed @M-Hazik ?

M-Hazik, 2024-07-22:
I have submitted MR: https://github.com/zephyrproject-rtos/zephyr/pull/76177 related to this issue. Can we wait for it to be closed?

zin-HXQ, 2024-10-28:
Hi, @krish2718. 

Maybe you can guide me. I am trying to run the same example code which worked for `esp32` to work for `nrf7002` driver but unable to do it successfully.

I tried https://github.com/krish2718/zephyr/commit/a4be2ade0247e05cb1d2b402eb1312b94a0d13dd changes for `nrf7002dk` by placing `.conf` and `.overlay` file in example `samples/boards/espressif/wifi_apsta_mode`. For reference, please, check https://github.com/zin-HXQ/zephyr/commit/4bf4f885b2c7442d1a4517f1a23e508f977192d6. 

- I get the following error when I tried to run the example.
[image]

-  When i try to debug the driver by settinG `CONFIG_WIFI_NRF70_LOG_LEVEL_DBG` , I get the following compile time error. Maybe, I'm missing some configurtion.
[image]

krish2718, 2024-10-29:
@M-Hazik are you sure you are using the latest code base? all below errors are resolved:

1. nordic_wlan0 -> nordic_wlan
2. Debug logs build errors

As for the WPA supplicant failure, you need to increase the FDs and pollable FDs, I would start with `CONFIG_POSIX_MAX_FDS`.

krish2718, 2024-12-03:
Related PR merged.

closed 2024-12-03 by krish2718

hanan-Zin, 2025-02-10:
@krish2718 
I am running the Shell app and trying to create two virtual interfaces on the nRF7002 EK.
I followed the changes you suggested https://github.com/krish2718/zephyr/commit/a4be2ade0247e05cb1d2b402eb1312b94a0d13dd ,  but I am encountering a runtime error.

[image]

I have attached a screenshot for reference. Any insights on what might be causing this?