zephyrproject-rtos/zephyr #74092: Twister reports invalid RAM/ROM size in twister.json when building with sysbuild and with flag `--footprint-from-buildlog`

RobertGalatNordic, 2024-06-11:
**Describe the bug**

When building sample application with command 
` west twister --testsuite-root <root to samples> --inline-logs --overflow-as-errors --enable-size-report --show-footprint --footprint-from-buildlog -vvv --build-only --subset <X/Y>`

The twister.json contains invalid values for RAM and ROM size and total size.
It seems like the buildlog parsing catches the mcuboot and reports it as application size. 
The results seem random, as most test cases are parsed correctly, and incorrectly parsed samples are not consistent
(in one run, the first sample may be OK, and in the next run the same sample will have size from mcuboot instead of app)

From my experience from 22 samples, around 6 will have different values in RAM/ROM compared to previous build (without code change) 

**To Reproduce**
pick a project with many build only samples
execute ` west twister --testsuite-root <root to samples> --inline-logs --overflow-as-errors --enable-size-report --show-footprint --footprint-from-buildlog -vvv --build-only`
twice and compare the generated twister.yml looking at the RAM/ROM size.

verify the results by repeating the test but without `--footprint-from-buildlog` flag.
This time the RAM/ROM reports in twister.yml are consistent.

**Expected behavior**
rerunning the twister command should result in the same RAM/ROM values when no code has been changed.

**Impact**
this affects automated tools for tracking the sample size.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: nrfConnect toolchain 2.6.0
 - Commit SHA or Version used

PerMac, 2024-06-19:
Thanks for the report. I had a chat with @tejlmand and it seems there is no easy fix for this. TL;DR build logs have memory statuses from all stages mixed up and nothing can be done with it. There is also no way to get clear markers that will allow the parser to reliably figure out which data comes from which image. This mean in case of builds with more than 1 image (sysbuild) --footprint-from-build log won't work.
Adding build logs from each image to their own folders might allow us to get the parser working correctly. I think it could also be a nice enhancement allowing to monitor sizes of different components. But it will be an additional task related to the build system.
@nashif The only solution I see for now is to add "This feature don't work with builds with more than one image" in the flag description in twister.

I am attaching a conversation with @tejlmand to provide a deeper context

<details>
  <summary>Conversation with @tejlmand</summary>
Perkowski, Maciej
This issue we already discussed a while ago. We made a script reading memory report from the build log, but it is always taking the last value. However, the order in which main app and others are build with sysbuild is random. So it makes not much sense. Back then, there were no clear to resolve it. I recall one couldn't affect the order without breaking the parallelism benefit and also adding some extra text than parser would understand to pick the main app and/or others was also not an option. Maybe it has changed?

- is it/can it be done to force the order of printing build logs, so the main is always first/last?
- is it possible to add some extra info to memory reports, stating which app it refers to?
 
Rasmussen, Torsten
`is it/can it be done to force the order of printing build logs, so the main is always first/last?`
if we want to do something like that, then it would require that one image strongly depends on another image, for example all images depends on main app, but as you also seem to be aware, then that would break parallelism .
But remember, each image has it's own sub-folder under the top-level build folder, like: `<build>/mcuboot`, `<build>/hello_world`, etc.
So a much better approach in my opinion would be to have each image build create its own independent build log.

`is it possible to add some extra info to memory reports, stating which app it refers to?` 
you can add print messages together with command execution, but as each print message is individual, then with concurrent running build commands you cannot guarantee that they don't happen at same time like this:
```
building / reporting foo
building / reporting bar
Mem-usage bar 
bar line 1
bar line 2
Mem-usage foo
foo line 1
foo line 2
```
each individual print will not be mixed, but because the mem report itself is printed by the linker itself (and thus all lines from there stays together), then any extra info we want must be printed individually, and thus there is a risk that this info message is interrupted as shown above.
Most of the time it will not be interrupted, simply because the size (and thus link time) of two images will in practice be different, but it cannot be guaranteed, and thus will fail in some cases.
  
</details>

RobertGalatNordic, 2024-06-20:
I found this issue because we used this flag to compare application size in our CI.
We overcome this bug by just using the default option in twister, which extracts flash and RAM usage from elf.

I understand that this issue is not as simple as it seems at the surface, and from my point of view adding a warning to this flag, is good enough for now. 

Eventually if it is decided that this flag should stay, the issue should be fixed, but for now I can live without it.

PR #74549 (2024-06-29): twister: Add warning to the "-footprint-from-buildlog" flag
The flag won't work with sysbuild since there is no way to reliably tell to a parser which data came from which image.

fixes: #74092
commit a033ba65 (2024-06-29): twister: Add warning to the "-footprint-from-buildlog" flag

The flag won't work with sysbuild since there is no way to
reliably tell to a parser which data came from which image.

fixes: #74092
closed 2024-06-29 by nashif