zephyrproject-rtos/zephyr #74142: twister does not understand HWMv2 board qualifiers for platform arguments

henrikbrixandersen, 2024-06-12:
**Describe the bug**
The `--platform` argument to twister does not understand HWMv2 board qualifiers.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b lpcxpresso55s36/lpc55s36 samples/hello_world/` (this works as expected)
2. `west twister -p lpcxpresso55s36/lpc55s36 -T samples/hello_world/` (this does not work)
3. See error

**Expected behavior**
Twister is able to understand board qualifiers.

**Impact**
Confusion, especially since our shell completions scripts completes the board qualifiers for `west twister -p ...` as well.

**Logs and console output**
```
$ west twister -p lpcxpresso55s36/lpc55s36 -T samples/hello_world/
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-5871-g97c3f87c82b4
INFO    - Using 'zephyr' toolchain.
ERROR   - platform_filter - unrecognized platform - lpcxpresso55s36/lpc55s36
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 2871c87aaec004cf4fe8e61a8d15c35b6d8ab677

henrikbrixandersen, 2024-06-26:
This falls back on Twister using the `scripts/schemas/twister/platform-schema.yaml` based files operating on "platforms" named from the `identifier` in those files, whereas West uses `scripts/schemas/board-schema.yml` operating on HWMv2 "board targets".

This is highly confusing and prevents running Twister on a given board variant _unless_ that board variant has a custom platform-schema file (e.g. `boards/native/native_sim/native_sim_native_64.yaml`).

nashif, 2024-06-26:
@henrikbrixandersen  i think i have  a way to deal with this, but it requires some major changes and already prototyping this. This is one of the issue we identified very early on in hwmv2 and plan to address. Not sure this can make it into 3.7 though, there are many inconsistencies of how boards are configured and that would require lots of harmonization before a general solution is applied.

henrikbrixandersen, 2024-06-27:
> @henrikbrixandersen i think i have a way to deal with this, but it requires some major changes and already prototyping this. This is one of the issue we identified very early on in hwmv2 and plan to [...]

Yeah, it will require some major changes. I've tried twice to solve it myself, but ran into unforeseen problems in both attempts.

I think it is unfortunate if this will not be fixed before v3.7.0 LTS, as it is confusing to users and makes the HWMv2 transition seem half-finished.

nashif, 2024-06-27:
i will spend some time on it sooner and see if I can address it for 3.7, agree this is an issue.

nashif, 2024-08-12:
note that I am actively looking into this now:
- will get list of boards from the build system (parsing board.yml files, etc.)
- looking into converting board yaml files used by twister (<target>.yaml) into one single file and removing duplicated data (vendor, ...). Instead of having a yaml file per board target/variant, we will have just one file.

Trying to keep things backward compatible as much as possible, but there will be some breaking changes, hopefully minimal.

tobiaskaestner, 2024-08-15:
@nashi, I am glad to help if you need some other pair of hands for testing

nashif, 2024-08-16:
> We have seen the same inconsistency downstream as well and there is a related issue #71761 that tried to partially solve it. Bottom line is that still we have diverging code bases in how west sees the workspace and how twister sees the same workspace. In the long run there is IMHO some effort needed to consolidate this into a common library that both west and twister can be based on. For now @rexut offered in #71761 some additional workarounds that did not get merged back then but would most likely help here.

@tobiaskaestner that is a different issue, probably need to reopen #71761 and adress this on its own.

PR #77250 (2024-10-22): twister: rework board enumeration

We now use hwmv2 to list boards instead of relying on twister specific
config files.

One yaml file (twister.yaml for now) will have all the data needed for
all possible targets and variations of a board reusing most of the data
where possible and variations can override the top level data.

Twister keeps track of 'aliases' of boards and identifies that for
example native_sim is the same as native_sim/native, so either names
will be possible in both test yaml files or on the command line,
however, the reporting will always use the full name, so no there is no [... cut]
closed 2024-10-22 by nashif