zephyrproject-rtos/zephyr #74158: doc: Documentation issue in 'samples/drivers/dac/README'

LysandrosAn, 2024-06-12:
**Describe the bug**

<< Hello the commands contained in the main.c file of this example, don’t seem to be documented anywhere...There are commands like DAC_RESOLUTION, DAC_RESOLUTION >> which are only mentioned in the example but not in the official documentation
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `samples/drivers/dac/README`
* Version: main
* SHA-1: 8dc3f856229ce083c956aa301c31a23e65bd8cd8

kartben, 2024-06-12:
Hi @LysandrosAn 

Can you elaborate on what you mean? The sample suggests to pass the DAC configuration via the `zephyr,user` node and then shows how properties of said node can be accessed via code and then passed to the `dac_channel_setup()` API. The API is documented in https://docs.zephyrproject.org/latest/hardware/peripherals/dac.html#api-reference

martinjaeger, 2024-06-18:
Hi @LysandrosAn,

the defines `DAC_NODE`, `DAC_CHANNEL_ID` and `DAC_RESOLUTION` are just used in the sample to make the code more easy to read. As said by @kartben, the defines are used to get data from devicetree to assign the device pointer and populate the `struct dac_channel_cfg`. They are not strictly needed, you can also set the values directly in your application.

@aescolar I'm going to remove the bug label, as this is more a question than a bug.