zephyrproject-rtos/zephyr #74205: mpfs_icicle: samples/drivers/flash_shell: Build failed in CI

erwango, 2024-06-13:
**Describe the bug**
Testing flah shell sample in CI fails on mpfs_icicle target
```
twister -p mpfs_icicle -T  samples/drivers/flash_shell/
```
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
CI blocker. See https://github.com/zephyrproject-rtos/zephyr/pull/74075

**Logs and console output**
```
zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:87:36: error: static assertion failed: "jedec,spi-nor jedec-id required for non-runtime SFDP"
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                    ^~~~~~~~~~~~~~

```

erwango, 2024-06-13:
@aescolar FYI this is blocking https://github.com/zephyrproject-rtos/zephyr/pull/74075 which is trying to fix another CI issue in https://github.com/zephyrproject-rtos/zephyr/pull/73837 which I'd need in V3.7.0

aescolar, 2024-06-13:
Bisected to adedf14c42594d167cb8f12855f7ae6097e285f1 @ashiroji

aescolar, 2024-06-13:
CC @de-nordic

aescolar, 2024-06-25:
Any update on this issue?

CC @VenkatKotakonda @albertofloyd

albertofloyd, 2024-06-25:
@aescolar @erwango how is this related to Microchip platforms and/or eSPI/PECI/KSCAN subsystems? 
Also I see https://github.com/zephyrproject-rtos/zephyr/pull/74075 and https://github.com/zephyrproject-rtos/zephyr/pull/73837 already solved, is this still valid? Or am I missing something?

albertofloyd, 2024-06-25:
I see this refers to Microchip PolarFire FPGA we are not using these boards. Only MEC152x/MEC172x Embedded controller evaluation boards. Need to check with original submitter and may need to nominate a maintainer

aescolar, 2024-06-26:
CC @con-pax

aescolar, 2024-06-26:
CC @HA-harshit @nagasureshmchp @petermcs-01

con-pax, 2024-06-26:
Hi @aescolar,
This is the first I have seen of this: I'm not sure why this test is being run against PolarFire SoC.

aescolar, 2024-06-26:
> Hi @aescolar, This is the first I have seen of this: I'm not sure why this test is being run against PolarFire SoC.

@con-pax All tests are run on all platforms unless they are filtered thru tags or thru kconfig or DT filters.

con-pax, 2024-06-26:
Hi @aescolar, 
So I tried to replicate the above 
using 

```bash
twister -p mpfs_icicle -T  samples/drivers/flash_shell/
```
except there are no tests associated with the target

```bash
Renaming output directory to /mnt/dev/zephyr_issue/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-rc1-422-g4a47803b0c06
INFO    - Using 'zephyr' toolchain.
Traceback (most recent call last):
  File "/mnt/dev/zephyr_issue/zephyr/scripts/twister", line 215, in <module>
    ret = main(options, default_options)
  File "/mnt/dev/zephyr_issue/zephyr/scripts/pylib/twister/twisterlib/twister_main.py", line 123, in main
    tplan.discover()
  File "/mnt/dev/zephyr_issue/zephyr/scripts/pylib/twister/twisterlib/testplan.py", line 185, in discover
    raise TwisterRuntimeError("No test cases found at the specified location...")
twisterlib.error.TwisterRuntimeError: No test cases found at the specified location...
```

aescolar, 2024-06-26:
> Hi @aescolar, So I tried to replicate the above using
> 
Be sure to run that command from ZEPHYR_BASE (the folder samples/drivers/flash_shell/ needs to be there) or otherwise run it as 

twister -p mpfs_icicle -T  ${ZEPHYR_BASE}/samples/drivers/flash_shell/

con-pax, 2024-06-26:
@aescolar Thanks, I can replicate this now:

```bash
twister -p mpfs_icicle -T ${ZEPHYR_BASE}/samples/drivers/flash_shell/
Renaming output directory to /mnt/dev/zephyr_issue/twister-out.2
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-rc1-422-g4a47803b0c06
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /mnt/dev/zephyr_issue/twister-out/testplan.json
INFO    - JOBS: 32
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - mpfs_icicle               samples/drivers/flash_shell/sample.drivers.flash.shell  ERROR : Build failure
ERROR   - see: /mnt/dev/zephyr_issue/twister-out/mpfs_icicle/samples/drivers/flash_shell/sample.drivers.flash.shell/build.log
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0, error:    1
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 0 failed, 1 errored, 0 skipped with 0 warnings in 8.25 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 2 platforms (50.00%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /mnt/dev/zephyr_issue/twister-out/twister.json
INFO    - Writing xunit report /mnt/dev/zephyr_issue/twister-out/twister.xml...
INFO    - Writing xunit report /mnt/dev/zephyr_issue/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples/drivers/flash_shell/sample.drivers.flash.shell on mpfs_icicle error (Build failure)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p mpfs_icicle -s samples/drivers/flash_shell/sample.drivers.flash.shell
INFO    - or with west:
INFO    - west build -p -b mpfs_icicle samples/drivers/flash_shell -T sample.drivers.flash.shell
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

To allow for things to progress in a timely manor for the release, I can propose a simple "fix" i.e, use a tag for flash:

```yaml
identifier: mpfs_icicle
name: Microchip PolarFire ICICLE kit
type: mcu
arch: riscv
toolchain:
  - zephyr
ram: 3840
testing:
  ignore_tags:
    - net
    - bluetooth
    - flash
vendor: microchip
```

With the above, the CI is skipped:

```bash
twister -p mpfs_icicle -T ${ZEPHYR_BASE}/samples/drivers/flash_shell/
Renaming output directory to /mnt/dev/zephyr_issue/twister-out.3
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-rc1-422-g4a47803b0c06
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /mnt/dev/zephyr_issue/twister-out/testplan.json
INFO    - JOBS: 32
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

INFO    - 1 test scenarios (1 test instances) selected, 1 configurations skipped (1 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 0 failed, 0 errored, 1 skipped with 0 warnings in 0.47 seconds
INFO    - In total 0 test cases were executed, 1 skipped on 1 out of total 2 platforms (50.00%)
INFO    - 0 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /mnt/dev/zephyr_issue/twister-out/twister.json
INFO    - Writing xunit report /mnt/dev/zephyr_issue/twister-out/twister.xml...
INFO    - Writing xunit report /mnt/dev/zephyr_issue/twister-out/twister_report.xml...
INFO    - Run completed
```

con-pax, 2024-06-26:
@aescolar Yep, I will do it now

con-pax, 2024-06-26:
@aescolar done: https://github.com/zephyrproject-rtos/zephyr/pull/75046

aescolar, 2024-06-26:
@nashif This board fails to build with flash with the same error for any sample:
```
cmake -GNinja -DBOARD=mpfs_icicle ../samples/hello_world/ -DCONFIG_FLASH=y
``` 
So I think this change in its yaml is appropriate, and would not blame the sample

PR #75046 (2024-06-27): boards: microchip: skip flash based CI tests for PolarFire SoC
Microchip's PolarFire SoC has a system controller that utilizes an external spi flash for performing "auto update" of the FPGA design. This is not a feauture fully integrated into Zephyr as of yet, so in an effort to not constantly fail the CI, add flash tag to the ignore list. 

Fixes  #74205
closed 2024-06-27 by aescolar