zephyrproject-rtos/zephyr #74212: apollo4p_evb: samples/drivers/flash_shell/sample.drivers.flash.shell   Build failed in CI

FRASTM, 2024-06-13:
building the samples/drivers/flash_shell for the apollo4p_evb target fails :

twister -p mpfs_icicle -T  samples/drivers/flash_shell/
or
wb -p auto -b apollo4p_evb samples/drivers/flash_shell/

SHA1 : d18fd46e991134ea29c8b52a52f2e4960327dfa4

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Impact**
CI failure blocking several PR

**Logs and console output**
```
./build/zephyr/include/generated/zephyr/devicetree_generated.h:7039:34: error: 'DT_N_S_soc_S_flash_controller_18000_S_flash_18000_P_write_block_size' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_flash_controller_18000_S_flash_18000_P_reg_IDX_1'?
 7039 | #define DT_N_INST_0_soc_nv_flash DT_N_S_soc_S_flash_controller_18000_S_flash_18000

```
**Environment (please complete the following information):**
 - OS: Ubuntu)
 - Toolchain 0.16.0
 - Commit SHA d18fd46e991134ea29c8b52a52f2e4960327dfa4

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2024-06-13:
It looks like write-block-size and erase-block-size properties are missing for the flash-controller
```
&flash0 {
	write-block-size = <16>;
	erase-block-size = <8192>;
};
```

aescolar, 2024-06-13:
Issue introduced by 008efecf5dddd1d54fb0726c4db5c3447bad5308 
@RichardSWheatley

aescolar, 2024-06-13:
Full log:
```
./build/zephyr/include/generated/devicetree_generated.h:6453:34: error: 'DT_N_S_soc_S_flash_controller_18000_S_flash_18000_P_write_block_size' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_flash_controller_18000_S_flash_18000_P_reg_IDX_1'?
 6453 | #define DT_N_INST_0_soc_nv_flash DT_N_S_soc_S_flash_controller_18000_S_flash_18000
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/zephyr/toolchain/gcc.h:87:51: note: in definition of macro 'BUILD_ASSERT'
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                                   ^~~~
./drivers/flash/flash_ambiq.c:22:32: note: in expansion of macro 'MAX'
   22 | #define FLASH_WRITE_BLOCK_SIZE MAX(DT_PROP(SOC_NV_FLASH_NODE, write_block_size), MIN_WRITE_SIZE)
      |                                ^~~
./include/zephyr/devicetree.h:636:32: note: in expansion of macro 'DT_CAT3'
  636 | #define DT_PROP(node_id, prop) DT_CAT3(node_id, _P_, prop)
      |                                ^~~~~~~
./drivers/flash/flash_ambiq.c:22:36: note: in expansion of macro 'DT_PROP'
   22 | #define FLASH_WRITE_BLOCK_SIZE MAX(DT_PROP(SOC_NV_FLASH_NODE, write_block_size), MIN_WRITE_SIZE)
      |                                    ^~~~~~~
./include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_N_INST_0_soc_nv_flash'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
./include/zephyr/sys/util_internal.h:104:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  104 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
./include/zephyr/devicetree.h:336:31: note: in expansion of macro 'UTIL_CAT'
  336 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
./drivers/flash/flash_ambiq.c:18:32: note: in expansion of macro 'DT_INST'
   18 | #define SOC_NV_FLASH_NODE      DT_INST(0, soc_nv_flash)
      |                                ^~~~~~~
./drivers/flash/flash_ambiq.c:22:44: note: in expansion of macro 'SOC_NV_FLASH_NODE'
   22 | #define FLASH_WRITE_BLOCK_SIZE MAX(DT_PROP(SOC_NV_FLASH_NODE, write_block_size), MIN_WRITE_SIZE)
      |                                            ^~~~~~~~~~~~~~~~~
./drivers/flash/flash_ambiq.c:25:15: note: in expansion of macro 'FLASH_WRITE_BLOCK_SIZE'
   25 | BUILD_ASSERT((FLASH_WRITE_BLOCK_SIZE & (MIN_WRITE_SIZE - 1)) == 0,
      |               ^~~~~~~~~~~~~~~~~~~~~~
./drivers/flash/flash_ambiq.c:25:14: error: expression in static assertion is not an integer
   25 | BUILD_ASSERT((FLASH_WRITE_BLOCK_SIZE & (MIN_WRITE_SIZE - 1)) == 0,
      |              ^
./include/zephyr/toolchain/gcc.h:87:51: note: in definition of macro 'BUILD_ASSERT'
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                                   ^~~~
./drivers/flash/flash_ambiq.c: In function 'flash_ambiq_write':
./drivers/flash/flash_ambiq.c:81:18: warning: unused variable 'aligned' [-Wunused-variable]
   81 |         uint32_t aligned[FLASH_WRITE_BLOCK_SIZE / sizeof(uint32_t)] = {0};
      |                  ^~~~~~~
In file included from ./include/zephyr/arch/arm/arch.h:20,
                 from ./include/zephyr/arch/cpu.h:19,
                 from ./include/zephyr/kernel_includes.h:36:
./drivers/flash/flash_ambiq.c: At top level:
./build/zephyr/include/generated/devicetree_generated.h:6453:34: error: 'DT_N_S_soc_S_flash_controller_18000_S_flash_18000_P_erase_block_size' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_flash_controller_18000_S_flash_18000_P_reg_IDX_1'?
 6453 | #define DT_N_INST_0_soc_nv_flash DT_N_S_soc_S_flash_controller_18000_S_flash_18000
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/zephyr/devicetree.h:4534:29: note: in definition of macro 'DT_CAT3'
 4534 | #define DT_CAT3(a1, a2, a3) a1 ## a2 ## a3
      |                             ^~
./drivers/flash/flash_ambiq.c:23:32: note: in expansion of macro 'DT_PROP'
   23 | #define FLASH_ERASE_BLOCK_SIZE DT_PROP(SOC_NV_FLASH_NODE, erase_block_size)
      |                                ^~~~~~~
./include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_N_INST_0_soc_nv_flash'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
./include/zephyr/sys/util_internal.h:104:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  104 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
./include/zephyr/devicetree.h:336:31: note: in expansion of macro 'UTIL_CAT'
  336 | #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
      |                               ^~~~~~~~
./drivers/flash/flash_ambiq.c:18:32: note: in expansion of macro 'DT_INST'
   18 | #define SOC_NV_FLASH_NODE      DT_INST(0, soc_nv_flash)
      |                                ^~~~~~~
./drivers/flash/flash_ambiq.c:23:40: note: in expansion of macro 'SOC_NV_FLASH_NODE'
   23 | #define FLASH_ERASE_BLOCK_SIZE DT_PROP(SOC_NV_FLASH_NODE, erase_block_size)
      |                                        ^~~~~~~~~~~~~~~~~
./drivers/flash/flash_ambiq.c:157:44: note: in expansion of macro 'FLASH_ERASE_BLOCK_SIZE'
  157 |         .pages_count = SOC_NV_FLASH_SIZE / FLASH_ERASE_BLOCK_SIZE,
      |                                            ^~~~~~~~~~~~~~~~~~~~~~
```

aescolar, 2024-06-13:
Lowering priority after the board has been disabled in CI

PR #74275 (2024-06-25): boards: ambiq: apollo4p_evb: fix multiple CI failure
This PR is trying to fix multiple CI failures which were introduced from https://github.com/zephyrproject-rtos/zephyr/pull/70842.
Adds the missing flash controller node in apollo4p_evb to address the CI failure https://github.com/zephyrproject-rtos/zephyr/issues/74212. 
Remove the untestable cache kconfig from apollo4x boards to address the CI failure https://github.com/zephyrproject-rtos/zephyr/issues/72775, https://github.com/zephyrproject-rtos/zephyr/issues/73443 at this moment.
And enable the twister for apollo4x platforms. Refer to http [... cut]
closed 2024-06-25 by nashif