zephyrproject-rtos/zephyr #74246: ESP32-S3 WiFi does not work if external SPIRAM is enabled

EricNRS, 2024-06-13:
**Describe the bug**
If SPIRAM is enabled, WiFi does not work on ESP32S3 WROOM N16R8.

WiFi Works if SPIRAM is disabled:
`west build --pristine -b xiao_esp32s3/esp32s3/procpu samples/net/wifi -DCONFIG_ESP_SPIRAM=n && west flash && west espressif monitor`

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048df9
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc94170,len:0x3804
load:0x40374000,len:0x1014c

SHA-256 comparison failed:
Calculated: 8923cad87525e34488aa27e9abb59530d31d7a56f3c68686442b1c27b70b23be
Expected: 0000000070c60000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x4037cdd8

I (94) boot: ESP Simple boot
I (94) boot: compile time Jun 14 2024 01:20:43
W (95) boot: Unicore bootloader
I (95) spi_flash: detected chip: gd
I (96) spi_flash: flash io: dio
W (99) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (111) boot: chip revision: v0.2
I (114) boot.esp32s3: Boot SPI Speed : 40MHz
I (118) boot.esp32s3: SPI Mode       : DIO
I (122) boot.esp32s3: SPI Flash Size : 8MB
I (126) boot: Enabling RNG early entropy source...
[esp32s3] [INF] DRAM: lma 0x00000020 vma 0x3fc94170 len 0x3804   (14340)
[esp32s3] [INF] IRAM: lma 0x0000382c vma 0x40374000 len 0x1014c  (65868)

[esp32s3] [INF] padd: lma 0x00013988 vma 0x00000000 len 0xc670   (50800)
[esp32s3] [INF] IMAP: lma 0x00020000 vma 0x42000000 len 0x53260  (340576)
0x42000000: _stext at ??:?

[esp32s3] [INF] padd: lma 0x00073268 vma 0x00000000 len 0xcd90   (52624)
[esp32s3] [INF] DMAP: lma 0x00080000 vma 0x3c060000 len 0x131ec  (78316)
[esp32s3] [INF] Image with 6 segments
[esp32s3] [INF] DROM segment: paddr=00080000h, vaddr=3c060000h, size=131F0h ( 78320) map
[esp32s3] [INF] IROM segment: paddr=00020000h, vaddr=42000000h, size=5325Eh (340574) map

*** Booting Zephyr OS build v3.6.0-5736-gd78a9429450f ***
uart:~$ wifi scan
Scan requested

Num  | SSID                             (len) | Chan (Band)   | RSSI | Security        | BSSID             | MFP     
1    | IoT                              3     | 1    (2.4GHz) | -67  | WPA2-PSK        |                   | Disable 
2    | IoT                              3     | 11   (2.4GHz) | -74  | WPA2-PSK        |                   | Disable 
3    | IoT                              3     | 6    (2.4GHz) | -87  | WPA2-PSK        |                   | Disable 
Scan request done
uart:~$ 
```

If SPIRAM is enabled, then WiFi returns nothing for `wifi scan` and will also not connect.  I only have octal SPI devices, so I am unable to determine if this is just an octal SPI issue or it it applies to quad SPI as well.

`west build --pristine -b xiao_esp32s3/esp32s3/procpu samples/net/wifi -DCONFIG_ESP_SPIRAM=y -DCONFIG_SPIRAM_MODE_OCT=y && west flash && west espressif monitor`

```
I (97) boot: ESP Simple boot
I (97) boot: compile time Jun 14 2024 01:13:09
W (97) boot: Unicore bootloader
I (97) spi_flash: detected chip: gd
I (98) spi_flash: flash io: dio
W (101) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (114) boot: chip revision: v0.2
I (117) boot.esp32s3: Boot SPI Speed : 40MHz
I (121) boot.esp32s3: SPI Mode       : DIO
I (124) boot.esp32s3: SPI Flash Size : 8MB
I (128) boot: Enabling RNG early entropy source...
[esp32s3] [INF] DRAM: lma 0x00000020 vma 0x3fc94a30 len 0x3b1c   (15132)
[esp32s3] [INF] IRAM: lma 0x00003b44 vma 0x40374000 len 0x10a10  (68112)
[esp32s3] [INF] padd: lma 0x00014568 vma 0x00000000 len 0xba90   (47760)
[esp32s3] [INF] IMAP: lma 0x00020000 vma 0x42000000 len 0x53598  (341400)
[esp32s3] [INF] padd: lma 0x000735a0 vma 0x00000000 len 0xca58   (51800)
[esp32s3] [INF] DMAP: lma 0x00080000 vma 0x3c060000 len 0x1337c  (78716)
[esp32s3] [INF] Image with 6 segments
[esp32s3] [INF] DROM segment: paddr=00080000h, vaddr=3c060000h, size=13380h ( 78720) map
[esp32s3] [INF] IROM segment: paddr=00020000h, vaddr=42000000h, size=53596h (341398) map
I (202) octal_psram: vendor id    : 0x0d (AP)
I (202) octal_psram: dev id       : 0x02 (generation 3)
I (202) octal_psram: density      : 0x03 (64 Mbit)
I (204) octal_psram: good-die     : 0x01 (Pass)
I (209) octal_psram: Latency      : 0x01 (Fixed)
I (213) octal_psram: VCC          : 0x01 (3V)
I (217) octal_psram: SRF          : 0x01 (Fast Refresh)
I (222) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (227) octal_psram: BurstLen     : 0x01 (32 Byte)
I (231) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (237) octal_psram: DriveStrength: 0x00 (1/1)
I (241) esp_psram: Found 8MB PSRAM device
I (244) esp_psram: Speed: 40MHz
I (978) esp_psram: SPI SRAM memory test OK

*** Booting Zephyr OS build v3.6.0-5736-gd78a9429450f ***
uart:~$ wifi scan
Scan requested
*** SHOULD BE A LIST OF ACCESS POINTS HERE ***
Scan request done
uart:~$ 
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use an ESP32S3 WROOM N16R8
2. Build with SPIRAM enabled:
`west build --pristine -b xiao_esp32s3/esp32s3/procpu samples/net/wifi -DCONFIG_ESP_SPIRAM=y -DCONFIG_SPIRAM_MODE_OCT=y && west flash && west espressif monitor`
3. Do a "wifi scan" command
4. See that no access points are found.  In addition, WiFi will not connect to any known access points

**Expected behavior**
WiFi scan finds access points and allows connecting.

**Impact**
Complete failure of WiFi

**Environment (please complete the following information):**
Zephyr 3.6.99 main @ d78a9429450f89de7829c8af7efe41fdb692fd50

EricNRS, 2024-06-14:
Doing a bit more debugging, the issue is that the ESP32-S3 WiFi stack does not support buffers allocated in SPIRAM and the minimum heap size of 51200 is too small.

```
*** Booting Zephyr OS build v3.6.0-5736-gd78a9429450f ***
uart:~$ kernel heap
free:           12
allocated:      50904
max. allocated: 50904
```

Increasing the heap size to 100KB allows the scan to work and shows that the maximum heap usage is 83400 bytes.

```
uart:~$ kernel heap 
free:           21068
allocated:      80892
max. allocated: 83400
```

The `HEAP_MEM_POOL_ADD_SIZE_BOARD` should be increased to 86016 or similar to allow WiFi to work.

However, this increased memory requirement makes the ESP32-S3 target very difficult to use since the internal SRAM is already in short supply with just this example already using 73% of the available memory.  Is there any chance that the ESP32-S3 WiFi stack could be fixed to support using buffers from SPIRAM?  That has been a long-term issue.

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      603244 B    8388352 B      7.19%
     iram0_0_seg:       68148 B       329 KB     20.23%
     dram0_0_seg:      234376 B       313 KB     73.13%
     irom0_0_seg:      341706 B         8 MB      4.07%
     drom0_0_seg:      472172 B         8 MB      5.63%
     ext_ram_seg:       1536 KB    8388544 B     18.75%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
```

EricNRS, 2024-06-14:
@sylvioalves

vigen1980, 2024-08-03:
Experiencing a similar issue - given the SRAM is quite limited, it'd be great to be able to utilize the SPIRAM for most memory allocations (performance is not [yet] an issue).

As I was experimenting with ```CONFIG_ESP_SPIRAM=y```, I noticed that WiFi scanning and connection works when ```CONFIG_ESP_HEAP_MIN_EXTRAM_THRESHOLD``` is set to ```1608```, that is only allocations above 1608 bytes are served from the SPIRAM. Logging various allocations, I noticed a set of allocations during startup as wifi was initialized of 1604 bytes which fits very well with the threshold number above. The call stack originates in one of the espressif blobs for which I wasn't able to find the source code:
```
#0  __wrap_k_malloc (size=1604) at modules/hal/espressif/zephyr/port/heap/heap_caps_zephyr.c:348
#1  0x40376e5c in wifi_malloc (size=1604)
    at modules/hal/espressif/zephyr/esp32s3/src/wifi/esp_wifi_adapter.c:69
#2  0x40376e9c in malloc_internal_wrapper (size=1604)
    at modules/hal/espressif/zephyr/esp32s3/src/wifi/esp_wifi_adapter.c:524
#3  0x42078c71 in wDev_Rxbuf_Init ()
#4  0x420734a0 in ic_init ()
#5  0x420630d1 in wifi_lmac_init ()
#6  0x4206317a in wifi_init_process ()
#7  0x420618e1 in ieee80211_ioctl_process ()
#8  0x40388278 in ppTask ()
#9  0x42009b84 in z_thread_entry (entry=0x403881d4 <ppTask>, p1=0x0, p2=0x0, p3=0x0)
    at zephyr/lib/os/thread_entry.c:48
```

After wifi connection is established, my app tries to connect to Azure IoT hub. It appears that to the threshold must be raised to 1620 bytes to have a successful connection (otherwise -116 error is reported by mqtt_connect). The callstack of the allocation still points to wifi stack, but I wasn't able to get any additional symbols:
```
#1  0x4200f543 in __wrap_k_malloc (size=1617)
    at modules/hal/espressif/zephyr/port/heap/heap_caps_zephyr.c:351
#2  0x40376e5c in wifi_malloc (size=1617)
    at modules/hal/espressif/zephyr/esp32s3/src/wifi/esp_wifi_adapter.c:69
#3  0x40376e9c in malloc_internal_wrapper (size=1617)
    at modules/hal/espressif/zephyr/esp32s3/src/wifi/esp_wifi_adapter.c:524
#4  0x4003ec98 in ?? ()
#5  0x4003ee9e in ?? ()
#6  0x400398b8 in ?? ()
#7  0x400399c4 in ?? ()
#8  0x40388290 in ppTask ()
#9  0x42009b84 in z_thread_entry (entry=0x403881d4 <ppTask>, p1=0x0, p2=0x0, p3=0x0)
    at zephyr/lib/os/thread_entry.c:48
```

With the setting above, I ended up using only 24964 bytes of SRAM. Raising the threshold to 1704 bytes (there are bunch of allocations of 1700 bytes in wifi stack) ends up consuming 79404 bytes of SRAM. But I'd like to keep the threshold at 1024 bytes to maximize hardware capabilities.

Hope this helps find the bug!

EricNRS, 2024-08-04:
@vigen1980 - Similar to your experiments, I finally determined that the `._malloc_internal` needs to allocated internal RAM (hence the name which seems obvious in hindsight) and `.wifi_malloc` can use SPIRAM.  I created two buffer pools for the WiFi and that used 26KB of internal RAM and 85KB of SPIRAM.

I'm really busy for a next week or so, but could share these patches if there is a chance they would be merged.  This can be a configuration option to use the internal buffer pools or continue using the heap as is currently done.

vigen1980, 2024-08-04:
@EricNRS Great find! I updated the impl of ```*alloc_internal_wrapper``` functions in esp_wifi_adapter.c to use ```k_aligned_alloc``` from ```zephyr/kernel/mempool.c``` and reduced the ```CONFIG_ESP_HEAP_MIN_EXTRAM_THRESHOLD``` to ```1024``` bytes, and got WiFi scanning and IoT Hub connection working with only 24452 bytes used from SRAM.

Here's what I've got so far:
```
static void *IRAM_ATTR malloc_internal_wrapper(size_t size)
{
  return k_aligned_alloc(sizeof(void*), size);
}

static void *IRAM_ATTR calloc_internal_wrapper(size_t n, size_t size)
{
  size_t bounds;
  if (__builtin_mul_overflow(n, size, &bounds)) {
    return NULL;
  }

  void* ret = k_aligned_alloc(sizeof(void*), size);
  if (ret != NULL) {
    (void)memset(ret, 0, bounds);
  }

  return ret;
}

static void *IRAM_ATTR zalloc_internal_wrapper(size_t size)
{
  return calloc_internal_wrapper(1, size);
}

```
@EricNRS Do you have something similar as well?

With all that - this solution does not seem like a bug fix, but more avoidance of the bug. I would love to still learn  why use of SPIRAM causes issues in WiFi stack.

EricNRS, 2024-08-04:
@vigen1980 - I used two different heaps and tracked which heap the allocated block came from since the same free call is used regardless of the heap allocation.  I have uploaded the code here if you want to give it a try:  https://github.com/zephyrproject-rtos/hal_espressif/compare/zephyr...EricNRS:hal_espressif:2024-08-04-wifi-spiram-heap

> this solution does not seem like a bug fix, but more avoidance of the bug. I would love to still learn why use of SPIRAM causes issues in WiFi stack.

It may just be a hardware requirement of the WiFi firmware.  Ideally, the driver should fail an assert or otherwise warn that it needed an internal buffer but got an SPIRAM change which it could do with pointer comparisons.

real-tintin, 2024-08-07:
I have the same issue on a ESP32-WROOM-32E-N8R2 (Zephyr 3.7.0), can't scan/connect when SPIRAM is enabled.

Increasing CONFIG_ESP_HEAP_MIN_EXTRAM_THRESHOLD to 16xx bytes also work for me. It doesn't seem like CONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM is either affecting the memory regions?

As you said, I would be great if WiFi could utilize the SPIRAM more.

@EricNRS I will try your fix!

marekmatej, 2024-09-06:
@EricNRS, @vigen1980  can you please try to disable the `CONFIG_FEATURE_CACHE_TX_BUF_BIT` flag from the `g_wifi_feature_caps` (in the `esp_wifi_adapter.c`) and give it a try? In my case, allocating those extra TX buffers renders the wifi to not working.

EricNRS, 2024-09-06:
@marekmatej  - are you saying that all of the heap memory could be in SPIRAM with CONFIG_FEATURE_CACHE_TX_BUF_BIT removed?  I will give it a try when I get a chance.

My changes in https://github.com/zephyrproject-rtos/hal_espressif/compare/zephyr...EricNRS:hal_espressif:2024-08-04-wifi-spiram-heap allocate using internal RAM for all `*_internal_wrapper()` functions.  It works well.  Sorry, there are some other debug changes in the patch.  With that change, I was able to reduce internal DRAM usage to 32KB and provide a generous 120KB from SPIRAM.  WiFi has been working well for months with these changes.

marekmatej, 2024-09-06:
> @marekmatej - are you saying that all of the heap memory could be in SPIRAM with CONFIG_FEATURE_CACHE_TX_BUF_BIT removed? I will give it a try when I get a chance.

No, I meant to say that enabling this flag will ask for additional dozens of kB of memory - it is automatically enabled when CONFIG_ESP_SPIRAM is enabled.

EricNRS, 2024-09-06:
> No, I meant to say that enabling this flag will ask for additional dozens of kB of memory - it is automatically enabled when CONFIG_ESP_SPIRAM is enabled.

Ah, nice.  With it disabled, it doesn't change the amount of internal RAM used, but does reduce the SPIRAM usage.  I'm not concerned about SPIRAM usage since I have over 8 MB of that.

```
# With CONFIG_FEATURE_CACHE_TX_BUF_BIT enabled (default) in g_wifi_feature_caps
uart> esp32_wifi heap_stats 
SPIRAM Heap: used 58 KB of 120 KB; max allocated 75 KB
DRAM Heap: used 20 KB of 32 KB; max allocated 21 KB

# With CONFIG_FEATURE_CACHE_TX_BUF_BIT removed from g_wifi_feature_caps
uart> esp32_wifi heap_stats
SPIRAM Heap: used 5 KB of 120 KB; max allocated 19 KB  <-- reduced
DRAM Heap: used 20 KB of 32 KB; max allocated 21 KB  <-- DRAM still the same
```

EricNRS, 2024-11-07:
I think this issue has been resolved as part of other changes, but it needs a retest and hopefully a reference to the fix(es).

sylvioalves, 2024-11-07:
> I think this issue has been resolved as part of other changes, but it needs a retest and hopefully a reference to the fix(es).

It is not fully fixed yet.

real-tintin, 2024-11-07:
@sylvioalves does this mean that its coming soon? :)

EricNRS, 2025-01-07:
Bumping to remove the stale flag.  @sylvioalves  - you mentioned there were still some changes to fully fix this.  Any update on the status?

marekmatej, 2025-01-07:
@EricNRS thanks for bumping the issue and your patience. I checked the following builds and both work well on my end. Can you update and try again please?

1. `west build -p -b esp32s3_devkitm/esp32s3/procpu samples/net/wifi/shell/ -DCONFIG_ESP_SPIRAM=y`

2. `west build -p -b esp32s3_devkitm/esp32s3/procpu samples/net/wifi/shell/ -DCONFIG_ESP_SPIRAM=y -DCONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y`

EricNRS, 2025-01-07:
They both work for me (I used `xiao_esp32s3/esp32s3/procpu` since that is the hardware that I have on hand), so I'm happy to mark this as fixed.

```
west build -p -b xiao_esp32s3/esp32s3/procpu samples/net/wifi/shell/ -DCONFIG_ESP_SPIRAM=y -DCONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=y

Memory region         Used Size  Region Size  %age Used
           FLASH:      615060 B    8388352 B      7.33%
     iram0_0_seg:       76288 B     343552 B     22.21%
     dram0_0_seg:      131688 B     327168 B     40.25% <-- 6% less DRAM usage with ALLOC_SPIRAM enabled
     irom0_0_seg:      348006 B         8 MB      4.15%
     drom0_0_seg:      483988 B         8 MB      5.77%
    ext_dram_seg:     1594128 B         8 MB     19.00%
    ext_iram_seg:          0 GB         8 MB      0.00%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%

```

```
west build -p -b xiao_esp32s3/esp32s3/procpu samples/net/wifi/shell/ -DCONFIG_ESP_SPIRAM=y -DCONFIG_ESP32_WIFI_NET_ALLOC_SPIRAM=n

Memory region         Used Size  Region Size  %age Used
           FLASH:      615060 B    8388352 B      7.33%
     iram0_0_seg:       76288 B     343552 B     22.21%
     dram0_0_seg:      152952 B     327168 B     46.75%
     irom0_0_seg:      348006 B         8 MB      4.15%
     drom0_0_seg:      483988 B         8 MB      5.77%
    ext_dram_seg:       1536 KB         8 MB     18.75%
    ext_iram_seg:          0 GB         8 MB      0.00%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
```

closed 2025-01-07 by marekmatej