zephyrproject-rtos/zephyr #74277: ESP32-S3 WiFi Triggers ASSERT Upon Connect

EricNRS, 2024-06-14:
**Describe the bug**
If asserts are enabled globally (`CONFIG_ASSERT=y`), then connecting to a WiFi access point will result in a failed assertion.  With asserts disabled, the connection is successful.

**To Reproduce**
Build and run the example on an ESP32-S3.  No SPIRAM is used here, so this should apply to all targets.

`west build --pristine -b xiao_esp32s3/esp32s3/procpu samples/net/wifi -DCONFIG_ASSERT=y -DCONFIG_HEAP_MEM_POOL_SIZE=85000 && west flash && west espressif monitor
`
Connect to WiFi and notice that the system crashes:
```
*** Booting Zephyr OS build v3.6.0-5736-gd78a9429450f ***
uart:~$ wifi connect -s "SSID" -p PASSWORD -k 1
Connection requested
[00:02:04.100,000] <err> os:  **  PC 0x40378c4f VADDR 0
0x40378c4f: xtensa_arch_except_epc at /home/work/zephyrproject-3.6/zephyr/arch/xtensa/core/xtensa_asm2_util.S:209

[00:02:04.101,000] <err> os:  ** A12 0x3fc9f640 A13 0x4 A14 0 A15 0x3fc9f720
[00:02:04.103,000] <err> os:  ** A12 0x3fc9f428 A13 0x4 A14 0x3fc9f438 A15 0xc
[00:02:04.105,000] <err> os:  ** A12 0x3fc9f230 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.107,000] <err> os:  ** A12 0x3fc9f038 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.109,000] <err> os:  ** A12 0x3fc9ee40 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.111,000] <err> os:  ** A12 0x3fc9ec48 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.113,000] <err> os:  ** A12 0x3fc9ea50 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.115,000] <err> os:  ** A12 0x3fc9e858 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.117,000] <err> os:  ** A12 0x3fc9e660 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.119,000] <err> os:  ** A12 0x3fc9e468 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.121,000] <err> os:  ** A12 0x3fc9e270 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.123,000] <err> os:  ** A12 0x3fc9e078 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.125,000] <err> os:  ** A12 0x3fc9de80 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.127,000] <err> os:  ** A12 0x3fc9dc88 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.129,000] <err> os:  ** A12 0x3fc9da90 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.131,000] <err> os:  ** A12 0x3fc9d898 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.133,000] <err> os:  ** A12 0x3fc9d6a0 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.135,000] <err> os:  ** A12 0x3fc9d4a8 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.137,000] <err> os:  ** A12 0x3fc9d2b0 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.139,000] <err> os:  ** A12 0x3fc9d0b8 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.141,000] <err> os:  ** A12 0x3fc9cec0 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.143,000] <err> os:  ** A12 0x3fc9ccc8 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.145,000] <err> os:  ** A12 0x3fc9cad0 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.147,000] <err> os:  ** A12 0x3fc9c8d8 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.149,000] <err> os:  ** A12 0x3fc9c6e0 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.151,000] <err> os:  ** A12 0x3fc9c4e8 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.153,000] <err> os:  ** A12 0x3fc9c2f0 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.155,000] <err> os:  ** A12 0x3fc9c0f8 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.157,000] <err> os:  ** A12 0x3fc9bf00 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.159,000] <err> os:  ** A12 0x3fc9bd08 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.161,000] <err> os:  ** A12 0x3fc9bb10 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.163,000] <err> os:  ** A12 0x3fc9b918 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.165,000] <err> os:  ** A12 0x3fc9b720 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.167,000] <err> os:  ** A12 0x3fc9b528 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.169,000] <err> os:  ** A12 0x3fc9b330 A13 0x4 A14 0xff0000 A15 0xff000000
[00:02:04.171,000] <err> > > > > > > > > > > > > > > > :  ** A12 0x3fc9b138 A13 0x4 A14 0xff0000 A15 0xff000000
```

With verbose asserts enabled and debug code enabled:
`ASSERTION FAIL [arch_irq_unlocked(key) || _kernel.cpus[0].current->base.thread_state & (((1UL << (0))) | ((1UL << (3))))] @ WEST_TOPDIR/zephyr/kernel/include/kswap.h:99`

Here is the `do_swap()` code that triggers the assertion.

```
static ALWAYS_INLINE unsigned int do_swap(unsigned int key,
					  struct k_spinlock *lock,
					  bool is_spinlock)
{
	ARG_UNUSED(lock);
	struct k_thread *new_thread, *old_thread;

#ifdef CONFIG_SPIN_VALIDATE
	/* Make sure the key acts to unmask interrupts, if it doesn't,
	 * then we are context switching out of a nested lock
	 * (i.e. breaking the lock of someone up the stack) which is
	 * forbidden!  The sole exception are dummy threads used
	 * during initialization (where we start with interrupts
	 * masked and switch away to begin scheduling) and the case of
	 * a dead current thread that was just aborted (where the
	 * damage was already done by the abort anyway).
	 *
	 * (Note that this is disabled on ARM64, where system calls
	 * can sometimes run with interrupts masked in ways that don't
	 * represent lock state.  See #35307)
	 */
# ifndef CONFIG_ARM64
	__ASSERT(arch_irq_unlocked(key) ||
		 _current->base.thread_state & (_THREAD_DUMMY | _THREAD_DEAD),
		 "Context switching while holding lock!");
# endif /* CONFIG_ARM64 */
#endif /* CONFIG_SPIN_VALIDATE */
```

The same test without the assert works:
`west build --pristine -b xiao_esp32s3/esp32s3/procpu samples/net/wifi -DCONFIG_HEAP_MEM_POOL_SIZE=85000 && west flash && west espressif monitor
`

**Expected behavior**
Connecting to WiFi should be possible with debug ASSERTs enabled.

**Impact**
This is a showstopper for me as I need asserts enabled during development and cannot release software for testing with known lock corruption.

**Environment (please complete the following information):**
Zephyr 3.6.99 at commit d78a9429450f89de7829c8af7efe41fdb692fd50

EricNRS, 2024-06-14:
Adding a breakpoint to the failed assertion returns this backtrace:

```
Thread 7 "wifi" received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1070403216]
gdb) bt
#0  do_swap (is_spinlock=true, lock=0x3fcd2c2c <_sched_spinlock>, key=396579)
    at /home/work/zephyrproject-3.6/zephyr/kernel/include/kswap.h:101
#1  z_swap (lock=0x3fcd2c2c <_sched_spinlock>, key=...) at /home/work/zephyrproject-3.6/zephyr/kernel/include/kswap.h:192
#2  0x4037bcb9 in z_pend_curr (lock=0x3fca136c <kheap_wifi_heap+2796>, key=..., wait_q=0x3fca1364 <kheap_wifi_heap+2788>, timeout=...)
    at /home/work/zephyrproject-3.6/zephyr/kernel/sched.c:696
#3  0x4037a100 in z_impl_k_msgq_get (msgq=0x3fca1364 <kheap_wifi_heap+2788>, data=0x3fc9c640 <wifi_stack+3520>, timeout=...)
    at /home/work/zephyrproject-3.6/zephyr/kernel/msg_q.c:272
#4  0x42009b74 in k_msgq_get (timeout=..., data=<optimized out>, msgq=<optimized out>)
    at /home/work/zephyrproject-3.6/zephyr/build/zephyr/include/generated/zephyr/syscalls/kernel.h:1207
#5  queue_recv_wrapper (queue=0x3fca1364 <kheap_wifi_heap+2788>, item=0x3fc9c640 <wifi_stack+3520>, block_time_tick=4294967295)
    at /home/work/zephyrproject-3.6/modules/hal/espressif/zephyr/esp32s3/src/wifi/esp_wifi_adapter.c:416
#6  0x40385794 in ppTask ()
#7  0x42003640 in z_thread_entry (entry=0x40385768 <ppTask>, p1=0x0, p2=0x0, p3=0x0)
    at /home/work/zephyrproject-3.6/zephyr/lib/os/thread_entry.c:48
(gdb) 
```

EricNRS, 2024-06-14:
@sylvioalves

EricNRS, 2024-06-20:
Just ran into another assert in the I2C code which may be related to this.  In this case, the assertion for `arch_is_in_isr()` failed since the timeout option is a compile-time constant of 500ms.

Line being executed that fails:
`	ret = k_sem_take(&data->cmd_sem, K_MSEC(I2C_TRANSFER_TIMEOUT_MSEC));`

Backtrace:
```
ASSERTION FAIL [((arch_is_in_isr() == 0) || ((timeout).ticks == (((k_timeout_t) {0})).ticks))] @ WEST_TOPDIR/zephyr/kernel/sem.c:136

Line 136 of "/home/work/zephyrproject-3.6/zephyr/kernel/sem.c" starts at address 0x4037f0d9 <z_impl_k_sem_take+21>
   and ends at 0x4037f108 <z_impl_k_sem_take+68>.
Line 1103 of "/home/work/zephyrproject-3.6/zephyr/build/zephyr/include/generated/zephyr/syscalls/kernel.h"
   starts at address 0x40378ce3 <i2c_esp32_transmit+59> and ends at 0x40378ce9 <i2c_esp32_transmit+65>.
Line 453 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x40378e08 <i2c_esp32_write_addr+212>
   and ends at 0x40378e0b <i2c_esp32_write_addr+215>.
Line 516 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x40378ffb <i2c_esp32_read_msg+39>
   and ends at 0x40379004 <i2c_esp32_read_msg+48>.
Line 690 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x4037942f <i2c_esp32_transfer+243>
   and ends at 0x4037943a <i2c_esp32_transfer+254>.
Line 788 of "/home/work/zephyrproject-3.6/zephyr/include/zephyr/drivers/i2c.h" starts at address 0x4207a012 <i2c_write_read+50>
   and ends at 0x4207a015 <i2c_write_read+53>.
Line 1387 of "/home/work/zephyrproject-3.6/zephyr/include/zephyr/drivers/i2c.h" starts at address 0x4207a04b <i2c_burst_read_dt+31>
   and ends at 0x4207a051 <i2c_burst_read_dt+37>.
Line 31 of "/home/work/zephyrproject-3.6/zephyr/drivers/sensor/ti/ina23x/ina23x_common.c"
   starts at address 0x4207a0c8 <ina23x_reg_read_16+16> and ends at 0x4207a0d4 <ina23x_reg_read_16+28>.
Line 150 of "/home/work/zephyrproject-3.6/zephyr/drivers/sensor/ti/ina23x/ina237.c" starts at address 0x4207aae2 <ina237_read_data+46>
   and ends at 0x4207aaed <ina237_read_data+57>.
Line 218 of "/home/work/zephyrproject-3.6/zephyr/drivers/sensor/ti/ina23x/ina237.c"
   starts at address 0x4207ae2c <ina237_sample_fetch+96> and ends at 0x4207ae33 <ina237_sample_fetch+103>.
. . .
```

EricNRS, 2024-06-24:
Just had another I2C failure due to `__ASSERT(arch_is_in_isr() == false)` failing.  I was able to capture a TRAX trace from this and going through each line, I didn't see any place where locks were not being properly released and the code was not being called in ISR context.

Percepio Tracealyzer shows that the `esp_timer` was active just before this sequence, so this looks to be the same WiFi issue where interrupts are getting disabled in the WiFi blob `libpp.a` or somewhere else in the sequence.

```
i2c_esp32_read_msg()
 -> i2c_esp32_master_read()
     -> i2c_esp32_transmit()

Line 337 of "/home/work/zephyrproject-3.6/zephyr/arch/xtensa/core/vector_handlers.c" starts at address 0x4037c81c <xtensa_int1_c+20> and ends at 0x4037c81f <xtensa_int1_c+23>.
Line 337 of "/home/work/zephyrproject-3.6/zephyr/arch/xtensa/core/vector_handlers.c" starts at address 0x4037c825 <xtensa_int1_c+29> and ends at 0x4037c838 <xtensa_int1_c+48>.
Line 121 of "/home/work/zephyrproject-3.6/zephyr/soc/espressif/common/include/_soc_inthandlers.h" starts at address 0x4037c538 <_xtensa_handle_one_int1> and ends at 0x4037c53b <_xtensa_handle_one_int1+3>.
Line 167 of "/home/work/zephyrproject-3.6/zephyr/soc/espressif/common/include/_soc_inthandlers.h" starts at address 0x4037c57c <_xtensa_handle_one_int1+68> and ends at 0x4037c57f <_xtensa_handle_one_int1+71>.
Line 192 of "/home/work/zephyrproject-3.6/zephyr/soc/espressif/common/include/_soc_inthandlers.h" starts at address 0x4037c5a0 <_xtensa_handle_one_int1+104> and ends at 0x4037c5a3 <_xtensa_handle_one_int1+107>.
Line 204 of "/home/work/zephyrproject-3.6/zephyr/soc/espressif/common/include/_soc_inthandlers.h" starts at address 0x4037c5b1 <_xtensa_handle_one_int1+121> and ends at 0x4037c5b4 <_xtensa_handle_one_int1+124>.
Line 211 of "/home/work/zephyrproject-3.6/zephyr/soc/espressif/common/include/_soc_inthandlers.h" starts at address 0x4037c5ba <_xtensa_handle_one_int1+130> and ends at 0x4037c5bc <_xtensa_handle_one_int1+132>.
Line 219 of "/home/work/zephyrproject-3.6/zephyr/soc/espressif/common/include/_soc_inthandlers.h" starts at address 0x4037c62b <_xtensa_handle_one_int1+243> and ends at 0x4037c631 <_xtensa_handle_one_int1+249>.
Line 706 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x40379854 <i2c_esp32_isr> and ends at 0x40379857 <i2c_esp32_isr+3>.
Line 713 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x40379880 <i2c_esp32_isr+44> and ends at 0x40379886 <i2c_esp32_isr+50>.
Line 33 of "/home/work/zephyrproject-3.6/modules/hal/espressif/components/hal/i2c_hal_iram.c" starts at address 0x4038c67c <i2c_hal_master_handle_rx_event> and ends at 0x4038c67f <i2c_hal_master_handle_rx_event+3>.
Line 808 of "/home/work/zephyrproject-3.6/modules/hal/espressif/zephyr/esp32s3/../../components/hal/esp32s3/include/hal/i2c_ll.h" starts at address 0x4038c6b0 <i2c_hal_master_handle_rx_event+52> and ends at 0x4038c6b5 <i2c_hal_master_handle_rx_event+57>.
Line 810 of "/home/work/zephyrproject-3.6/modules/hal/espressif/zephyr/esp32s3/../../components/hal/esp32s3/include/hal/i2c_ll.h" starts at address 0x4038c6c0 <i2c_hal_master_handle_rx_event+68> and ends at 0x4038c6c5 <i2c_hal_master_handle_rx_event+73>.
Line 812 of "/home/work/zephyrproject-3.6/modules/hal/espressif/zephyr/esp32s3/../../components/hal/esp32s3/include/hal/i2c_ll.h" starts at address 0x4038c6d0 <i2c_hal_master_handle_rx_event+84> and ends at 0x4038c6d5 <i2c_hal_master_handle_rx_event+89>.
Line 38 of "/home/work/zephyrproject-3.6/modules/hal/espressif/components/hal/i2c_hal_iram.c" starts at address 0x4038c6f2 <i2c_hal_master_handle_rx_event+118> and ends at 0x4038c6f4 <i2c_hal_master_handle_rx_event+120>.
Line 42 of "/home/work/zephyrproject-3.6/modules/hal/espressif/components/hal/i2c_hal_iram.c" starts at address 0x4038c72c <i2c_hal_master_handle_rx_event+176> and ends at 0x4038c72f <i2c_hal_master_handle_rx_event+179>.
Line 46 of "/home/work/zephyrproject-3.6/modules/hal/espressif/components/hal/i2c_hal_iram.c" starts at address 0x4038c737 <i2c_hal_master_handle_rx_event+187> and ends at 0x4038c74c <wdt_hal_init>.
Line 717 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x40379891 <i2c_esp32_isr+61> and ends at 0x40379893 <i2c_esp32_isr+63>.
Line 719 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x403798a0 <i2c_esp32_isr+76> and ends at 0x403798a3 <i2c_esp32_isr+79>.
Line 721 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x403798ad <i2c_esp32_isr+89> and ends at 0x403798b0 <i2c_esp32_isr+92>.
Line 723 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x403798bc <i2c_esp32_isr+104> and ends at 0x403798bf <i2c_esp32_isr+107>.
Line 727 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x403798c6 <i2c_esp32_isr+114> and ends at 0x403798c9 <i2c_esp32_isr+117>.
Line 97 of "/home/work/zephyrproject-3.6/zephyr/kernel/sem.c" starts at address 0x4037f850 <z_impl_k_sem_give> and ends at 0x4037f853 <z_impl_k_sem_give+3>.
Line 10 of "/home/work/zephyrproject-3.6/zephyr/kernel/spinlock_validate.c" starts at address 0x403836a4 <z_spin_lock_valid> and ends at 0x403836a7 <z_spin_lock_valid+3>.
Line 18 of "/home/work/zephyrproject-3.6/zephyr/kernel/spinlock_validate.c" starts at address 0x403836c9 <z_spin_lock_valid+37> and ends at 0x403836cb <z_spin_lock_valid+39>.
[... cut ...]
# 	__ASSERT(((arch_is_in_isr() == false) || K_TIMEOUT_EQ(timeout, K_NO_WAIT)), "");

Line 373 of "/home/work/zephyrproject-3.6/zephyr/drivers/i2c/i2c_esp32.c" starts at address 0x40378fc4 <i2c_ll_write_cmd_reg> and ends at 0x40378fc7 <i2c_esp32_transmit+3>.
Line 133 of "/home/work/zephyrproject-3.6/zephyr/kernel/sem.c" starts at address 0x4037f8f0 <z_impl_k_sem_take> and ends at 0x4037f8f3 <z_impl_k_sem_take+3>.
Line 48 of "/home/work/zephyrproject-3.6/zephyr/lib/os/assert.c" starts at address 0x4201c4a8 <assert_print> and ends at 0x4201c4ab <assert_print+3>.
```

EricNRS, 2024-06-24:
@sylvioalves Can you or someone on your team comment on whether the code in `libpp.a` disables interrupts anywhere -- especially when calling the `queue_recv_wrapper()`?  It seems odd that it would do that since it has function calls such as `_wifi_int_disable()`, `_wifi_int_restore()`, `_ints_on()`, and `_ints_off()` for that and logging is showing that those functions are being called correctly.

sylvioalves, 2024-06-24:
@EricNRS I am able to reproduce it here and have been checking how to solve this. However, I can't reply to your question above (yet).

EricNRS, 2024-06-24:
Awesome, thanks @sylvioalves, I sincerely appreciate the help.  I have been going through the TRM and code to determine exactly how the interrupts work to see if I can find any clues such as maybe the `arch_irq_lock()` `XCHAL_EXCM_LEVEL` isn't masking an interrupt correctly or something similar.

Let me know if you need me to run any tests.

EricNRS, 2024-06-25:
@sylvioalves -  I found the issue!

I added the following code to grab the PS.INTLEVEL to all of the interface functions in `modules/hal/espressif/zephyr/esp32s3/src/wifi/esp_wifi_adapter.c` and found that the cause of the interrupts being incorrectly enabled is actually the call to `timer_disarm_wrapper()` which we do have code for!

```
/**
 * Return the PS.INTLEVEL which is active interrupt mask level.
 * 
 * 0 means all interrupts enabled, 3 is XCHAL_EXCM_LEVEL and means
 * interrupts disabled. 
 */
static unsigned is_lock_held(void)
{
	unsigned int key;

	__asm__ volatile("rsr.ps %0" : "=r"(key) :: "memory");

	// interrupt level is the lower 4-bits of PS
	return key & 0x0F;
}

// Added the before and after checks to all functions
static void IRAM_ATTR timer_disarm_wrapper(void *timer)
{
	unsigned before = is_lock_held();
	ets_timer_disarm(timer);
	unsigned after = is_lock_held();

	if (before != after) {
		__asm__ volatile("break	1, 4");
	}
}
```

That identified the location of the issue and stepping through the debugger finally revealed the issue.  Here is the call sequence:

Turns out the issue is the irq_lock() key is getting overwritten because timer_list_lock() is being called twice.
```
timer_disarm_wrapper()
 -> esp_timer_stop()
      -> timer_list_lock()
      -> timer_remove()
            -> timer_list_lock()   <=== this overwrites the PS key returned by irq_lock()
            . . .
      -> timer_list_unlock()  <== wrong interrupt key restored
```

I just removed the lock and ran tests for an hour and it is working great.  I will cleanup the code and upload for review.

EricNRS, 2024-06-25:
Fix PR:  https://github.com/zephyrproject-rtos/hal_espressif/pull/290

EricNRS, 2024-06-25:
It looks like some of the timer functionality isn't used, so I wonder if the entire `esp_timer.c` task could be replaced by Zephyr timers.  That would save 4KB of RAM used for the timer stack.  May be something to consider as a lower-priority task once things are stable.

PR #74986 (2024-06-27): west.yml: update hal_espressif for bugfixes
This update includes bugfixes related to esp_timer, responsible for handling Wi-Fi thread events.
Also removes all constructor attributes from non-used hal functions.

Fixes #74368
Fixes #74277
Fixes #75063
commit c029fa5b (2024-06-27): west.yml: update hal_espressif for bugfixes

This update includes bugfixes related to esp_timer,
responsible for handling Wi-Fi thread events.
This also removes all constructor attributes from non-used
hal functions.

Fixes #74368
Fixes #74277
Fixes #75063
closed 2024-06-27 by nashif