zephyrproject-rtos/zephyr #74284: ESP32-S3 Linker Source Snippets does not work for SPIRAM Relocations

EricNRS, 2024-06-14:
**Describe the bug**
It is not possible to relocate data to the SPIRAM using the CMake linker snippets such as:

```
# in app CMakeLists.txt
zephyr_linker_sources(RAM_SECTIONS spiram_noinit_relocations.ld)

# File:  spiram_noinit_relocations.ld
SECTION_PROLOGUE(spiram_noinit_relocations, (NOLOAD), ALIGN(4))
{
    *net_pkt.c.obj(.noinit .noinit.*)
} > ext_ram_seg
```

The reason for this is that the SPI Flash (DROM) and SPI RAM (DRAM) both share a 32MB address space and different combinations of SPI Flash and SPIRAM can be fitted.  In Zephyr 3.6 with the EspressIf HAL v5.1, the linker descriptor `soc/espressif/esp32s3/default.ld` maps both of these sections to the DROM_SEG_ORG and then later in the file, the address of the ` .ext_ram.dummy` dummy is shifted to the end of the SPI Flash contents.

This means that anything defined in `snippets-ram-sections.ld` will included before the ` .ext_ram.dummy` section and will therefore be located at DROM_SEG_ORG instead of DROM_SEG_ORG + sizeof(Flash Data).

Here is a summary of the `soc/espressif/esp32s3/default.ld` file:

```
  drom0_0_seg(R):  org = DROM_SEG_ORG, len = DROM_SEG_LEN
#if defined(CONFIG_ESP_SPIRAM)
  ext_ram_seg(RWX): org = DROM_SEG_ORG, len = CONFIG_ESP_SPIRAM_SIZE - 0x40
#endif
. . . 
  #include <snippets-ram-sections.ld>
. . .
  /* this will capture any remaining .noinit items, so "snippets-ram-sections.ld" must be included before this section.
  .dram0.noinit (NOLOAD):
  {
    . = ALIGN(4);
    *(.noinit)
    *(.noinit.*)
    . = ALIGN(4) ;
  } GROUP_LINK_IN(RAMABLE_REGION)
. . .
#if defined(CONFIG_ESP_SPIRAM)

  .ext_ram.dummy (NOLOAD):
  {
    . = ORIGIN(ext_ram_seg) + (_rodata_reserved_end - _flash_rodata_dummy_start);
    . = ALIGN (CACHE_ALIGN);
  } GROUP_LINK_IN(ext_ram_seg)

  /* This section holds .ext_ram.bss data, and will be put in PSRAM */
  .ext_ram.bss (NOLOAD) :
  {
    _ext_ram_bss_start = ABSOLUTE(.);
    *(.ext_ram.bss*)
    . = ALIGN(4);

    _spiram_heap_start = ABSOLUTE(.);
    . = . + CONFIG_ESP_SPIRAM_HEAP_SIZE - (_spiram_heap_start - _ext_ram_bss_start);
    . = ALIGN(4);

    _ext_ram_bss_end = ABSOLUTE(.);
  } GROUP_LINK_IN(ext_ram_seg)

#endif /* CONFIG_ESP_SPIRAM */
```

I am not aware of any other way to force the linker to match the items later.  The only solution I can think of is to go back to the solution used in Zephyr 3.5 which placed the SPIRAM at the end of the segment.

See the following discussions for additional details:
- https://discord.com/channels/720317445772017664/1250370791313248286
- https://discord.com/channels/720317445772017664/883444902971727882/1250695683338666089

**To Reproduce**
In application CMakeLists.txt:
`zephyr_linker_sources(RAM_SECTIONS spiram_relocations.ld)`

In spiram_relocations.ld:
```
SECTION_PROLOGUE(spiram_noinit, (NOLOAD), ALIGN(4))
{
    *net_pkt.c.obj(.noinit .noinit.*)
} > ext_ram_seg
```

Build the image and filter based upon net_pkt objects:
`west build -t ram_report | grep -C20 net_pkt.c`

This results in the spiram_noinit section getting placed at the beginning of the ext_ram_seg, when it needs to be placed after .ext_ram.dummy.

```
├── net_pkt.c                          25940   0.31%  - 
│   ├── _k_mem_slab_buf_rx_pkts         2048   0.02%  0x3c005c60 spiram_noinit
│   ├── _k_mem_slab_buf_tx_pkts         2048   0.02%  0x3c005460 spiram_noinit
│   ├── _net_buf_rx_bufs                 224   0.00%  0x3c005380 spiram_noinit
│   ├── _net_buf_tx_bufs                 896   0.01%  0x3c004000 spiram_noinit
│   ├── log_dynamic_net_pkt                4   0.00%  0x3fca87e8 log_dynamic_area
│   ├── net_buf_data_rx_bufs            4096   0.05%  0x3c004380 spiram_noinit
│   ├── net_buf_data_tx_bufs           16384   0.20%  0x3c000000 spiram_noinit  <-- located at beginning which is SPI Flash
│   ├── net_buf_fixed_alloc_rx_bufs       12   0.00%  0x3c0d51a0 .ext_ram.dummy
│   ├── net_buf_fixed_alloc_tx_bufs       12   0.00%  0x3c0d5190 .ext_ram.dummy
│   ├── net_buf_fixed_rx_bufs              4   0.00%  0x3c0d51ac .ext_ram.dummy
│   ├── net_buf_fixed_tx_bufs              4   0.00%  0x3c0d519c .ext_ram.dummy
│   ├── rx_bufs                           68   0.00%  0x3fca8d9c net_buf_pool_area
│   ├── rx_pkts                           36   0.00%  0x3fca8920 k_mem_slab_area
│   ├── tx_bufs                           68   0.00%  0x3fca8de0 net_buf_pool_area
```

**Impact**
This is a major annoyance as a workaround is to go through the Zephyr kernel and manually add `__attribute__ ((section (".ext_ram.noinit")))` and `__attribute__ ((section (".ext_ram.bss")))`.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
Zephyr v3.6.99 at commit d78a9429450.

EricNRS, 2024-08-14:
@sylvioalves - any chance we can also get this issue resolved in the current cleanup efforts?

EricNRS, 2024-09-06:
@marekmatej - thanks, I will take a look at that.  This particular issue is for for general relocation, I just used the network stacks as an example.  I was able to successfully move about 40KB of the WiFi heap to SPIRAM and added the code to https://github.com/zephyrproject-rtos/zephyr/issues/74246#issuecomment-2267228781.

EricNRS, 2024-11-06:
*Bump* to keep this open so the bot doesn't think it has been abandoned.

KevinJordil, 2024-11-18:
Is anyone currently working on a fix?

EricNRS, 2025-01-07:
I'm not aware of any active work on this, but the EspressIf team looks to be over the HAL upgrade hump, so hopefully some of the issues like this one will be next.

marekmatej, 2025-01-07:
@EricNRS I've investigated the issue, and the fix for `.ext_ram_noinit` segment is still pending.
However, you can use the `EXT_RAM_BSS_ATTR` macro which will place your variable into ".ext_ram.bss" segment in the SPIRAM.
Also, you can use the `shm_psram` and allocate your variable on the SPIRAM heap.

brandon-exact, 2025-04-03:
can we reopen this issue? I do not believe the linked PR fixed the reported issue.

EricNRS, 2025-04-05:
> can we reopen this issue? I do not believe the linked PR fixed the reported issue.

Correct, the code snippets are still not working correctly and I am not sure if they can be made to work correctly given linker script limitations (as described in my original description).

The Code and Data Relocation (https://docs.zephyrproject.org/latest/kernel/code-relocation.html) seems to work on a file-level as a quick test shows this works for relocating any BSS data to external RAM.

main.c
```c
static uint8_t big_buffer[KB(512)];

int main(void)
{
    // avoid code removal
    big_buffer[0] = 0xAA;
}
```

CMakeLists.txt
```
zephyr_code_relocate(FILES main.c LOCATION ext_ram_seg_BSS)
```

I think the relocation script should be extended to allow moving specific variables to external memory since there are cases where some variables must remaining in internal memory.

We need some good examples of this feature in the ESP32-S* documentation due to ongoing memory limitations on the target.