zephyrproject-rtos/zephyr #74296: tests/posix/common/ failing to build for riscv

aescolar, 2024-06-14:
**Describe the bug**
tests/posix/common/ failing to build for m2gl025_miv

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p m2gl025_miv -T tests/posix/common -s portability.posix.common.static_stack
4. See error

**Expected behavior**
No build errors

**Impact**
main CI failing

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/9510842407/job/26216040417#step:11:1441

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: main after b98a60773b3f2a592b9a1dea4899ec26ca2ab4fb

**Additional context**

aescolar, 2024-06-14:
CC @fkokosinski @kgugala @tgorochowik 
@mgielda @katsuster @edersondisouza @carlocaione @npitre

ycsin, 2024-06-14:
Fix is in #74282

PR #74282 (2024-06-14): arch: riscv: stacktrace: conditionally check stack_info
Check if an address is in the thread stack only when `CONFIG_THREAD_STACK_INFO` is enabled, since otherwise the `stack_info` will not be available.

This fixes compilation error when `CONFIG_THREAD_STACK_INFO` is explicitly disabled.

fixes #74296
closed 2024-06-14 by nashif