zephyrproject-rtos/zephyr #74313: Bluetooth: Sysbuild broken on nrf5340

alwa-nordic, 2024-06-14:
Bluetooth sample peripheral_hr fails `bt_enable` on nrf53 with sysbuild.

```
<wrn> bt_hci_core: opcode 0x0c33 status 0x12
Bluetooth init failed (err -22)
```

Cause: Value differs between images: `CONFIG_BT_ACL_RX_SIZE`. The `0x12` error code is returned from the controller when the host size is smaller. This is true for both ZLL and SDC.

Why are the values different?

Sysbuild overrides the config for the net core, but not for the app core. The app core uses the default value.

https://github.com/zephyrproject-rtos/zephyr/blob/d45605e6a3dfe68e43207d15e3bfc003cbfaa72f/samples/bluetooth/peripheral_hr/sysbuild.cmake#L16-L19

https://github.com/zephyrproject-rtos/zephyr/blob/9ae5352372dd9965805980e0c712612f21ed2229/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf#L26

Not caught in tests because: There is a workaround applied specifically for bsim53 instead of fixing the issue.

https://github.com/zephyrproject-rtos/zephyr/blob/9ae5352372dd9965805980e0c712612f21ed2229/samples/bluetooth/peripheral_hr/boards/nrf5340bsim_nrf5340_cpuapp.conf#L1-L3

Expected fix: Propagate `CONFIG_BT_ACL_RX_SIZE` from the app to net core.

[image]

carlescufi, 2024-07-10:
@nordicjm could you take a look?

nordicjm, 2024-07-10:
Not a sysbuild issue, user issue. You would need to set the value in both images, sysbuild will not do that for you.

aescolar, 2024-07-22:
@alwa-nordic given that you are looking into it, and that we need to assign issues to somebody, if you don't mind let's leave it a assigned to you by now, unless you have a better assignee in mind.