zephyrproject-rtos/zephyr #74326: Introduce BLE controller-only firmware to make STEVAL-MKBOXPRO board working

avisconti, 2024-06-14:
## Origin

Import following binary blob:
- ble-fw/steval-mkboxpro/dtm.bin

It is a BLE controller-only firmware image imported from:
https://github.com/STMicroelectronics/stsw-mkbox-bleco

## Type

- [ ] Precompiled library
- [x] Firmware image

## Module

hal_stm32: https://github.com/zephyrproject-rtos/hal_stm32/.

## Purpose

This BLE f/w image (controller-only configuration) is required to upgrade a standard STM STEVAL-MKBOXPRO board in order to work with zephyr bluetooth stack.

## Pull Request

Zephyr: #74255
hal_stm32: https://github.com/zephyrproject-rtos/hal_stm32/pull/217

## Dependencies

none

## License

Blobs are licensed using STMicroeletronics "ULTIMATE LIBERTY SOFTWARE LICENSE". License text is available [here](https://github.com/STMicroelectronics/STM32CubeWBA/blob/main/Middlewares/ST/STM32_WPAN/LICENSE.md) and provided in hal_stm32 PR.

avisconti, 2024-08-05:
@carlescufi do I need to do anything else for this blob request?

erwango, 2024-08-05:
Blob introduction has been approved on Zephyr Project TSC - 10 July 2024.

avisconti, 2024-08-08:
> Blob introduction has been approved on Zephyr Project TSC - 10 July 2024.

Yes, I think I can close it down as completed.

closed 2024-08-08 by avisconti