zephyrproject-rtos/zephyr #74329: Attempting to use MQTT with ESP32 fails

ASerbinski, 2024-06-14:
**Describe the bug**

I'm trying to adapt samples/net/cloud/aws_iot_mqtt/ to;
a) esp32/wifi
b) local mosquitto server rather than aws.

The device in question is a generic ESP32 dev board utilizing an ESP32-WROVER-B module. I'm using it as esp32_devkitc_wrover, which is close enough.

Its not working, while the esp32 is able to connect with the mosquitto server, it errors out with;

Zephyr console/ESP32:
```
[00:00:10.197,000] <err> aws: Failed to connect: -22 backoff delay: 394 ms
[00:00:10.606,000] <err> aws: Failed to connect: -22 backoff delay: 1536 ms
[00:00:12.160,000] <err> aws: Failed to connect: -22 backoff delay: 2698 ms
[00:00:14.873,000] <err> aws: Failed to connect: -22 backoff delay: 7140 ms
[00:00:22.033,000] <err> aws: Failed to connect: -22 backoff delay: 15403 ms
```

Mosquitto:
```
1718380307: Client connection from 192.168.1.137 failed: error:0A000126:SSL routines::unexpected eof while reading.
1718380308: Client connection from 192.168.1.137 failed: error:0A000126:SSL routines::unexpected eof while reading.
1718380309: Client connection from 192.168.1.137 failed: error:0A000126:SSL routines::unexpected eof while reading.
1718380312: Client connection from 192.168.1.137 failed: error:0A000126:SSL routines::unexpected eof while reading.
1718380319: Client connection from 192.168.1.137 failed: error:0A000126:SSL routines::unexpected eof while reading.
```

**To Reproduce**
I've attached my patch for the sample (its renamed .txt since github didn't like it when it was .patch).

- Assumes mosquitto is running on IP address 192.168.1.223, adjust as needed in prj.conf
- SSID/PSK need to be set on lines 43/44 of src/main.c
- From mosquitto/, start mosquitto using "mosquitto -c mosquitto.conf"
- Run mosquitto_pub to validate credentials and ssl as "mosquitto_pub -h localhost -u user -P password --cafile ca.crt --cert client@host.crt --key client@host.key -t 'topic/value' -m true"
- Note output from mosquitto (server) indicates success:
```
1718379782: New connection from ::1:54836 on port 8883.
1718379782: New client connected from ::1:54836 as auto-0146A82B-918C-0B83-7EBE-3060A3A7F1BB (p2, c1, k60, u'user').
1718379782: No will message specified.
1718379782: Sending CONNACK to auto-0146A82B-918C-0B83-7EBE-3060A3A7F1BB (0, 0)
1718379782: Received PUBLISH from auto-0146A82B-918C-0B83-7EBE-3060A3A7F1BB (d0, q0, r0, m0, 'topic/value', ... (4 bytes))
1718379782: Received DISCONNECT from auto-0146A82B-918C-0B83-7EBE-3060A3A7F1BB
1718379782: Client auto-0146A82B-918C-0B83-7EBE-3060A3A7F1BB disconnected.
```

- The same cafile, cert, and key as used for mosquitto_pub are also in src/creds/, and convert_keys.py has been run.
- Build using "west build -b esp32_devkitc_wrover samples/net/cloud/aws_iot_mqtt" and flash.
- Output is as listed above.

**Expected behavior**
Obviously its expected to connect without error.

**Impact**
showstopper

[mqtt_esp32.txt](https://github.com/user-attachments/files/15840420/mqtt_esp32.txt)

rlubos, 2024-06-18:
> [00:00:10.197,000] <err> aws: Failed to connect: -22 backoff delay: 394 ms

Have you tried to pinpoint the exact reason for getting `EINVAL` from `mqtt_connect()`? My bet is it's mbed TLS misconfiguration, i. e. some mbed TLS config needed to handle credentials you provide is not set (some cipher or hash algorithm missing or something like that). But with the information provided, it's just a guess.

rlubos, 2024-06-18:
@nashif I took the liberty and lowered the priority, it's most likely just some misconfiguration.

ASerbinski, 2024-06-18:
> > [00:00:10.197,000]  aws: Failed to connect: -22 backoff delay: 394 ms
> 
> Have you tried to pinpoint the exact reason for getting `EINVAL` from `mqtt_connect()`?

Its a seemingly ambiguous error. Do you have any suggestion on how to understand what it means?

> My bet is it's mbed TLS misconfiguration, i. e. some mbed TLS config needed to handle credentials you provide is not set (some cipher or hash algorithm missing or something like that).

I would expect the example code I'm working with to be correctly configured so that it can actually work. If not, then can you suggest changes needed in order to make it work?

> But with the information provided, it's just a guess.

I have provided all the information needed to precisely reproduce everything I've done in exact detail. What additional information would you suggest be provided?

rlubos, 2024-06-18:
> > > [00:00:10.197,000]  aws: Failed to connect: -22 backoff delay: 394 ms
> > 
> > 
> > Have you tried to pinpoint the exact reason for getting `EINVAL` from `mqtt_connect()`?
> 
> Its a seemingly ambiguous error. Do you have any suggestion on how to understand what it means?

Use a debugger, see where it fails. `EINVAL` is so commonly used, that's it's hard to locate the actual source of the error based on the error code only. But as I said before, given that this error is returned when socket layer fails to load credentials, mbed TLS misconfiguration is quite probable reason.

> I would expect the example code I'm working with to be correctly configured so that it can actually work. If not, then can you suggest changes needed in order to make it work?

I would expect that example code works out of the box as well. But didn't you just provide a pretty large patch, modifying the example, including TLS credentials from a custom source? That's no longer the example code we provide in Zephyr, hence expectation that everything will work with the default configuration is amiss.

Looking at the custom CA certificate provided in the diff, I can already see it's using different hash algorithm:
[image]
So I'd start by enabling `CONFIG_MBEDTLS_SHA512=y`.

> > But with the information provided, it's just a guess.
> 
> I have provided all the information needed to precisely reproduce everything I've done in exact detail. What additional information would you suggest be provided?

Well, anything that helps isolating the issue would be helpful. Already mentioned feedback from the debugging session, mbed TLS logs perhaps. I don't own the board, so if reproducing on dedicated board is expectation here, I can't help.

closed 2024-06-18 by ASerbinski