zephyrproject-rtos/zephyr #74332: Problems with tests/subsys/pm and native_sim

jeremybettis, 2024-06-14:
**Describe the bug**

I did a clean checkout of Zephyr on a Debian rodete 20240525.01.00RL system using the Zephyr getting started guide. I have no local changes to zephyr at all.

But there are some tests under tests/subsys/pm that work fine with native_posix, but fail to build with native_sim.

**To Reproduce**

host toolchain fails on native_sim, passes on native_posix
```
(.venv) jbettis@jbettis13:~/zephyrproject/zephyr$ ZEPHYR_TOOLCHAIN_VARIANT=host west twister -p native_sim -p native_posix -s tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime 
Renaming output directory to /home/jbettis/zephyrproject/zephyr/twister-out.8
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-6330-gfe05606630f8
INFO    - Using 'host' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/jbettis/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - native_sim                tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime  ERROR : Build failure
ERROR   - see: /home/jbettis/zephyrproject/zephyr/twister-out/native_sim/tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime/build.log
INFO    - Total complete:    2/   2  100%  skipped:    0, failed:    0, error:    1
INFO    - 1 test scenarios (2 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 2 test configurations passed (50.00%), 0 failed, 1 errored, 0 skipped with 0 warnings in 8.93 seconds
INFO    - In total 2 test cases were executed, 0 skipped on 2 out of total 4 platforms (50.00%)
INFO    - 1 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/jbettis/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/jbettis/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/jbettis/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime on native_sim error (Build failure)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p native_sim -s tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime
INFO    - or with west:
INFO    - west build -p -b native_sim tests/subsys/pm/device_driver_init -T pm.device_driver_init.pm_device_runtime
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

llvm toolchain passes on both
```
(.venv) jbettis@jbettis13:~/zephyrproject/zephyr$ ZEPHYR_TOOLCHAIN_VARIANT=llvm west twister -p native_sim -p native_posix -s tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime 
Renaming output directory to /home/jbettis/zephyrproject/zephyr/twister-out.9
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-6330-gfe05606630f8
INFO    - Using 'llvm' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/jbettis/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    2/   2  100%  skipped:    0, failed:    0, error:    0
INFO    - 1 test scenarios (2 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 2 of 2 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 10.29 seconds
INFO    - In total 2 test cases were executed, 0 skipped on 2 out of total 4 platforms (50.00%)
INFO    - 2 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/jbettis/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/jbettis/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/jbettis/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

zephyr toolchain fails on native_sim
```
(.venv) jbettis@jbettis13:~/zephyrproject/zephyr$ ZEPHYR_TOOLCHAIN_VARIANT=zephyr west twister -p native_sim -p native_posix -s tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime 
Renaming output directory to /home/jbettis/zephyrproject/zephyr/twister-out.10
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-6330-gfe05606630f8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/jbettis/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - native_sim                tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime  ERROR : Build failure
ERROR   - see: /home/jbettis/zephyrproject/zephyr/twister-out/native_sim/tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime/build.log
INFO    - Total complete:    2/   2  100%  skipped:    0, failed:    0, error:    1
INFO    - 1 test scenarios (2 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 2 test configurations passed (50.00%), 0 failed, 1 errored, 0 skipped with 0 warnings in 8.91 seconds
INFO    - In total 2 test cases were executed, 0 skipped on 2 out of total 4 platforms (50.00%)
INFO    - 1 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/jbettis/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/jbettis/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/jbettis/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime on native_sim error (Build failure)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p native_sim -s tests/subsys/pm/device_driver_init/pm.device_driver_init.pm_device_runtime
INFO    - or with west:
INFO    - west build -p -b native_sim tests/subsys/pm/device_driver_init -T pm.device_driver_init.pm_device_runtime
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```
[zephyr_build.log](https://github.com/user-attachments/files/15841312/zephyr_build.log)
[host_build.log](https://github.com/user-attachments/files/15841313/host_build.log)

fabiobaltieri, 2024-06-14:
Seems related to `DEVICE_DEPS`:

```console
$ west build -p -b native_sim samples/hello_world
-- Zephyr version: 3.6.99 (/home/fabiobaltieri/zephyrproject/zephyr), build: v3.6.0-6330-gfe05606630f8
[82/84] Linking C executable zephyr/zephyr.elf
Generating files from /home/fabiobaltieri/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: native_sim
[84/84] Running utility command for native_runner_executable

$ west build -p -b native_sim samples/hello_world/ -DCONFIG_DEVICE_DEPS=y
...
-- Zephyr version: 3.6.99 (/home/fabiobaltieri/zephyrproject/zephyr), build: v3.6.0-6330-gfe05606630f8
[83/89] Linking C executable zephyr/zephyr_pre0.elf
/usr/bin/ld.bfd: warning: zephyr/zephyr_pre0.elf has a LOAD segment with RWX permissions
[87/89] Linking C executable zephyr/zephyr.elf
Generating files from /home/fabiobaltieri/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: native_sim
[89/89] Running utility command for native_runner_executable
```

it's enabled indirectly in those samples

aescolar, 2024-06-17:
As a note w gcc 12.3.0 it builds fine.
This is ld version dependent (the report is made with gcc 13.2.0)

aescolar, 2024-06-17:
Summary: the linker is just warning us about a potentially exploitable situation in the elf (executable code being in the same segment as write data). This in itself is not much of a real problem for the native targets as they are test targets you build with whatever you want and run on your own.
But the warning is the issue, as native builds would produce it at "random" resulting in twister failures in whichever sample/test.
The issue is not due to one particular sample or another, but due to the luck of how the linker splits the elf in segments and what collection ends up in one or another.
One option is just to disable the warning (`--no-warn-rwx-segments` if the linker supports it).
But given that a simple enough solution seems to be to change where we place things in the linker script for the native (posix arch) targets, I just sent a PR with that change instead.

PR #74410 (2024-06-19): native/posix arch: Reorder sections to avoid a RWE segment
GNU ld since 2.39, produces more security related warnings. Due to the way we ordered sections the linker may end up with a segment in the elf which contains both executable code and read and write data.
This results in this new versions of the linker warning us.

So, in the linker script,
let's move the executable native_sim_if section from being placed with the other native port related segments after .data to being placed after .text instead.

Fixes #74332
closed 2024-06-19 by nashif