zephyrproject-rtos/zephyr #74347: Kconfig tree littered with non applicable options

thedjnK, 2024-06-15:
**Describe the bug**
A MSPI Kconfig has appear that is always enabled despite having no applicable drivers for the hardware and gives a cmake empty source warning: `CONFIG_FLASH_MSPI`, additionally `CONFIG_I2C_AMBIQ_DMA` and `CONFIG_I2C_DMA_TCB_BUFFER_SIZE` are shown to non-ambiq platforms. Additional Kconfigs shown on completely non-applicable platforms:
`CONFIG_MTK_ADSP_TIMER`
`CONFIG_INTC_MTK_ADSP`
`CONFIG_INTC_ESP32`

There might be more present. These should be limited properly and not shown on incompatible hardware.

**Environment (please complete the following information):**
 - Commit SHA or Version used: 66630961627f4d8625e2dbe1cb7afe4abd690b2e

thedjnK, 2024-06-15:
@raffarost @swift-tk this also applies to your submissions, please send patches

rexut, 2024-06-15:
same issue: #74349

swift-tk, 2024-06-15:
One question, how do I keep CONFIG_FLASH_API with a mspi-device compatible? The device driver binding all included that but have a different compact string and DT_HAS_* macro.
If it is not possible, then it is kind of tedious to write all the same kconfig to all mspi devices.

thedjnK, 2024-06-16:
> One question, how do I keep CONFIG_FLASH_MSPI with a mspi-device compatible? The device driver binding all included that but have a different compact string and DT_HAS_* macro. If it is not possible, then it is kind of tedious to write all the same kconfig to all mspi devices.

I don't know what you mean, every different device has a different compatible name e.g. nordic flash and stm32 flash are both flash devices but they are not the same flash device, each has a different binding for it

AlessandroLuo, 2024-06-17:
will be fixed in https://github.com/zephyrproject-rtos/zephyr/pull/74356 and https://github.com/zephyrproject-rtos/zephyr/pull/74380

nashif, 2024-06-17:
duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/74349

nordicjm, 2024-06-17:
> duplicate of #74349

This is the original issue which lists all of the Kconfig options with issues which that issue does not:

> CONFIG_MTK_ADSP_TIMER
> CONFIG_INTC_MTK_ADSP
> CONFIG_INTC_ESP32

nashif, 2024-06-17:
> This is the original issue which lists all of the Kconfig options with issues which that issue does not:

then please closwe the other one once you have reopened this.

sylvioalves, 2024-06-19:
Is there anything related to `CONFIG_INTC_ESP32` that still requires fixing?

nordicjm, 2024-06-19:
> Is there anything related to `CONFIG_INTC_ESP32` that still requires fixing? If yes, how to reproduce/check?

```
  config INTC_ESP32
        bool "Interrupt allocator for Xtensa-based Espressif SoCs"
        default y if SOC_FAMILY_ESPRESSIF_ESP32 && !SOC_SERIES_ESP32C3 && !SOC_SERIES_ESP32C6
        help
          Enable custom interrupt allocator for Espressif SoCs based on Xtensa
          architecture.
```
This is visible and selectable to any board/soc when it should not be, it should be
```
        default y
        depends on SOC_FAMILY_ESPRESSIF_ESP32
        depends on !SOC_SERIES_ESP32C3 && !SOC_SERIES_ESP32C6
```

nordicjm, 2024-06-19:
@andyross fix needed for `CONFIG_MTK_ADSP_TIMER`

thedjnK, 2024-06-21:
Ping @andyross

PR #74535 (2024-06-21): drivers: intc: esp32: fix kconfig visibility
Kconfig options in those drivers are visible and selectable to any board/soc when it should not. This makes sure both depends on proper family.

Fixes #74347
closed 2024-07-08 by aescolar