zephyrproject-rtos/zephyr #74348: PWM Clock does not work if SOC clock is changed [NXP S32K1]

vChavezB, 2024-06-15:
**Describe the bug**
If I change the system clock configuration of the SOC NXP S32K146 through the SoC definition [here](https://github.com/zephyrproject-rtos/hal_nxp/blob/bee15c31e19ff1982583d9b750ef844d64320160/s32/soc/s32k146/src/Clock_Ip_Cfg.c), the PWM clock is not updated correctly.

**To Reproduce**
1. Change the system clock (`BUS_CLK`) which is named as `CORE_RUN_CLK` in [`Clock_Ip_Cfg.c`](https://github.com/zephyrproject-rtos/hal_nxp/blob/bee15c31e19ff1982583d9b750ef844d64320160/s32/soc/s32k146/src/Clock_Ip_Cfg.c). This can be done by for example changing the Predivider to 4 [here](https://github.com/zephyrproject-rtos/hal_nxp/blob/bee15c31e19ff1982583d9b750ef844d64320160/s32/soc/s32k146/src/Clock_Ip_Cfg.c#L524).

2. Create a PWM node for tests

```ini
my_pwm: my-pwm-node {
	pwms = <&ftm0 1 PWM_HZ(2400) PWM_POLARITY_NORMAL>;
	label = "My NXP PWM;
};

&ftm0 {
	clock-source = "system";
};
```

3. Add some source code to test this

```c
void test(){
  static const struct pwm_dt_spec pwm_out= PWM_DT_SPEC_GET(DT_ALIAS(my_pwm));
  pwm_set_pulse_dt(&pwm_out, pwm_out.period/2);  // 50 % duty cycle	
}
```

**Expected behavior**
That the PWM frequency corresponds to the one set in the device tree.

**Impact**
I have to manually modify the PWM frequency speed in the device tree to match 2400 Hz (in this example) each time I change the system clock speed.

## Possible solution

After checking the source code for the HAL NXP S32K146 I noticed that the FTM clock source is set to `SPLLDIV1_CLK `[here](https://github.com/zephyrproject-rtos/hal_nxp/blob/bee15c31e19ff1982583d9b750ef844d64320160/s32/soc/s32k146/src/Clock_Ip_Cfg.c#L312).

As per the Reference Manual I understand that the System clock is not `SPLLDIV1_CLK` but rather `CORE_RUN_CLK` defined in [`Clock_Ip_Cfg.c`](https://github.com/zephyrproject-rtos/hal_nxp/blob/bee15c31e19ff1982583d9b750ef844d64320160/s32/soc/s32k146/src/Clock_Ip_Cfg.c). 

[image]

vChavezB, 2024-06-15:
Could you have a look @manuargue  ?

manuargue, 2024-06-15:
Well we haven't supported running with other clocks configurations than the existing one in hal_nxp, but is good that you have tried it out.

I believe the problem is that `NXP_S32_FTMx_CLK` is referencing the `PCC_FTMx` clock (selected by `PCC_FLEXTMRn[PCS]`) and not the effective `FTM_CLK` selected by `FTMn_SC[CLKS]`.

Leaving the clock config in `Clock_Ip_Cfg.c` as is (`FTM0_CLK` to `SPLLDIV1_CLK`), but changing in devicetree:
```devicetree
&ftm0 {
    clocks = <&clock NXP_S32_CORE_CLK>;
};
```
does it solve your issue?

vChavezB, 2024-06-16:
Thanks for quick reply. I do not have the board with me but will try this out on the next days.

manuargue, 2024-06-16:
tested with `pwm usec ftm@40039000 1 1000 500`

1. with existing clock configuration:
[image]

2. changing the predivider to 4 [here](https://github.com/zephyrproject-rtos/hal_nxp/blob/bee15c31e19ff1982583d9b750ef844d64320160/s32/soc/s32k146/src/Clock_Ip_Cfg.c#L524)
[image]

3. same change in predivider + patch from https://github.com/zephyrproject-rtos/zephyr/pull/74355
[image]

vChavezB, 2024-06-17:
That looks good. Thanks for the test results.

Just as a small context on why I noticed this. I am trying to reduce the current consumption for my application and was testing different clock speeds. The other peripherals behaved the same but since I have some pwm outputs thats when I noticed this.

PR #74355 (2024-06-18): dts: arm: nxp: fix clock name for FTM nodes
Clock `NXP_S32_FTMx_CLK` reference the output of PCC FTM clock, but FTM has an internal clock mux to select the clock source of the counter, which for `ucans32k1sic` board is set to system clock. Fix the clock nodes to use the correct clock name. So far this was working because both `NXP_S32_FTMx_CLK` and `NXP_S32_CORE_CLK` are configured to the same frequency.

Fixes #74348

```
west twister -p ucans32k1sic --device-testing --device-serial=/dev/ttyUSB_ucans32k1sic  --fixture pwm_loopback   -T tests/drivers/pwm  -T samples/drivers/led_pwm/  -T sample [... cut]
commit 3799073e (2024-06-18): dts: arm: nxp: fix clock name for FTM nodes

Clock NXP_S32_FTMx_CLK reference the output of PCC_FTMx clocks, but
FTM has an internal clock mux to select the clock source of the
counter, which for ucans32k1sic board is set to system clock. Fix the
clock nodes to use the correct clock name. So far this was working
because both NXP_S32_FTMx_CLK and NXP_S32_CORE_CLK are configured to
the same frequency.

Fixes #74348
closed 2024-06-18 by nashif