zephyrproject-rtos/zephyr #74349: CMake warning because of new MSPI support

rexut, 2024-06-15:
**Describe the bug**

Kconfig leads to a wrong data base for CMake since commit https://github.com/zephyrproject-rtos/zephyr/commit/ece0c9b0d3b6d7a4107c49caf46f9b8375ba5d4e:

https://github.com/zephyrproject-rtos/zephyr/blob/66630961627f4d8625e2dbe1cb7afe4abd690b2e/drivers/flash/Kconfig.mspi#L4-L9

As soon as the high-level symbol `FLASH` is activated, which is a common procedure at application level, an MSPI driver is automatically activated. However, this is now only available for a single instance (MSPI ATXP032 driver). This means that the drivers for the SoC flash controllers can still be built, but unfortunately only with an extremely unattractive CMake warning:

```
CMake Warning at /.../workspace/zephyr/CMakeLists.txt:915 (message):
  No SOURCES given to Zephyr library: drivers__mspi

  Excluding target from build.
```

**To Reproduce**

Steps to reproduce the behavior:
1. properly arranged Zephyr West workspace
2. `west build -b nucleo_f746zg -p always -d build/cjmcu_752 zephyr/samples/drivers/flash_shell`
3. See warning above

**Expected behavior**

Application `zephyr/samples/drivers/flash_shell` will build without any warnings.

**Impact**

Showstopper, a really good QA pipelines leads build warnings to errors.

**Logs and console output**

Non.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.5
 - latest Zephyr main branch (v3.7.0-rc1)

**Additional context**

Non.

rexut, 2024-06-15:
@swift-tk and @nashif: Can one of you take a closer look at this? I assume that there is a systematic problem with the now existing transitive dependency:

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/drivers/flash_shell/prj.conf#L4:
```
CONFIG_FLASH=y
```
–> https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/Kconfig.mspi#L4-L9
```
config FLASH_MSPI
	default y
	depends on FLASH
	select MSPI
```
–> `build/zephyr/.config`
```
CONFIG_FLASH=y
CONFIG_FLASH_MSPI=y
CONFIG_MSPI=y
```

That isn't the intention of the sample for a Nucleo STM32xxxx board. There is no MSPI node in devicetree. The sample should only work with the STM32 SoC Flash controller, nothing else.

nashif, 2024-06-17:
duplicate it of https://github.com/zephyrproject-rtos/zephyr/pull/74409

closed 2024-06-17 by nashif