zephyrproject-rtos/zephyr #74353: Improve the ov5640 video driver to provide parallel interface (DVP) support.

CharlesDias, 2024-06-15:
**Is your enhancement proposal related to a problem? Please describe.**
The current ov5640 supports MIPI CSI-2 interface.

**Describe the solution you'd like**
Improve the ov5640 video driver to provide parallel interface support as well.

**Additional context**
Anyone planning on implementing this improvement?

Recently, the OV5640 video driver was added. However, it only supports the MIPI CSI-2 interface. I plan to implement this driver update, if no one is implementing it, and validate it on a STMicroelectronics board via DCMI driver.

/cc @ngphibang  @kartben @erwango @ABOSTM @FRASTM @josuah @loicpoulain @danieldegrasse

josuah, 2024-06-16:
I have worked a bit with that sensor in the past in parallel (non-MIPI) mode, so could give a hand.
Available in the chat.

This might be the first sensor that would be possible to configure in either modes.

There is a `source = <&phandle>;` devicetree property that would allow the controller to configure the sensor into the correct mode, maybe that is the way.

On Linux, the MIPI parameters are passed as part of the endpoint:
```
ports {
        port {
                endpoint {
                        mipi-lanes = <1 2 3 4>;
                        remote-endpoint = <&other>;
                };
        }
};
```

That opens these possibilities:

- Encode the selection of the mode in the devicetree
- Encode a phandle in the devicetree and have the controller call control commands to select the DVP or MIPI configuration for the sensor.

CharlesDias, 2024-06-16:
Thanks for your availability, @josuah! I am not a specialist, but could be a good opportunity to learn more about camera sensor and Zephyr drivers as well.

CharlesDias, 2024-06-16:
This was the partial DTS that I'm using in the initial tests with OV5640 sensor and ST DCMI driver on STM32H7B3I-DK board.

```
&i2c4 {
	pinctrl-0 = <&i2c4_scl_pd12 &i2c4_sda_pd13>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

	ov5640: ov5640@3c {
		compatible = "ovti,ov5640";
		reg = <0x3c>;
		status = "okay";
		reset-gpios = <&gpioa 0 GPIO_ACTIVE_LOW>;
		powerdown-gpios = <&gpioa 7 GPIO_ACTIVE_HIGH>;

		port {
			ov5640_ep_out: endpoint {
				remote-endpoint = <&dcmi_ep_in>;
			};
		};
	};
};

&dcmi {

	status = "okay";
	sensor = <&ov5640>;
	pinctrl-0 = <&dcmi_hsync_pa4 &dcmi_pixclk_pa6 &dcmi_vsync_pb7
		&dcmi_d0_pc6 &dcmi_d1_pc7 &dcmi_d2_pg10 &dcmi_d3_pc9
		&dcmi_d4_pc11 &dcmi_d5_pd3 &dcmi_d6_pb8 &dcmi_d7_pb9>;
	pinctrl-names = "default";
	bus-width = <8>;
	hsync-active = <0>;
	vsync-active = <0>;
	pixelclk-active = <1>;
	capture-rate = <1>;
	dmas = <&dma1 0 75 (STM32_DMA_PERIPH_TO_MEMORY | STM32_DMA_PERIPH_NO_INC |
		STM32_DMA_MEM_INC | STM32_DMA_PERIPH_8BITS | STM32_DMA_MEM_32BITS |
		STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_1_4>;

	port {
		dcmi_ep_in: endpoint {
			remote-endpoint = <&ov5640_ep_out>;
		};
	};
};
```

https://github.com/zephyrproject-rtos/zephyr/assets/27859307/1bf60853-ae4d-4bd8-a28b-a6dcab5d5a8f

CharlesDias, 2024-06-16:
@josuah, if I got your suggestion, the previous DTS should be updated to something like below? I'm using the `interface-type`property to specify the interface. 

**Parallel interface is used.**

```
&i2c4 {
	pinctrl-0 = <&i2c4_scl_pd12 &i2c4_sda_pd13>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

	ov5640: ov5640@3c {
		compatible = "ovti,ov5640";
		reg = <0x3c>;
		status = "okay";
	      
		reset-gpios = <&gpioa 0 GPIO_ACTIVE_LOW>;
		powerdown-gpios = <&gpioa 7 GPIO_ACTIVE_HIGH>;
		
	        interface-type = "dvp";
		port {
		    dvp_ep_out: endpoint {
			bus-width = <8>;
			hsync-active = <0>;
			vsync-active = <0>;
			pixelclk-active = <1>;
		        remote-endpoint = <&dcmi_ep_in>;
		    };
		};
	};
};

&dcmi {

	status = "okay";
	sensor = <&ov5640>;
	pinctrl-0 = <&dcmi_hsync_pa4 &dcmi_pixclk_pa6 &dcmi_vsync_pb7
		&dcmi_d0_pc6 &dcmi_d1_pc7 &dcmi_d2_pg10 &dcmi_d3_pc9
		&dcmi_d4_pc11 &dcmi_d5_pd3 &dcmi_d6_pb8 &dcmi_d7_pb9>;
	pinctrl-names = "default";
	capture-rate = <1>;
	dmas = <&dma1 0 75 (STM32_DMA_PERIPH_TO_MEMORY | STM32_DMA_PERIPH_NO_INC |
		STM32_DMA_MEM_INC | STM32_DMA_PERIPH_8BITS | STM32_DMA_MEM_32BITS |
		STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_1_4>;

	port {
		dcmi_ep_in: endpoint {
			remote-endpoint = <&dvp_ep_out>;
		};
	};
};
```

**Parallel interface is used.**

```
&i2c4 {
	pinctrl-0 = <&i2c4_scl_pd12 &i2c4_sda_pd13>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

	ov5640: ov5640@3c {
		compatible = "ovti,ov5640";
		reg = <0x3c>;
		status = "okay";
	       
		reset-gpios = <&gpioa 0 GPIO_ACTIVE_LOW>;
		powerdown-gpios = <&gpioa 7 GPIO_ACTIVE_HIGH>;
		
	        interface-type = "mipi";
		port {
		    mipi_ep_out: endpoint {
		        data-lanes = <1 2>;
		        clock-lanes = <0>;
		        bus-width = <2>;
		        remote-endpoint = <&mipi_csi_ep_in>;
		    };
		};
	};
};

&mipi_csi {
    status = "okay";

    /* Add other configs. /*
    
    port {
        mipi_csi_ep_in: endpoint {
            remote-endpoint = <&mipi_ep_out>;
        };
    };
};
```

josuah, 2024-06-16:
That would be something like this in Linux (with the `mipi-lanes` on the `mipi_csi` controller block I think), but currently, there is no semantics given in Zephyr for the `remote-endpoint`. Configuring it is pure documentation and not used in the driver.

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/st/stm32mp15xx-dhcor-avenger96.dtsi#n156

```
&dcmi {
        pinctrl-names = "default", "sleep";
        pinctrl-0 = <&dcmi_pins_c>;
        pinctrl-1 = <&dcmi_sleep_pins_c>;
        status = "disabled";

        port {
                dcmi_0: endpoint {
                        remote-endpoint = <&stmipi_2>;
                        bus-type = <5>;
                        bus-width = <8>;
                        pclk-sample = <0>;
                };
        };
};
```

And another example when MIPI is used:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/nxp/imx/imx6q-h100.dts#n337

```
&mipi_csi {
        status = "okay";

        port {
                mipi_csi2_in: endpoint {
                        remote-endpoint = <&tc358743_out>;
                        data-lanes = <1 2 3 4>;
                        clock-lanes = <0>;
                        clock-noncontinuous;
                        link-frequencies = /bits/ 64 <297000000>;
                };
        };
};
```

Although Zephyr does not have any of these, and configuring them here might not be necessary to get them to work.

Even having any `remote-endpoint` declared at all is just symbolic currently.

CharlesDias, 2024-06-16:
Thanks, @josuah ! I'll take a better look at those links.

josuah, 2024-06-16:
In Linux, the sensor detects whether it is configured on a MIPI bus or on a DVP/DMCI bus:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ov5640.c#n2712

This means a OV5640 camera devicetree node would be nested into either a MIPI or DMCI node/block, and this would sets which type it is configured with.

I think there lack a devicetree bindings declaration `on-bus: mipi;` which would allow to introduce that difference in type.

But of course, MIPI devices aare "on-bus" for their *configuration* interface: I2C, not MIPI or CSI!

This is the "on-bus" for their *remote-endpopint* that is considered here: sensor->**ep**.bus_type

This means we would probably need something like this to be merged first: #73023

Maybe a simple devicetree configuration could be done in the meantime? So that you are not blocked by this.
Some `bus-type = ` with a comment mentioning #73023, at least to allow other members to give their feedback on this.

ngphibang, 2024-06-17:
+1 from me. I think this would be a nice extension for ov5640. FYI, we are currently adding framerate changing to this driver but this shouldn't be a problem.

> I think there lack a devicetree bindings declaration on-bus: mipi; which would allow to introduce that difference in type.
>
> But of course, MIPI devices aare "on-bus" for their configuration [...]

@josuah , @CharlesDias : For this CSI / DVP (parallel) extension, as you said, we just need a "bus-type" property defined in the devicetree. We should group all these common properties into a "video-interfaces.yaml" binding (to avoid repeating them for each driver), some thing like [this ](https://github.com/zephyrproject-rtos/zephyr/pull/74415) which I picked from Linux. Then, in the driver, we can get the property by : `DT_PROP(DT_INST_CHILD(DT_INST_CHILD(id, port), endpoint), bus_type)`.

This is what actually done for display, as [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/display/display_mcux_elcdif.c#L371).

Support of remote-endpoint in Zephyr would be nicer but I think it is not needed and not a blocking point for this case.

josuah, 2024-06-17:
> some thing like https://github.com/zephyrproject-rtos/zephyr/pull/74415which I picked from Linux

Exciting! This is some good way to structure the various video drivers.

> Then, in the driver, we can get the property by : `DT_PROP(DT_INST_CHILD(DT_INST_CHILD(id, port), endpoint), bus_type)`.

I agree and this is good to have it this way in the meantime.

@CharlesDias would a `bus-type` property on the devicetree work for you?

Something like this:

```
&i2c4 {
		port {
			mipi_ep_out: endpoint {
				bus-type = <...>;
			};
		};
	};
};
```

CharlesDias, 2024-06-17:
> @CharlesDias would a `bus-type` property on the devicetree work for you?

@josuah, I think so!

Thank you, @josuah and @ngphibang! I'll wait a little more for other members' to confirm that no one is working on this.

CharlesDias, 2024-06-18:
> https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/st/stm32mp15xx-dhcor-avenger96.dtsi#n156

Hi, @josuah and @ngphibang. I hope you can help me with the following question. On the Linux driver, from the link above, we have duplicate properties declared for both endpoints, dcmi and i2c. I thought this duplication strange. My questions are:

1. Will we do this duplication on Zephyr?
2. On Linux, the `bus-type` is defined into DCMI endpoint instead of OV5640 endpoint, but I believe for Zephyr it doesn't make sense. So, will we add the `bus-type` in the OV5640 endpoint, right?

Thanks!

```
&i2c4 {
	stmipi: stmipi@14 {
                . . . 
		ports {
                 . . .
			port@2 {
				reg = <2>;
				stmipi_2: endpoint {
					bus-width = <8>;
					hsync-active = <0>;
					vsync-active = <0>;
					pclk-sample = <0>;
					remote-endpoint = <&dcmi_0>;
				};
			};
		};
	};

```

```
&dcmi {
        . . . 

	port {
		dcmi_0: endpoint {
			remote-endpoint = <&stmipi_2>;
			bus-type = <5>;
			bus-width = <8>;
			pclk-sample = <0>;
		};
	};
};
```

ngphibang, 2024-06-18:
@CharlesDias 

- Firstly, I think this is not a full description of the ov5640 camera node, as you can see, it lacks some things like `reset-gpios`, `pwdn-gpios`, etc. These things will be completed by a camera overlay dts and there you may find the `bus-type` property. Unlike Zephyr, overlay devicetrees are not integrated into Linux so you will not see them in upstream.
- Secondly, with the present of `bus-width`, it is a sign that this is a parallel interface as pointed out [here](https://github.com/zephyrproject-rtos/zephyr/pull/74415/commits/a757b3d2e16e175a02355d2b9281ead822a50ef9#diff-26092480ecf94c52029511ae7ecffae4a084d7cd172034e7a2b4496ae911bdeaR23).
- Thirdly, what we should base on is the ov5640 driver, you can see [here ](https://elixir.bootlin.com/linux/latest/source/drivers/media/i2c/ov5640.c#L3882) and [here ](https://elixir.bootlin.com/linux/latest/source/drivers/media/v4l2-core/v4l2-fwnode.c#L429), it really parses the endpoint of the ov5640 to get the `bus-type`.
- Then, as a reference, you can see [here](https://elixir.bootlin.com/linux/latest/source/arch/arm/boot/dts/renesas/r8a7742-iwg21d-q7-dbcm-ov5640-single.dtsi#L31) where `bus-type` is declared for ov5640 node.
- Finally, there is a _**low**_ possibility that we don't need to define `bus-type` in the sensor node but we get it from the dcmi via remote-endpoint. But, remote-endpoint is currently not supported in Zephyr.

So, in any above case, I think we should define `bus-type` in ov5640 node.

ngphibang, 2024-06-18:
For the "duplicated" properties, do you mean `bus-width` and `pclk-sample` ? IMO, it is because these properties does not necessarily have the same value for sensor and dcmi (although we see that they are the same in devicetree), for example `pclk-sample` can be rising edge for ov5640 and falling edge for dcmi. Anyway, in ov5640 driver, it is also obtained from the endpoint of ov5640 as [here](https://elixir.bootlin.com/linux/latest/source/drivers/media/i2c/ov5640.c#L2655)

CharlesDias, 2024-06-19:
> So, in any of above cases, I think we should define `bus-type` in ov5640 node.

Also make sense for me to define the `bus-type` in ov5640 node. I will continue with that in mind.

> For the "duplicated" properties, do you mean `bus-width` and `pclk-sample` ? IMO, it is because these properties does not necessarily have the same value for sensor and dcmi (although we see that they are the same in devicetree),

I did not know that! :-)

Thanks so much @ngphibang for the explanation!

CodingGhost, 2024-10-24:
@CharlesDias I am trying to use the ov5460 on a STM32H7a3zi. Im using latest Zephyr master but I get 
 video_ov5640: Unsupported pixel format or resolution.

Is there any PR I need for this to run thats not yet in master? Or did I do something wrong? Im using the lvgl sample but removed all lvgl code as I dont have a display.

mjs513, 2025-07-12:
@josuah 

@KurtE pointed out this issue so figured this would be a better spot to continue what I am finding with both the arduino giga and the portenta H7 while trying to get the OV5640 working with those devices.  This is a continuation of https://github.com/zephyrproject-rtos/zephyr/issues/92354.

Anyways I did manage to get several frames from the OV5640 using the H7 using the following configuration (using a ST7796 lib that @KurtE put together).

For the Overlay:
```
&i2c3 {
	status = "okay";
	ov5640: ov5640@3c {
		compatible = "ovti,ov5640";
		reg = <0x3c>;
		status = "okay";
		reset-gpios = <&gpioe 3 GPIO_ACTIVE_LOW>;
		powerdown-gpios = <&gpiod 5 GPIO_ACTIVE_HIGH>;

		port {
			ov5640_ep_out: endpoint {
				remote-endpoint-label = "dcmi_ep_in";
				bus-type = <5>;
			};
		};
	};
};

.....

&dcmi {
	status = "okay";
	/* ext-sdram = <&sdram1>; */
	pinctrl-0 = <&dcmi_hsync_pa4 &dcmi_pixclk_pa6 &dcmi_vsync_pi5
		     &dcmi_d0_ph9 &dcmi_d1_ph10 &dcmi_d2_ph11 &dcmi_d3_ph12
		     &dcmi_d4_ph14 &dcmi_d5_pi4 &dcmi_d6_pi6 &dcmi_d7_pi7>;
	pinctrl-names = "default";
	dmas = <&dma1 0 75 (STM32_DMA_PERIPH_TO_MEMORY | STM32_DMA_PERIPH_NO_INC |
			    STM32_DMA_MEM_INC | STM32_DMA_PERIPH_32BITS | STM32_DMA_MEM_32BITS |
			    STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_1_4>; //FULL for 7670, default FIFO_FULL

	port {
		dcmi_ep_in: endpoint {
      //remote-endpoint-label = "gc2145_ep_out";
      //remote-endpoint-label = "ov7670_ep_out";
      remote-endpoint-label = "ov5640_ep_out";
      //remote-endpoint-label = "ov2640_ep_out";
			bus-width = <8>;
			hsync-active = <0>;
			vsync-active = <0>;
			//pclk-sample = <0>;
      pclk-sample = <1>;
      
		};
	};
};
```
along with a 24Mhz camera clock
```
&pwm1 {
	/* Camera source clock */
	pwmclock: pwmclock {
		status = "okay";
		compatible = "pwm-clock";
		clock-frequency = <0>;
		#clock-cells = <1>;
		pwms = <&pwm1 1 PWM_HZ(24000000) PWM_POLARITY_NORMAL>;
		/* 12MHz for 7670, default is also 12MHz */
	};
};
```
I did have to make one change to OV5640.c.  Changed the pwdn pin from
```
	if (cfg->powerdown_gpio.port != NULL) {
		gpio_pin_set_dt(&cfg->powerdown_gpio, 0);
	}
```
to
```
	if (cfg->powerdown_gpio.port != NULL) {
		gpio_pin_set_dt(&cfg->powerdown_gpio, 1);
	}
```
Also in the camera sketch lib I added:
```
  struct video_frmival frmival = {
      .numerator = 1,
      .denominator = 30,
  };

  if(video_set_frmival(this->vdev, &frmival) ) {
        Serial.println("Failed to set video framerate");
  }
```

With this config I managed to get a couple dozen frames at 320x240 res before I got the video_stm32_dcmi: HAL_DCMI_ErrorCallback.  Unfortunately the images look like this:

[image]

With the giga having other issues but thats later - have to experiment more on that board

josuah, 2025-07-12:
> I did manage to get several frames from the OV5640 using the H7

Very glad to see that! 

> I did have to make one change to OV5640.c. Changed the pwdn pin from [...]
> along with a 24Mhz camera clock

We might need to add an extra devicetree property to support this but this is a minor config detail.

> Unfortunately the images look like this

It is possibly due to an inversion of the pixel format between `RGB565X` and `RGB565` in the capture sample: https://github.com/zephyrproject-rtos/zephyr/pull/79996

One other way to test is using the [USB Video Webam](https://docs.zephyrproject.org/latest/samples/subsys/usb/uvc/README.html) sample to turn the board into a webcam.

CharlesDias, 2025-07-12:
Hi @mjs513 and @KurtE, I believe this is a different issue and would be better addressed by https://github.com/zephyrproject-rtos/zephyr/issues/92354.

@josuah and @ngphibang, since the DVP has been merged, I suggest closing this issue. Do you agree?

josuah, 2025-07-12:
You are right, better keep issues focused on one aspect.

I'm closing this issue and opening a [forum](https://github.com/zephyrproject-rtos/zephyr/discussions) thread => https://github.com/zephyrproject-rtos/zephyr/discussions/93058

Thanks all!

closed 2025-07-12 by josuah